jgyates/genmon #132: Service required but GenMon Pi is not responded correctly

rhombixt, 2018-06-09:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below.

My service LED came on the Generator - looks like the battery needs inspecting. The Genmon Should set the service output active on the genpio to that it alerts me to a service required and it should log the proper alarm on the web page

{Please write here}

### Actual Behavior

Output did not respond with the service led lit - says that it's an unknown alarm 0000000014 on the webpage
[image]

Checking the display on the genset - says inspect battery - but ready to run led is still lit.

### Steps to Reproduce (including precondition)
Service LED is currently lit - I won't clear the alarm so that genmon can be fixed so it responds correctly

{Please write here}

### Screenshot on This Problem (if possible)

{Please write here}

### Your Environment

- Genereator Model: {Please write here}
- Generator Registers: {Please write here}
- Genmon Version: {Please write here}

jgyates, 2018-06-09:
You have not included your the version number of the software, your generator model or your registers.

Based on the info that I have (assuming you have and Evolution Air cooled), alarm code 14 is check battery, so you don't have the latest software (if my assumptions are correct)

You mentioned "genpio", are you referring to gengpio.py?   If you are referring to gengpio.py then alarm conditions take priority over service events. if you clear you alarm do does your LED light up?

Please provide more details as I cant not keep track of every detail of everyone using the software.
@rhombixt I know we have corresponded before I correspond with a lot of folks in this forum.

rhombixt, 2018-06-09:
Sorry - Model is 11kw Honeywell Air cooled 70580
I dont know the version of genmon but it's probably out of date. The issue is that the gengpio.py is supposed to activate when there is an alarm or service condition. That did not happen - the status_service GPIO 25 (pin 22) was supposed to activate like it was supposed to. This is CRITICAL for my application that the outputs work so this is a good test with a minor check battery condition. I don't want to clear the alarm yet - I want to make sure that the service light comes on to match the service light on the outside of the generator case.

Registers:
Num Regs: 208 
Not Changed: 630452836 Changed 10296748 Total Changed 0.02
0038:0000			0034:0000
0037:1071			0032:4000
05f5:0000			05f4:0000
05f1:0000			01f1:0010
05ed:0064			023b:0000
023e:0000			003b:0000
05fa:0000			0059:0000
0058:0000			0053:0000
0052:0000			0057:0000
0056:0000			0055:0000
0054:0000			0208:03f1
005c:0000			005a:0000
005f:0000			005d:0000
003c:0000			000d:0000
000e:162a			000f:0608
000a:0081			000c:000b
0005:1000			0006:0000
0000:0009			0001:00000014
0008:0000			0009:00f2
01f2:0c05			0012:0000
0011:009c			0010:0512
0007:0000			0019:0015
002b:0005			002c:1000
002a:6a75			002f:0000
002e:0000			0020:0000
0021:0000			001f:4956
001e:0184			001d:0000
001c:0000			001b:467c
001a:00bc			
01f4:33303031323930373335

rhombixt, 2018-06-09:
Or maybe I am getting an alarm condition vs a service condition - GPIO 24 is active at the moment - is that what you mean by alarms vs service events?

jgyates, 2018-06-09:
@rhombixt ,

Unfortunately  the controller does not work that way. It reports the highest priority state. Once that state is clear it reports the next highest priority state and so on. The alarm state is higher priority than service due state. If you notice your register 0001 in your registers. It is reporting a 14 hex.

    0001:00000014

this is an alarm of check battery

If the generator was reporting service due it would be a 1F hex like this:

  0001:0000001f

The nexux and evolution both work this way. The industrial version support reporting multiple conditions at the same time but the residential versions do not.

The software reports what the generator is reporting so since the generator controller is only reporting an alarm this is what the software is reporting. If your service is due then once you clear the alarm then it should report service due once the alarm is cleared.

rhombixt, 2018-06-09:
ok - but what about the unknown registers etc.. should I update in that case and are there any changes to that procedure ?

jgyates, 2018-06-09:
You are running a very old version of the software, yes I would update.

https://github.com/jgyates/genmon/wiki/4--Updating-to-the-Latest-Version

closed 2018-06-09 by jgyates

rhombixt, 2018-06-09:
I tried doing an update and got some merge issues.. the case is closed but I commented on it with the details.

https://github.com/jgyates/genmon/wiki/4--Updating-to-the-Latest-Version

jgyates, 2018-06-09:
you comment does not include the command you used to update. Did you use the script file or did you enter the commands manually. Are you seeing any issues or or you just posting details? you message is not clear.

rhombixt, 2018-06-09:
I used the command git pull and then the output was as follows:

Updating a7d2e73..dcc5f75
error: Your local changes to the following files would be overwritten by merge:
	ClientInterface.py
	OtherApps/serialtest.py
	README.md
	gengpio.py
	gengpioin.py
	genlog.py
	genmon.conf
	genmon.py
	genserv.py
	gensms.py
	myclient.py
	mymail.conf
	mymail.py
	mynotify.py
	startgenmon.sh
	static/genmon.css
	static/genmon.js
	static/index.html
Please, commit your changes or stash them before you can merge.
error: The following untracked working tree files would be overwritten by merge:
	templates/command_template.html
Please move or remove them before you can merge.
Aborting

rhombixt, 2018-06-09:
Nothing updated.

rhombixt, 2018-06-09:
I posted all the details on the forum under issue 132.. not sure if you are seeing that information.

jgyates, 2018-06-09:
None of these messages are odd, they are normal. 

From the link that I sent earlier:

git pull

This will update to the latest, or, if you have modified any files and want to overwrite them then type:

git fetch origin
git reset --hard origin/master

"git pull" will not update any files that have the time stamp different than the repository or that were not originally checked out from the repository. Because of the way you initially put the files there you need to contine reading the document and use the seconds set of commands.

rhombixt, 2018-06-09:
Sorry, I am not a programmer and the statement is confusing - I read it as you use "git pull" 
This will update to latest.

The if you have modified files and want to overwite them .. which I did not modify any files so I did not do the rest of the procedure.

jgyates, 2018-06-09:
This is the error you get when you have loaded the software twice. I am going to need more info.

"The application did not automatically start when I rebooted. " how did you come to this conclusion? you did not tell me specifically what is working and what is not working.

rhombixt, 2018-06-09:
Sorry, I am not good at this. It has been a long time since I installed this software and it has worked flawlessly so I don't need to really update it. I have the software start somehow when I reboot the machine. It starts the genmon.sh somehow along with all the genpio (input output files) . After the update - none of the outputs updated, when running my clientinterface.py from the command line, it shows that I am no longer receiving data. I killed the genmon app and started it manually and cut and paste the output above.

I can give you remote access to this machine if you want.

rhombixt, 2018-06-09:
I don't remember how most of this works - i know there is a file somehow that starts when I reboot the machine so that all these applications start together.

rhombixt, 2018-06-09:
Rebooted again - tried to run genmon.py
Traceback (most recent call last):
  File "genmon.py", line 946, in <module>
    MyMonitor = Monitor(ConfigFilePath = ConfigFilePath)
  File "genmon.py", line 95, in __init__
    self.log = mylog.SetupLogger("genmon", self.LogLocation + "genmon.log")
  File "/home/pi/genmon/genmonlib/mylog.py", line 26, in SetupLogger
    rotate = logging.handlers.RotatingFileHandler(log_file, mode='a',maxBytes=50000,backupCount=5)
  File "/usr/lib/python2.7/logging/handlers.py", line 117, in __init__
    BaseRotatingHandler.__init__(self, filename, mode, encoding, delay)
  File "/usr/lib/python2.7/logging/handlers.py", line 64, in __init__
    logging.FileHandler.__init__(self, filename, mode, encoding, delay)
  File "/usr/lib/python2.7/logging/__init__.py", line 911, in __init__
    StreamHandler.__init__(self, self._open())
  File "/usr/lib/python2.7/logging/__init__.py", line 936, in _open
    stream = open(self.baseFilename, self.mode)
IOError: [Errno 13] Permission denied: '/var/log/genmon.log'

rhombixt, 2018-06-09:
2018-06-09 15:18:09,405 : New version detected: Old = V1.9.23, New = V1.9.23
2018-06-09 15:18:09,527 : DetectController auto-detect override (Liquid Cooled). Liquid Cooled now is False
2018-06-09 15:18:11,987 : Looking up model info on internet
2018-06-09 15:18:18,147 : Found: Model: G0070580
2018-06-09 15:18:18,148 : Notice: product ID not found online. Using serial number to complete lookup. : generac_evolution.py:461
2018-06-09 15:18:18,166 : GenMon Loadded for site: 169 Sherwood Blvd W
2018-06-09 15:22:37,746 : GenMon Loadded for site: 169 Sherwood Blvd W
2018-06-09 15:25:42,981 : Error getting thread name in KillThread: ComWatchDog
2018-06-09 15:25:49,063 : Generator Monitor Shutdown
2018-06-09 15:26:53,151 : Validation Error: Invalid register length (Start) 0130 1000

jgyates, 2018-06-09:
Do you have my email address?

jgyates, 2018-06-09:
if you do then send me the remote info for login and I will take a look at it this evening.

rhombixt, 2018-06-09:
yes

rhombixt, 2018-06-09:
2018-06-09 15:22:37,746 : GenMon Loadded for site: 169 Sherwood Blvd W
2018-06-09 15:25:42,981 : Error getting thread name in KillThread: ComWatchDog
2018-06-09 15:25:49,063 : Generator Monitor Shutdown
2018-06-09 15:26:53,151 : Validation Error: Invalid register length (Start) 0130 1000