jgyates/genmon #133: New Serial Port Timeouts - Used To Work Fine

bowood, 2018-06-09:
Long time user.  Love the software.  Today I upgraded to latest software via the genmonmaint.sh process.  After doing this I get continuous mymodbus.log timeouts:

2018-06-09 11:03:35,947 : Error: timeout receiving slave packet for register 02f Buffer:0
2018-06-09 11:03:38,021 : Error: timeout receiving slave packet for register 01 Buffer:0
2018-06-09 11:03:40,093 : Error: timeout receiving slave packet for register 0a Buffer:0

So I downgraded but could not find a version that worked properly again.  In desperation I did some tweaks to current version.  Specifically I added 1s onto the timeouts:

                #msElapsed = self.MillisecondsElapsed(SentTime)
                msElapsed = self.MillisecondsElapsed(SentTime) + 1000

This kind of works, but still getting too many timeouts:

        Serial Stats :
            Packet Count : M: 6557, S: 6072
            CRC Errors : 0 
            CRC Percent Errors : 0.00
            Packet Timeouts : 484
            Discarded Bytes : 0
            Comm Restarts : 0
            Packets Per Second : 10.86
            Average Transaction Time : 0.0308 sec

My theory is that I'm getting more timeouts than I used to get, and that they are more problematic during genmon startup so my tweak brute forces it through the genmon startup period but works ok during the rest of the run.

Is my MAX232 adapter dying?  Is my Nexus controller struggling?  My raspberry pi 3 going bad?  I'm at a loss for what could cause this.  Thanks for any suggestions.

- Generator Model: Generac 22kw air cooled
- Genmon Version: V1.9.22

jgyates, 2018-06-09:
I don't see anything in the logs that is glaring. Did you do any other updates, like to the system system software?

What does the monitor page show for CPU utilization? I assume you have tried rebooting? Also, what is the version that you initially upgraded from and how did you perform the upgrade?

One thing your could try is to make a copy of your /etc/genmon.conf file and then overwrite your file with the latest from the repository. I do not include downgrading in my testing so I would stick with the latest version.

jgyates, 2018-06-09:
This is just a suggestion, I don't have any specific data that points me to a problem, but you could try to update your pi to the latest os software and firmware.

     sudo apt-get update
     sudo apt-get dist-upgrade
     sudo rpi-update

On my pi I run these commands fairly regularly.

jgyates, 2018-06-09:
you will need to reboot after these commands.

bowood, 2018-06-09:
Tried rebooting.  Switched to the genmon.conf from the repository.  cpu looks close to idle at all times. 
 Guesstimated V1.9.7 for previous version, but not certain, could be older.  We had two tornadoes come through on May 15th giving us 4 days of continuous outage.  I think I upgraded after that, but might have been on a version from before that.  I'm suspicious of the msElapsed update, but I haven't thought through your calculation:

https://github.com/jgyates/genmon/commit/c696614de543fc3779457e5c9e916eb46b978013

I'll update my pi.  Probably out of date there.

jgyates, 2018-06-09:
It is almost as if your OS tick time is off which would cause problems with time.sleep, but I have no data to support that, only the symptoms. If the data rate was off then you would see serial errors or no data. 

the changes that you point out make the modbus wait time include the actual data rate instead of assuming a rate of 9600. This should only be different than 9600 baud if you use the /OtherApps/modbusdump.py . Also that change is from a month ago and a lot of people have been using this since then with no timeout issues. Of course it could be that change coupled with something specific about your system. 

Are your running raspbian Jessie, stretch or some other distro?

jgyates, 2018-06-09:
also, let's look at your pip libraries, type this command and post your output:

     pip freeze

jgyates, 2018-06-09:
It should not update to stretch, it will just update the kernel and any bug fixes, security patches, etc that are specific to jessie. I don't think there is a way to "upgrade" from jessie to stretch reliably.

bowood, 2018-06-09:
I see it downloading stretch packages, so I guess I'm on stretch:

Get:233 http://archive.raspberrypi.org/debian **stretch**/main armhf python3-fourletterphat all 0.1.0 [7,762 B]

Maybe half way done.

jgyates, 2018-06-09:
This will tell you what release your are running:

     cat /etc/os-release 

PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)"
NAME="Raspbian GNU/Linux"
VERSION_ID="8"
VERSION="8 (jessie)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

bowood, 2018-06-09:
Assuming this is reliable when half way through a dist-upgrade:

$ cat /etc/os-release 
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"

jgyates, 2018-06-09:
FYI, I checked in few updates but there is nothing related to serial or modbus, only web UI updates so if you see V1.9.23 that is what the update was for. Just thought I would let you know since you are in the middle of an issue.

bowood, 2018-06-09:
Finished the update/distupgrade/rpi updates.  Hundreds of packages updated.  I resynced to the repository to remove my +1000ms hack.  Very slow startup with lots of mymodbus.log timeouts, but it did come up.  Maybe a little better on timeouts, but hard to say.  I couldn't get it to this point without the hack last time, so I think its progress.

    Monitor :

        Generator Monitor Stats :
            Monitor Health : OK
            Controller : Evolution, Air Cooled
            Run time : Generator Monitor running for 0:06:40.
            Generator Monitor Version : V1.9.23

        Serial Stats :
            Packet Count : M: 2349, S: 2242
            CRC Errors : 0 
            CRC Percent Errors : 0.00
            Packet Timeouts : 106
            Discarded Bytes : 0
            Comm Restarts : 0
            Packets Per Second : 11.47
            Average Transaction Time : 0.0314 sec

Here are my pip library versions.

$ pip freeze
automationhat==0.1.0
blinker==1.3
blinkt==0.1.2
Cap1xxx==0.1.3
chardet==2.3.0
click==6.7
colorama==0.3.7
configparser==3.5.0
crcmod==1.7
cryptography==1.7.1
drumhat==0.1.0
enum34==1.1.6
envirophat==1.0.0
ExplorerHAT==0.4.2
Flask==0.12.2
fourletterphat==0.1.0
gpiozero==1.4.1
idna==2.2
ipaddress==1.0.17
itsdangerous==0.24
Jinja2==2.10
keyring==10.1
keyrings.alt==1.3
MarkupSafe==1.0
mcpi==0.1.1
microdotphat==0.2.1
mote==0.0.3
motephat==0.0.2
numpy==1.12.1
oauthlib==2.0.1
phatbeat==0.1.1
pianohat==0.1.0
picamera==1.13
picraft==1.0
piglow==1.2.4
pigpio==1.38
Pillow==4.0.0
pyasn1==0.1.9
pycrypto==2.6.1
pygame==1.9.3
pygobject==3.22.0
pyinotify==0.9.6
PyJWT==1.4.2
pyOpenSSL==16.2.0
pyserial==3.2.1
pytz==2018.3
pyxdg==0.25
rainbowhat==0.1.0
requests==2.12.4
requests-oauthlib==0.7.0
RPi.GPIO==0.6.3
RTIMULib==7.2.1
scrollphat==0.0.7
scrollphathd==1.2.1
SecretStorage==2.3.1
sense-emu==1.0
sense-hat==2.2.0
simplejson==3.10.0
six==1.10.0
skywriter==0.0.7
sn3218==1.2.7
spidev==3.3
touchphat==0.0.1
twython==3.4.0
urllib3==1.19.1
Werkzeug==0.14.1

jgyates, 2018-06-09:
Looking at your serial stats:

        Packets Per Second : 10.86
        Average Transaction Time : 0.0308 sec

Average transaction time is correct, Packets per second is not, normally this would be around 64. This tells me that when you do transmit and the controller responds that it responds in the expected timeframe (e.g. average transaction time). One thing that could cause the behavior we are seeing is an intermittent connection on your transmit line. This would cause no CRC or other serial errors on the pi, but would cause timeouts since the controller would be receiving the bad packets and not responding if the CRC was bad.

It is possibly that this condition occurred before you upgraded. The way the software works is that there is a thread that only loops, reading the list of mod bus registers and stores them. Other threads read the stored values and act on them. So if you started have a bad tx connection then you will stop updating registers that you do not get a response, but the other threads will continue to use the cached values. Based on the registers that were sent, the majority of things that are missing are the large modbus transactions (ones that read multiple registers at once) like the serial number registers or log registers.

jgyates, 2018-06-09:
Here is the output of mine:

argh==0.26.2
asn1crypto==0.24.0
certifi==2018.4.16
cffi==1.11.5
chardet==3.0.4
click==6.7
configparser==3.5.0
crcmod==1.7
cryptography==2.2.2
enum34==1.1.6
Flask==0.12.2
idna==2.6
ipaddress==1.0.22
itsdangerous==0.24
Jinja2==2.9.6
MarkupSafe==1.0
pathtools==0.1.2
pycparser==2.18
PyJWT==1.6.4
pyOpenSSL==18.0.0
pyowm==2.8.0
pyserial==2.6
python-apt==0.9.3.12
pytz==2017.3
PyYAML==3.12
requests==2.18.4
RPi.GPIO==0.6.3
six==1.11.0
twilio==6.14.3
urllib3==1.22
virtualenv==15.1.0
Werkzeug==0.12.2
wiringpi==2.32.1

bowood, 2018-06-09:
hmm, maybe RFI?  My cable is not shielded.  If I recall I used thermostat wire from the serial port to the molex.  This was rock solid in the past though.  Maybe something is loose.  Not looking forward to taking it apart given some hornets have decided that the breaker compartment is a good spot for a nest.  It'll take me a few weeks to get to it but I'll dismantle and check the connections and re-open ticket if I find anything interesting.  Feel free to close ticket for now.  Thanks for the diagnostic thoughts.

jgyates, 2018-06-09:
our pyserial version is different, I will upgrade mine just to rule that out.

jgyates, 2018-06-09:
Installed pyserial with apt-get not pip so mine is already update for my distro (jessie). I have a test pi that has stretch, I will check the version there. I am guessing it the difference is due to stretch vs jessie

I thought I had an RF problem for a while (I have a 48kw generator), but it turns out to be vibrations. I ended up replacing my cable with a few of [these](https://smile.amazon.com/gp/product/B000M802RG/ref=oh_aui_search_detailpage?ie=UTF8&psc=1) and I stopped dropping packets when the generator started.

bowood, 2018-06-09:
Maybe after 4 days of continuous running something came loose.  I'll take it all apart and check the connections.  Good catch that the checksums will be on RX and could simply be that many of my TX aren't getting through.  I got the serial/ttl converter linked on the wiki, but there were some bad reviews in there.  I might also have overheated it and broken it.  I'll see if anything obvious on that while I have it open.

jgyates, 2018-06-09:
ok. Going to close this thread. Feel free to post here if you have questions.

closed 2018-06-09 by jgyates

bowood, 2018-06-09:
I'm noticing a lot of undervoltage throttling errors. I have a USB cable. I'll try that too when I get it out next time. Maybe it messes up TX.

bowood, 2018-06-16:
I got it apart.  Added the USB cable you have in the wiki and reseated everything.  No more under voltage errors, and no more lost packet errors.  I actually suspect the undervoltage/throttling errors are likely culprit.  Actually powering off generator and cable at the same time.  I figure with common ground that should be ok.  Not sure how the generator provides the power.  Via a drop resistor?  Maybe the Pi 3 is simply too much for it.

jgyates, 2018-06-16:
outstanding. Thanks for the feedback. I am glad it is working now.