jgyates/genmon #135: Did I damage my controller, or has anyone gotten a USB serial adapter to work?

JustApu, 2018-06-12:
I had a Pi Zero successfully running genmon via a MAX3232 but apparently the waterproof enclosure it was all in wasn't. The Pi & serial/TTL board both got damp and died.

While waiting for a new MAX3232 to be delivered, I replaced it with a [USB serial adapter](https://smile.amazon.com/dp/B07B9SC1TL) that I had and a new Pi. But genmon is not talking to the generator.

a) What are the chances I've damaged the Modbus connection on the generator? Anyway I can test it besides actually getting genmon working?

b) Has anyone gotten a USB serial adapter working?

Some notes that might be helpful:

- myserial.log reports
`2018-06-11 22:29:13,281 : Error in SerialDevice:Flush : /dev/ttyUSB0:(9, 'Bad file descriptor') : myserial.py:131`
`2018-06-11 22:29:13,285 : Error in SerialDevice:Flush : /dev/ttyUSB0:(9, 'Bad file descriptor') : myserial.py:130`
`2018-06-11 22:29:13,400 : Resetting SerialDevice:ReadThread Error: /dev/ttyUSB0:device reports readiness to read but returned no data (device disconnected?) : myserial.py:90`
`2018-06-11 22:29:16,521 : Resetting SerialDevice:ReadThread Error: /dev/ttyUSB0:device reports readiness to read but returned no data (device disconnected?) : myserial.py:90`
`2018-06-11 22:29:16,643 : Resetting SerialDevice:ReadThread Error: /dev/ttyUSB0:device reports readiness to read but returned no data (device disconnected?) : myserial.py:90`

- the Pi is powered via the Molex connector. The four pins are split out into five connectors -- ground & power to the Pi and ground/rx/tx to the serial adapter.

- throttled=0x0

- serialconfig.py & serialtest.py both report no errors. The adapter stays listed in the output of lsusb and no disconnect messages or other errors in /var/log/messages

[image]

[IMG_0258.MOV.zip](https://github.com/jgyates/genmon/files/2092345/IMG_0258.MOV.zip)

jgyates, 2018-06-12:
Hi @JustApu ,

I can answer definitively, but I can share some of my experiences.  USB does work, however you need to be very mindful with the power consumption of the USB adapter as not all USB serial adapters have he same power requirements and also, not all are "pi compatible", but is is possible if everything is matched up correctly. If your new pi that you you are using is a pi zero then the matching number of  compatible usb serial adapters is smaller, and if you are powering over the molex connector it could limit it even more just based on empirical data as the power supplied by the molex *appears* to be less (have not actually measured it, but I have seen a copule of issues with USB adapters when powered over the molex). I can't speak for the pi zero on the firmware, but I know on the pi 2 and 3 if you firmware is not up to date then you may not get the disconnect messages in /var/log/messages. You can update your firmware to the latest with these commands and see if you get USB disconnect messages in your log:

    sudo apt-get update
    sudo apt-get dist-upgrade
    sudo rpi-update

the first two commands update the system software and the last updates the firmware. It is recommended to update the system software before the firmware update. 

In general if you get a "device reports readiness to read but returned no data (device disconnected?)" error this means the serial device once once there but now it is not. 

the good news is that you have a lot of discarded bytes, which indicates that you have received some data, but not a full packet. Discarded bytes are bytes received but they do not align with a packet. If you serial port is going in and out then this could happen.

JustApu, 2018-06-12:
Thanks, @jgyates 

update / dist-upgrade were already done before I put this Pi Zero W in service. I just did the firmware update with no changes that I can tell. I'll try powering the Pi from a USB battery pack tomorrow and only using the ground/TX/RX from the generator and see if that makes a difference. Otherwise, its just waiting for a new serial/TTL converter chip to arrive.

liltux, 2018-06-20:
@JustApu I have used a USB/Serial adapter, both with prolific and tdi chipsets.  I have not tested on an air cooled home standby yet but I have used on industrial with serial port.  The only thing populated on the serial port on genset is TX, RX, GND/SHLD.  So I am assuming as long as the TX, RX, and GND are routed correctly between the serial adapter and the controller it should work.  @jgyates I can post something here later once I get my cable made and tested.

JustApu, 2018-06-21:
Thanks, @liltux. I did try powering the Pi from the USB battery pack (disconnecting only the +5V pin so that the shared ground remained) and had even worse communications than before. So I'm back to waiting for the new MAX3232 to arrive. Fingers crossed...

liltux, 2018-06-22:
@JustApu , @jgyates Ok, so I have genmon running on rpi0w with a prolific chipset serial to usb adapter powered by a wall wart and connected and communicating with an air cooled evolution controlled generator.  Works like a charm.  Requirement is the pins 7, 8, and 2 (gnd) must be in your cable between the evolution controller and the serial adapter.  Also you would need to make the needed changes to genmon.conf for using your adapter, i.e. /dev/ttyUSB0.  see rough pics below.
[image]
[image]
[image]

jgyates, 2018-07-29:
Closing this issue for now. Feel free to reopen or comment in this thread if you are still having issues.

closed 2018-07-29 by jgyates