jgyates/genmon #136: Long range wifi antenna suggestions or link?

jpt3333, 2018-06-13:
My generator is about 150 ft from my router, and I have tried 2 different USB wifi "extended range" antennas to no avail, getting no signal at my Pi at the generator. Does anyone have a suggestion for a product or link that would bridge this difference via wifi?

@jgyates I noticed that you have a pic of a pretty hefty looking antenna. Is there a link anywhere for that product? Much appreciated for any help!

jgyates, 2018-06-13:
@jpt3333 

I have used a couple of different ones that i got from https://www.data-alliance.net. The one in the picture is this one i believe:

https://www.data-alliance.net/antenna-15dbi-2-4ghz-omni-directional-wifi-vertical-polarization/

My generator is also about 150 feet away from the closest access point. I was never really happy with the performance I was getting, but probably due to the USB adapter or my RF cable. I eventually bit the bullet and ran ethernet to my generator. the odd thing was that I was using the pi 3 and enabled the internal wifi as well and I got equal or slightly better with the internal adapter, but like I mentioned, it could be because of the USB adapter. I was going to attempt to user a Pi Zero, which has a way to connect an external antenna:

https://hackaday.com/2017/03/07/adding-an-external-antenna-to-the-raspberry-pi-zero-w/

But I decided to skip that and just go with Ethernet. If you go the wifi round you may consider the pi zero with an external antenna. Data alliance has omni and uni directional antennas as well. 

I have also bridged longer gaps with some Mikrotik products like [this](https://smile.amazon.com/gp/product/B00BTKPTNI/ref=oh_aui_search_detailpage?ie=UTF8&psc=1), but you would need to determine how to power it from your generator (possibly the generator battery to USB to PoE...) .

jpt3333, 2018-06-13:
> directional-wifi-vertical-polarization/
> to-the-raspberry-pi-zero-w/

Awesome much thanks, perfect!

closed 2018-06-15 by jgyates