jgyates/genmon #138: Question - During Utility Outage

jonathanpisarczyk, 2018-06-19:
### Expected Behavior
Question - During utility outage should I be able to use the "Stop Generator" button in the web gui to shut down the generator? 

### Actual Behavior
I had a utility outage yesterday and clicking the "stop generator" button and confirming with the pop up did nothing. 

### Steps to Reproduce (including precondition)
I tried to stop generator a couple times. 

### Your Environment

- Genereator Model: Evolution Air Cooled G0062500 (20Kw)
- Generator Registers: Submitted using web gui
- Genmon Version: V1.9.23

jgyates, 2018-06-19:
Hi @jonathanpisarczyk 

the software does not do any thing except pass the stop command to the firmware so this behavior is controlled by the controller firmware. There are a few things that generac requires the owner to physically be at the generator (resetting alarms, turning off the generator if it started from a utility loss). 

Thanks
Jason

jonathanpisarczyk, 2018-06-19:
@jgyates Thanks.  Makes sense.

closed 2018-06-19 by jgyates