jgyates/genmon #139: CPU temps high after update

jcherney, 2018-06-20:
Hi,

Just updated to 1.9 from 1.1. Previously CPU temps were in the 57-60 deg range. Now they are in the 118-121 deg range. Seems really high.

Any thoughts on why I’m seeing this?

jcherney, 2018-06-20:
[image]

jgyates, 2018-06-20:
I am guessing what you were  reading before was Celsius and what you are reading now is Fahrenheit. The UI has an option to display Fahrenheit. If you go to the settings page there is an option to "Use Metric Units" If enabled it will display in  Celsius. 60 C is normal for a pi, so is 116 F.

When the platform stats first rolled out it was metric only, then we added the weather stuff and gave an option to display metric or imperial units.

jcherney, 2018-06-20:
Well that makes sense- thanks! I’ll check that out.

closed 2018-06-21 by jgyates