jgyates/genmon #141: Performance differences between versions

gmqest, 2018-06-24:
First, thanks to the developers and several supportive users for this terrific resource. If this question is not appropriate for this forum, please let me know or move it.

We have just updated our older build from 1.4 to the latest build 1.9.27. We note that the prior installation consumed roughly 2-4% of the CPU resources on our raspberry pi zero. The new build appears to be using roughly 30-40% CPU (python process mainly).

The new build has many valuable new features, but the fundamental feature set of 1.4 was fine so we are assessing whether this build is necessary for us on our hardware. 

Is our observed performance difference expected (are others seeing similar CPU usage)? If so, are there parameters we can disable to decrease the CPU usage?

jgyates, 2018-06-25:
@gmqest 

If you are seeing continued CPU utilization of 30-40 percent then something is not acting as it should. Here are my comments / questions regarding this issue:

1) When the pi starts up CPU utilization is higher due to other processes starting so make sure you are reading the CPU utilization once the system has been running for a few minutes.
2) How are you determining the CPU utilization? The software will display the CPU utilization on the monitor tab. Note that the "top" command will show load average and CPU utilization, which in linux are two different metrics.
3) Check your logs for errors. If your system is not configured properly then you could have higher CPU utilization
4) there is always the possibility of a bug that could cause this, but I have not heard of anyone reporting CPU utilization in this range and I am not seeing it on my system.

mattm55, 2018-06-25:
From a command prompt you can also run "top" to see what's using the system resources. That might give you an indication of what's really bringing you to 30-40%.

Matt

jgyates, 2018-06-25:
Do determine which python scrip is using the CPU utilization, make a note of the process ID  (PID is first column) using the most CPU time. Then type this command:

     ps aux | grep python

This will display the process ID (second column) and full path of each process that has python in its load command (not this will include the 'sudo' command as a process so you will see somethings twice if you load multiple commands on one line like "sudo python program")

Find the process ID that you made a note of from top and cross-reference it to the output of the "ps" command to determine which python program is taking up the cpu time.

Regarding the logs. The primary logs are in the /var/log directory:

/var/log/genmon.log
/var/log/genserv.log
/var/log/mymail.log
/var/log/myserail.log
/var/log/mymodbus.log

Other logs my be there as well if you are loading programs besides genmon and genserv. 
 
You can also send the logs by typing this command in your browser (if you are using the web interface):

       http://piIPaddress:portcmd/sendlogfiles

for example:

      http://192.168.1.50:8000/cmd/sendlogfiles

just add /cmd/sendlogfiles to the end of the URL you normally use to access the web interface.

Are you running programs besides genmon and and genserv?
Are you using startgenmon.sh to load the programs?

gmqest, 2018-06-25:
@jgyates See output of ps aux... below. The main CPU consumer is PID 714:

root       709  0.0  0.7   7216  3436 ?        S    11:59   0:00 sudo python /home/pi/genmon/genmon.py
root       714 28.2  2.9 116356 13032 ?        Sl   11:59   1:09 python /home/pi/genmon/genmon.py
root       732  0.0  0.7   7216  3220 ?        S    11:59   0:00 sudo python /home/pi/genmon/genserv.py
root       737  1.0  3.3  19128 15088 ?        S    11:59   0:02 python /home/pi/genmon/genserv.py
root       808  0.9  3.0  22784 13604 ?        S    11:59   0:02 /usr/bin/python -O /usr/share/wicd/daemon/monitor.py

We will go through all of the logs carefully and let you know if there is anything unusual (or send them to you if we do not understand something).

We are running Nagios to monitor the genmon process using your script.

We do use startgenmon.sh to load the programs from upon reboot with cron.

jgyates, 2018-06-25:
I would do one of two things (or both)

1) Look at your logs to see if there are any errors.
2) Validate that you have replaced your /etc/genmon.conf file with the latest one from the repository.

Normally this would not be an issue, but since you are updating from version 1.4 to 1.9.27 there have been a lot of changes since then and what is likely happening is that you have a feature misconfigured.

If you have not done this then you can perform the following steps:

save your old configuration file:
     
     cp /etc/genmon.conf /etc/genmon.conf.bak

copy the new file:
   
     sudo cp /home/pi/genmon/genmon.conf /etc/genmon.conf

restart the sofrtware

       ./startgenmon.sh restart

You email information would be in mymail.conf so noting to speak of has chanted there. Then go the the web interface settings page and fill in your site name, and any other settings specific to your site.

Or you can attempt merge the old genmon.conf fiile with the new one but the former is likely easier.

gmqest, 2018-06-25:
[logs.zip](https://github.com/jgyates/genmon/files/2134011/logs.zip)
@jgyates 

1. We have been through all of the logs you referenced and cannot find an obvious problem (only the usual intermittent errors). The logs are attached - the CPU issues began on the 24th when we upgraded the build.

2. We were not using the latest /etc/genmon.conf so did replace that file with the one from the repository per your instructions. If anything, this version appears to be using even a bit *more* CPU.

A few other notes which may be obvious: we have updated and upgraded the server to latest version so of python, rebooted and restarted genmon many times. Thanks for looking into this, let us know if the logs point you to any issues we missed.

jgyates, 2018-06-25:
I did not see your myseral.log file in the zip.

Nothing is jumping out as a problem in the logs you did send, however you do have some entries that you may need to track down:

      Validation Error: Invalid register length 

This error occurs when you ask for a modbus register but recieve one of a different length. If you see any of these it can mean you have a communication error, which could have multiple causes.

Also you have this line multiple times:

New version detected: Old = V1.9.27, New = V1.9.27

This message  only occurs when there is not version information in /etc/genmon.conf. You should see this once when you first use a new conf file,but your log has it multiple times. Did you copy the file twice?

I will need more info on what OS and hardware platform you are using and anything unique about your use of the software to help further. Since you are the only person that I am aware of with this issue, at this point it appears that this is unique to you system in some way. If we find what is unique to your system we can potentially compensate for this in the  software if that is appropriate, depending on what we find.

Also what version of python are you using and have you loaded any other modules that would effect the runtime environment of python? What is the output of this command:

      python -V

gmqest, 2018-06-25:
@jgyates 

There is a typo in both references to the missing log above so we missed it. Attached is that log (**myserial**.log) as well as hard/software details and all python modules.

In our troubleshooting attempts, we may have overwritten the conf files twice (no more).

This raspberry pi zero does no activity other than genmon (and the associated Nagios monitoring). Happy to help troubleshoot further if there is anything more we can add, but if we are truly the only ones with the problem our plan is to delete and reinstall genmon completely. Following that we will revert to v1.4.

[logs2.zip](https://github.com/jgyates/genmon/files/2134281/logs2.zip)

gmqest, 2018-06-25:
@jgyates 

We had most of those settings already, but have done exactly as indicated. All permutations result in the same CPU usage for python and specifically genmon.py.

- "Do Not Check for Outages", "Sync Daylight Savings Time" and "Sync Time" **ENABLED** (green in interface)
- "Outbound Email Settings", "Inbound Email Commands Processing" and "Display Current Weather" **DISABLED**

The only difference from our previous configuration is "Do Not Check for Outages" and "Outbound Email Settings". We also tried *disabling* "Sync Daylight Savings Time" and "Sync Time" as a permutation.

jgyates, 2018-06-25:
Can you send me your registers? If you enable email you can do this from the monitor page, or you can get the output using "python ClientInterface.py" with the "generator: allregs" command.

this will allow me to simulate your generator based on the internal registers.

gmqest, 2018-06-25:
Please see attached @jgyates. Let me know if you also need register logs.

[Registers.txt](https://github.com/jgyates/genmon/files/2134729/Registers.txt)

jgyates, 2018-06-25:
A question about your hardware. I have several sets of register dumps that people have submitted via the web interface. Starting with around version 1.9.1 I starting including the info from the monitor page (CPU info, OS info, etc) in the register dump. After examining these I found that some users are showing higher CPU utilization. Those that show higher CPU utilization are all using the Pi Zero, which has a slower CPU. When you were running V1.4.0 were you also on a Pi Zero?

jgyates, 2018-06-25:
I also noticed your packets per second was around 49. Does it stay at this value or does it ever get higher. It should gradually ramp up to around 64 (at least it does on the Pi3)

gmqest, 2018-06-26:
@jgyates all versions of genmon were run on the same hardware but if we do roll back to v1.4 and find that CPU usage is still this high I will report it here for the record. Our recollection is that usage was much lower previously but if your other pi zero users have similar results with this version then that is simply the expected outcome of upgrading.

Our packets per second has always been ~50-52, even in the previous version.

jgyates, 2018-06-26:
ok. that is good to know. I have never really done an analysis on the pi zero. I have one on order to test with. I also am working on a few optimizations for slower CPUs but I still need to do some testing to make sure they work as expected.

jgyates, 2018-06-26:
I checked in an update that may help some. Once you update there is a new setting on the Settings page named "Optimize for slower CPUs". If you enable this then it will modify the wait in the modbus read. I doubt this is the difference between V1.4 and 1.9.x. The difference is more likely due to the addition of features (and threads to support these features) coupled with the lower CPU speed and memory of the Pi Zero. Let me know how it performs.

As I mentioned I have a pi zero on order so I will do some more analysis of how it runs when the pi zero arrives.

gmqest, 2018-06-26:
Many thanks for your efforts on this @jgyates - the new setting in v1.9.29 decreases the python CPU usage by ~26%: in absolute terms, CPU usage for genmon.py drops from ~30% to ~22%.

Configuration: **ENABLED** (green in interface) - "Sync Daylight Savings Time", "Sync Time", " Optimize for slower CPUs", and "Outbound Email Settings"

We will consider the system optimized for now; probably not necessary to roll back version. If you find that your (or others) pi zero tests produce better results, we would look into it further.

jgyates, 2018-06-26:
@gmqest , does this effect packets per second and average transaction time?

gmqest, 2018-06-26:
Yes:
- Packets per second decreases to 35 from low 50s.
- Average transaction time increases to 0.055 seconds from ~0.03 s.

jgyates, 2018-06-26:
ok. Just checking. I figured this would be the trade off. The main thing that this feature does is change the wait time between the modbus request and how long to sleep between attempts to read the respons from the generator. The default will sleep 0.01 seconds, the optimized sleeps 0.03 seconds.

gmqest, 2018-06-26:
Understood - that tradeoff seems very reasonable for the performance improvement.

I will leave it to you to close the comment unless you need anything else from us.

closed 2018-06-26 by jgyates

madisdev, 2018-06-26:
A little late to this, but I am running a PI Zero W with an old version of the code. My CPU for the python process is around 12~. And my packets per second is at 60.

jgyates, 2018-06-26:
"old version" is what specific version number?

Another think to keep in mind when comparing numbers is that the turn around time for the controller to respond can vary depending on how old your generator is, in theory any way. An older nexus and a new evolution may not have the same speed CPU so the faster one would respond quicker. The variable here is this: how long to wait for a response before checking the serial buffer for a packet. If you controller responds quicker you should see higher packets per second and lower CPU utilization.