jgyates/genmon #142: Possible enhancement request: USB wlan adapter name matching

smeisner, 2018-06-25:
I am using a Raspberry PI Zero W with the integrated wifi adapter. The range is pretty bad, so I decided to try an external USB based adapter which works better. The only problem is that the adapter is getting a name that's not "wlan" based. Instead it's "wlx64700219195e". This causes the Monitor page to not show signal status. If I comment out the `if "wlan" in adapter:` line (or modify it to `if "wlx" in adapter:`) in genmonlib/myplatform.py, then the status is shown.

Is there a way to "normalize" the wlan adapter names or allow genmon to query adapters with varying names? Or maybe provide a "prefix string" in the settings?

smeisner, 2018-06-25:
The max value for the signal strength has also changed. For the TP-Link USB adapter I am using, it's 100. So I have the following 2 changes in genmonlib/myplatform.py and it's working well:

Near line 224:
```
if ("wlan" in adapter) or ("wlx" in adapter):
    LinuxInfo = self.MergeDicts(LinuxInfo, self.GetWiFiInfo(adapter))
```

and near line 259:

```
WiFiInfo["WLAN Signal Level"] = ListItems[3].replace(".", "") + " dBm"
if "wlx" in adapter:
    WiFiInfo["WLAN Signal Quality"] = ListItems[2].replace(".", "") + "/100"
else:
    WiFiInfo["WLAN Signal Quality"] = ListItems[2].replace(".", "") + "/70"
WiFiInfo["WLAN Signal Noise"] = ListItems[4].replace(".", "") + " dBm"
```

jgyates, 2018-06-25:
Your system has the new linux feature enabled to show "predictable" network names. You can avoid this by running this command

    sudo raspi-config

Then from the menu select network and disable predictable network names.

This will revert back to the wlan0 naming convention.

I will add the change near line 224, but I need the output of this command to validate the second change:

    cat /proc/net/wireless

smeisner, 2018-06-25:
I will try the change in raspi-config next.

The output of cat /proc/net/wireless
```
pi@generator:~ $ cat /proc/net/wireless
Inter-| sta-|   Quality        |   Discarded packets               | Missed | WE
 face | tus | link level noise |  nwid  crypt   frag  retry   misc | beacon | 22
wlx64700219195e: 0000   61.   48.    0.       0      0      0      0      0        0

```

smeisner, 2018-06-25:
Yup! The raspi-config change set the name to wlan0. I did not realize that would change the name like that.

```
root@generator:~# cat /proc/net/wireless
Inter-| sta-|   Quality        |   Discarded packets               | Missed | WE
 face | tus | link level noise |  nwid  crypt   frag  retry   misc | beacon | 22
 wlan0: 0000   68.   47.    0.       0      0      0      0      0        0

```

jgyates, 2018-06-25:
After looking at this, it appears that different drivers handle signal quality different ways. The TP Link is based from 0 - 100, the built in driver is 0 - 70. The linux docs say the driver developer can make this any value so the appropriate thing to do here is to just remove the " / 70". I will check in an update later today on this.

smeisner, 2018-06-25:
That's too bad. For the user to interpret the number correctly, the "/70" helps. If it's up to the driver, then there should be a call to get the maximum. That would've made more sense.

jgyates, 2018-06-25:
I will put this on the list. I should be able to reg the /70 vs /100 from iwconfig. I just need to make more code changes.

jgyates, 2018-06-25:
I just checked in an update that should fix this. Let me know how it works.

smeisner, 2018-06-25:
Looks good. The wlan device name is still wlan0 (as mentioned above). After I updated, the numbers are:

```
Network Interface Used : wlan0
WLAN Signal Level : 48 dBm
WLAN Signal Quality : 46/100
WLAN Signal Noise : 0 dBm
```

**By the way**...I am running genmon, genserv and gensms as services (started with systemd). My CPU usage never goes over ~22% (used to be ~50 - 60% constantly) and the temps are down ~5 - 10deg. Would you be interested n the systemd unit files?

If so, changes to the installer and updater would be required as well. I can also do a PR if that'd be easier.

jgyates, 2018-06-25:
Sure, that would be great. I have some changes that I am going to be making to the install script so this may be something that I want to consider adding. It may be a few weeks before I get to thoes features so you can just attached the files as a zip to this thread.

That is a good data point on the CPU utilization. I typically see around 2 - 3% on a Pi3, but I think the pi zero is a good bit slower so I expect the CPU to be a little higher. I have another user who is experiencing CPU around 30% and I am trying to determine if that is the PI speed or the software so this data point helps.

Thanks Steve,
Jason

smeisner, 2018-06-26:
Here you go:
[genmon.zip](https://github.com/jgyates/genmon/files/2135404/genmon.zip)

smeisner, 2018-06-26:
Forgot to mention...

My RPi has been running using the systemctl unit files in the zip for a few hours now and the cpu utilization is up at 45% now. Not sure what to make of this.

closed 2018-06-26 by jgyates

smeisner, 2018-06-26:
Good catch! I used your startgenmon.sh to create the unit files. I missed the kill options. Glad you caught it.

I will upgrade and let you know how it changes the CPU usage.