jgyates/genmon #143: Open Weather API Blocked

zekyl314, 2018-06-26:
### Expected Behavior
OWM API calls should be very minimal to OWM.

### Actual Behavior
I received an email from OWM stating the following - Your OpenWeatherMap API key is temporary blocked due to the continual sufficient exceeding of the calls per minute limit by performing 550 requests within a minute but the limit for the Free account is 60 rpm.  I have emailed OWM to ask them for logging details on my calls to see if they have anything more solid as well I could share.

### Steps to Reproduce (including precondition)
I don't know how to reproduce, I looked at the logging and I see exception errors, but wondering if that is because I am blocked?

### Your Environment

- Genereator Model: Honeywell / Generac 20KW
- Genmon Version: 1.9.29

jgyates, 2018-06-26:
What is the frequency of errors in the log? Each log entry in /var/log/genmon.log is time stamped.

The behavior of the weather module is this:

1) Request the weather every 10 min
2) f an error occurs, retry in 1 min.

Questions:
1) Did this ever work for you
2) What is the first error in you log related to weather failures.
3) what is the frequency of the errors.

Also, you have the ability to create new keys once your account is unlocked. If you key was used by another person then you can determine this by creating a new API key and deleting the old key and see if your account is blocked again.

zekyl314, 2018-06-26:
It is every minute from what I can see.

1 - It did work for me in the past, yes.  
2 - I don't know the first error, it seems to have rolled the files because there are so many errors in the log.
```
/var/log# grep OWM genmon.log*| wc -l
1844
```

I have asked OWM to provide details on my key, as I would never expect your application to call it that many times in a minute.

jgyates, 2018-06-26:
The rolled over logs are named genmon.log.1, genmon.log.2, etc

zekyl314, 2018-06-26:
Yes sir, I saw that, my line count above includes all the logs, and the only date in the logs is today 2018/06/26.

closed 2018-06-27 by zekyl314

kalex1114, 2018-06-28:
I actually got the same email as well
pi@genmon:/var/log $ sudo grep OWM genmon.log*| wc -l
1900

Your OpenWeatherMap API key  is temporary blocked due to the continual sufficient exceeding of the calls per minute limit by performing 1125 requests within a minute but the limit for the Free account is 60 rpm.

Got that email twice so far.

zekyl314, 2018-06-28:
I asked OWM for an IP listing, and there were 6 IPs on there, 5 were not mine.  I asked for more detail, but not getting much.  I deleted that key and generated a new one.  They tried to blame me publishing on a site like github or something.

kalex1114, 2018-06-28:
Good idea. I'll generate a new one after my outage is over.  Currently have no utility power

Ali-Guliyev, 2021-03-18:
I got the same email
and i dont know how to solve it
Please someone help me!