jgyates/genmon #144: startgenmon.sh after upgrade

zekyl314, 2018-06-29:
### Expected Behavior
When I upgrade, I wouldn't expect to have to edit startgenmon.sh

### Actual Behavior
When I upgrade to the new version, I have to edit startgenmon.sh and enable SMS and logging again.

### Steps to Reproduce (including precondition)
copy startgenmon.sh to /tmp, upgrade and diff the two startgenmon.sh files to determine what comments need to be removed from the upgraded copy.

jgyates, 2018-06-29:
The program genmon and genserv are the only two programs enabled by default so this is the expected behavior. You can work around this in several ways, but the easiest is to make a copy of startgenmon.sh and put it in your /home/pi directory. The modify the crontab to use the copy.

closed 2018-06-29 by jgyates

zekyl314, 2018-06-29:
Good point about that, as long as that file doesn't change, but that makes good sense.  Thank you sir.