jgyates/genmon #145: pi power supply question

steve-bailey1, 2018-07-03:
Just a question not a problem with the software.  What does everyone run for power supplies for their Pi's?  I am having a problem with under voltage on my pi2 with the usb wifi.  I found pretty nice 120v to 5v converters but wondering what everyone is using for 12v to 5v converters.

bowood, 2018-07-03:
I was trying to use just molex pins into pi3 but kept getting undervoltage. I ended up also adding the USB motorcycle connector listed on the wiki. I think by itself it too might not be enough for a pi 3 but both it and the molex together have ended my undervoltage errors.

proppilot, 2018-07-03:
https://www.amazon.ca/gp/product/B00QTJWRFW/ref=oh_aui_detailpage_o07_s00?ie=UTF8&psc=1

works like a charm.  Connect via fuse to the +/- on the battery

krisasmith, 2018-07-03:
https://www.amazon.com/gp/product/B01MDLUSE7/

steve-bailey1, 2018-07-03:
Proppilot - That is very similar to what I am using but I am still getting the undervoltage condition.

jgyates, 2018-07-05:
@steve-bailey1 

I had a similar problem with a USB wifi adapter. I power via the battery. I ended up running an Ethernet cable to the generator. While the wifi with an external antenna worked, I could was not able to remove the undervoltage issues until I removed the USB wifi adapter. I suppose that you  could try a different WiFi adapter, disable the onboard wifi (to reduce the pi current usage), or go with a PI Zero which, as I understand, has a more defined way to solder an external antenna.

Pi Zero external antenna:
https://hackaday.com/2017/03/07/adding-an-external-antenna-to-the-raspberry-pi-zero-w/

Pi 3 external antenna
https://www.dorkbotpdx.org/blog/wramsdell/external_antenna_modifications_for_the_raspberry_pi_3

madisdev, 2018-07-05:
Again, I don't want anyone think that adding external antenna to PI Zero is a simple task. It requires some micro soldering. And it was only by sheer luck I was able to do it.

closed 2018-07-29 by jgyates

farberm, 2019-06-15:
> Proppilot - That is very similar to what I am using but I am still getting the undervoltage condition.

How did you rin the ethernet cable in the the housing?