jgyates/genmon #147: Propane usage monitor

SethL, 2018-07-10:
This is just a general feature request.  I have a 22kw generac air cooled generator with the evolution controller.  Part of my issue is trying to estimate my propane usage.  I was wondering if it would be possible to calculate the propane usage for each outage. The generator specs regardimg propane usage are published so I would think it might be possible based on run time and %load.  

Thanks and btw I LOVE GENMON!

bowood, 2018-07-10:
I've got the same generator. Last 4 day outage at 600watts (I was away) was about 100gallons of propane. Less than I was expecting. I have a tankutility.com on the propane tank. I'd happily feed in accurate readings if we wanted to jointly build some load curves as a feature.

zekyl314, 2018-07-10:
Hey StehL, I have been working on an idea for my propane tank, using the electrical read out sensor for my tank.  I found one on-line and purchased it.  I was going to see how I could read it to measure my tank volume.  I haven't gotten very far yet though, I need to run a cable from my tank to my generator and connect to my pi, then I can start messing around with reading it.  

https://www.ebay.com/itm/NEW-Rochester-Gauges-Inc-6200-6300-8900-Liquid-Level-Transmitter-/183308492937?hash=item2aae09a089

jgyates, 2018-07-11:
I think @bowood has the right idea since this is going to be dependent on the model / size of engine and fuel type. Having the fuel consumption data along with the kW log contained in /home/pi/genmon/kwlog.txt should enough to create the load curves. I may be able to extrapolate the curves for other models with a little bit if digging in the manuals..

bowood, 2018-07-11:
Hmm.  My outage was on May 15th and my kwlog.txt only goes back to June.  I might have lost it when I was trying to fix my dropped packet situation.  But here is as accurate as I have to get started.  If we had a log to upload readings I'd be happy to do so:
97 gallons of propane (per tankutility.com tank monitor)
92.7 hour outage (per genmon outage event log)
1.08kwh/h energy usage during the 4 days in question (per TED Pro energy monitor)

So this is roughly a gallon per hour when basically idle for 4 days on a 22kw generac.  I recall generac materials saying it burns 2.1 gallons per hour at 1/2 load so I was very happy to have it so low when mostly idle.  I was away and trees from the storm took out internet so I couldn't use genmon to shut it down or check cameras or anything.  Was happy to come back to find minimal damage and lower propane usage than I expected.

jgyates, 2018-07-17:
An initial version of the code to measure fuel consumption is checked in.

* This only support systems that have the power meter functionality (Evolution controller)
* Past version of the code only supported a power log that holds one month of data, this version supports 12 months of data (or if the log reaches the maximum size). Queries will only show data for one month until you have accumulated more data in your log.
* The GUI will show the fuel usage for the past month if "Display Unknown Sensors" is enabled in the settings (This is where I display unsupported or new features that are being tested)
* The initial version only supports Propane load curve and returns the data in gallons. Future versions will may support NG, and possibly other units.
* The initial version only supports Evo Air Cooled units that are 16kW, 20kW and 22kW. To support other models I will need to create the load formula and add it to the code, which requires some manual importing of data.
* The data from the spec sheets that I found only have data points for half and full load to the load curve is linear. Based on the data from a diesel generator (which has 4 data points instead of 2) the load curve is not exactly linear, but it is close.

Fuel usage for different time frames can be displayed by using the following URL in the browser (replace your PI IP address / port)

     http://192.168.1.50:8000/cmd/power_log_json?power_log_json=XXX,fuel

where XXX is the number of minutes to examine

Fuel use for the last month

     http://192.168.1.50:8000/cmd/power_log_json?power_log_json=43800,fuel

Fuel use for the last week

     http://192.168.1.50:8000/cmd/power_log_json?power_log_json=10080,fuel

Let me know if there are any issues or questions.
-Jason

SethL, 2018-07-17:
Bravo sir.  Thank you!  This is EXACTLY what I was wanting.  

 

 http://192.168.1.50:8000/cmd/power_log_json?power_log_json=XXX,fuel

 http://192.168.1.50:8000/cmd/power_log_json?power_log_json=43800,fuel

 http://192.168.1.50:8000/cmd/power_log_json?power_log_json=10080,fuel

bowood, 2018-07-17:
Amazing. Nicely done. I'll give it a try this weekend.

SethL, 2018-07-17:
What would be really cool is for you to be able to input a date of your last fill and a capacity of the tank.  Then have a meter on the status screen for fuel remaining 😊

So in my case with my underground tank was filled 2 weeks ago and has 425 gallons in it.  I input 7/1/18 and 425gal and then as genmon calculates my consumption it deducts it from the 425 number and gives me the estimated remaining fuel in a meter on the status screen

 

Just a thought, I can run the report manually based on the special url example by inputting the specific numbers of days.  Just a nice to have….

 

My status screen:

Fuel Consumption in last 30 days : 1.01 gal

Display something like Estimated Fuel Remaining: 423.99gal

jgyates, 2018-07-17:
You can reset the power log with this command:

https://192.168.1.50:8000/cmd/power_log_clear

or you can delete (or backup) the file kwlog.txt in the genmon directory

This will reset the log, which will allow you to start the calculation on a new tank fill. Then you can issue the command:

 http://192.168.1.50:8000/cmd/power_log_json?power_log_json=0,fuel

which will tell you how much fuel has been used in the entire power log.

Thanks for the input, I will think about how best to implement this in the UI. There are some items I am still determining how to handle, for example what should be maximum size / time allowed for the power log. If the file gets too large it can effect performance.

SethL, 2018-07-17:
Thanks!

 

https://192.168.1.50:8000/cmd/power_log_clear

http://192.168.1.50:8000/cmd/power_log_json?power_log_json=0,fuel

jgyates, 2018-07-18:
I checked in a version that supports more variants. I tested with simulation for all Evolution models that I have registers for and it appears to work on all of them but one (a Honeywell G0070580 11kW EvoAC submitted by @rhombixt ). I cant find a spec sheet for that one. The Air Cooled Evolution units that I have NOT added support for are:

* Three phase models
* 50 Hz models
* 13kw and 15kw models (can't find spec sheets on these)
* 11kw Honeywell (it should work on the generac 11kw but the honewell is a slightly different model)

rhombixt, 2018-07-18:
I use natural gas on mine - is there something you need sent/tested?

jgyates, 2018-07-18:
If you have a spec sheet for your generator that has fuel consumption stats that would be helpful. I don't have NG support added in yet, but I will likely add that.

rhombixt, 2018-07-23:
did that work?

jgyates, 2018-07-23:
Hi @rhombixt 

Yes, this was exactly what I needed. The update that includes this info was checked in a few days ago.

thanks
Jason

jgyates, 2018-08-12:
The info for this feature is located [here](https://github.com/jgyates/genmon/wiki/Appendix-8-Monitoring-Fuel-and-Power-Usage)

closed 2018-08-12 by jgyates

SethL, 2018-08-12:
I just now saw the fuel gauge.  This is awesome!!  Thanks for everything

 

Closed #147.