jgyates/genmon #149: Evolution 2.0

jgyates, 2018-07-15:
This thread is related to discussion on the new Evolution 2.0 controller. If you have have one feel free to post questions or answers here.

jgyates, 2018-07-15:
According to this doc:

http://gens.generac.com/generaccorporate/media/library/content/parts-accessories/wifi-manual-revb.pdf?ext=.pdf

The wifi module uses RS-485. This is a guess, but if pin 1 is still 5V and pin 2 is still ground then pin 5 and 6 must be the RS-485  connection (RS-485 is a two wire bi-directional link).  Pin 7 and 8 may still work with RS-232, but again, this is unverified.

zentec, 2018-07-15:
RS-485 may be solid given its immunity from noise.  Pi daughter boards seem inexpensive. 

I don't think disconnecting the wifi module is going to result in much of a loss.  I captured the ethernet for a day and nearly all the traffic was NTP.  It sent off about 300 bytes of data not long after an exercise and that small amount of data is not surprising given the disappointing information on the web site and mobile app.  There are no open ports listening on the IP used by the controller.

My only caution in disconnecting the wifi would be to probably make sure the wifi is turned off on the controller before pulling the plug to keep it from trying to update the time.

Once I hunt down the connector type, I'll order up some pins (and extractor tool), and see if the RS-232 port is alive with the wifi module running.  

[image]
[image]

ewouk61, 2018-07-17:
the industrial h panel gens have had a 485 for the last couple of years. they use it for communicating to remote annunciaters, with the panel being the master and the annunciaters being the slave, haven't done any poking around on it

ebzimny01, 2018-09-02:
I had a new generac 22kw air cooled installed in June. I know it has Evolution controller, but how do I know if it is v1 or v2?

P.S. Been trying to get genmon working. I'm new to this. Tested loopback successfully all the way at the molex end of my cable. But when hooked up to controller I get nothing. . .

jgyates, 2018-09-02:
The V2 support WiFi. Also the version number of the controller firmware is lower (~1.04 for Evolution 2.0 and ~1.20 for Evolution 1.0).

If you have 1.0 version you can open up a new thread to request help on questions. You may also want to look at this page if you have not already:

https://github.com/jgyates/genmon/wiki/Appendix-6---Serial-Troubleshooting

ebzimny01, 2018-09-03:
I'm all good now. Have v1 and finally got it all working. After checking everything again it was user error on making the cable.

jgyates, 2018-09-22:
It appears the MobileLink uses pins 1,2,5,6 and the GenMon RS-232 connection is thru pins 7,8.

jgyates, 2018-09-22:
On the Evolution 2.0 controller, it appears the MobileLink uses pins 1,2,5,6 and the GenMon RS-232 connection is thru pins 7,8. This is a good sign for compatibility with Genon.

jgyates, 2018-09-24:
Looks like the software does work with Evolution 2.0. Update to 1.10.5 or higher for full support.

closed 2018-09-24 by jgyates

jgyates, 2018-09-30:
Also, it appears you can run the built in mobile link and genmon at the same time (https://github.com/jgyates/genmon/issues/183#issuecomment-425757594)