jgyates/genmon #150: Losing my USB networking...

danorum, 2018-07-28:
I have a raspberry pi zero running genmon using a pluggable OTG ethernet adapter.

The one I put in for my Dad had locked up (at least I thought so), and I just had the issue happen to me. I now think I was mistaken and that the pi zero was running fine, but lost its network adapter.

After rebooting mine, I set up the Rasbperry Pi watchdog thinking that might at least reboot it for me. However, I decided to look closer at the syslog and found the entries at the bottom below.

It looks like I am losing the usb ethernet. Should I do a ping against the router and reboot? I am wondering if this is a usb issue that would be fixed with a reboot or if the power cycle will be required to fix.

Any advice appreciated.

Would this be the next thing to try? 
`#!/bin/bash

#router ip
ping -c4 192.168.0.1  > /dev/null

if [ $? != 0 ]
then
     reboot
fi`

Thanks,
Dan

Jul 28 09:58:01 genmon avahi-daemon[227]: Interface eth0.IPv6 no longer relevant
 for mDNS.
Jul 28 09:58:01 genmon avahi-daemon[227]: Leaving mDNS multicast group on interf
ace eth0.IPv6 with address fe80::7a4c:b36d:5683:cb88.
Jul 28 09:58:01 genmon kernel: [1703749.534379] usb usb1-port1: disabled by hub
(EMI?), re-enabling...
Jul 28 09:58:01 genmon kernel: [1703749.534419] usb 1-1: USB disconnect, device
number 4
Jul 28 09:58:01 genmon kernel: [1703749.535145] asix 1-1:1.0 eth0: unregister 'a
six' usb-20980000.usb-1, ASIX AX88772 USB 2.0 Ethernet
Jul 28 09:58:01 genmon dhcpcd[340]: eth0: carrier lost
Jul 28 09:58:01 genmon avahi-daemon[227]: Interface eth0.IPv4 no longer relevant
 for mDNS.
Jul 28 09:58:01 genmon avahi-daemon[227]: Leaving mDNS multicast group on interf
ace eth0.IPv4 with address 192.168.0.229.
Jul 28 09:58:01 genmon avahi-daemon[227]: Withdrawing address record for fe80::7
a4c:b36d:5683:cb88 on eth0.
Jul 28 09:58:01 genmon avahi-daemon[227]: Withdrawing address record for 192.168
.0.229 on eth0.
Jul 28 09:58:01 genmon kernel: [1703749.764263] Indeed it is in host mode hprt0
= 00001501
Jul 28 09:58:01 genmon kernel: [1703749.994236] Indeed it is in host mode hprt0
= 00001501
Jul 28 09:58:01 genmon dhcpcd[340]: eth0: deleting address fe80::7a4c:b36d:5683:
cb88
Jul 28 09:58:01 genmon dhcpcd[340]: eth0: deleting default route via 192.168.0.1
Jul 28 09:58:01 genmon dhcpcd[340]: eth0: deleting route to 192.168.0.0/24
Jul 28 09:58:01 genmon dhcpcd[340]: eth0: removing interface
Jul 28 09:58:02 genmon kernel: [1703750.354227] Indeed it is in host mode hprt0
= 00001501
Jul 28 09:58:02 genmon kernel: [1703750.704179] Indeed it is in host mode hprt0
= 00001501
Jul 28 09:58:02 genmon kernel: [1703750.984187] Indeed it is in host mode hprt0
= 00001501
Jul 28 09:58:02 genmon kernel: [1703751.264186] usb usb1-port1: Cannot enable. M
aybe the USB cable is bad?
Jul 28 09:58:02 genmon kernel: [1703751.264306] Indeed it is in host mode hprt0
= 00001501
Jul 28 09:58:03 genmon kernel: [1703751.404214] Indeed it is in host mode hprt0
= 00001501
Jul 28 09:58:03 genmon kernel: [1703751.684184] Indeed it is in host mode hprt0
= 00001501
Jul 28 09:58:03 genmon kernel: [1703751.964340] Indeed it is in host mode hprt0
= 00001501
Jul 28 09:58:03 genmon kernel: [1703752.254213] Indeed it is in host mode hprt0
= 00001501
Jul 28 09:58:04 genmon kernel: [1703752.534194] usb usb1-port1: Cannot enable. M
aybe the USB cable is bad?
Jul 28 09:58:04 genmon kernel: [1703752.534270] usb usb1-port1: attempt power cy
cle

jgyates, 2018-07-28:
This is likely a power issue. If the total power used by the pi and USB devices causes the pi to be in an undervoltage condition, then USB devices can be powerd off by the pi firmware. An easy way to resolve this is to use the pi3 which has built in Ethernet instead of a pi zero with a USB ethernet adapter.

alternatively you could look at how you are powering your pi. Based on feedback from others it appears that powering the pi from the molex connector is more likely to cause undervolt errors than powering from the battery.

danorum, 2018-07-28:
I am powering with a 12V to USB from the battery. I just looked and the one I bought says it is 2.1A, so I am going to try the one from amazon that is 3.1A.

Is there somewhere the pi would have logged the power issue?

krisasmith, 2018-07-28:
Try this command.

`/opt/vc/bin/vcgencmd get_throttled`

If it returns: `throttled=0x0` you're good. if you get something else you're pi does not have enough power.

jgyates, 2018-07-28:
This info is also captured on the Monitor page:

[image]

danorum, 2018-07-28:
Both of those look good. I will receive the 3.1A 12v to USB on Monday and may also try adding that network check to run every 5 minutes. Tough to debug something that happens somewhat randomly.

Thanks for the suggestions. Feel free to close this issue and if I figure something out I will still add to the thread.

closed 2018-07-29 by jgyates