jgyates/genmon #151: Request Info : Nexus Liquid Cooled

jgyates, 2018-07-29:
If anyone is using the latest version (V1.9.36 or higher) with a Nexus Liquid Cooled model (QT series) it would be helpful to have some new register dumps (from the Monitor page "Submit Registers to Developers" button). I have some new code that will better auto-detect the model and kw rating that I would like to test and the latest version has the register info needed for this test.

@justapu
@jchenery
@JungleJeep
@JeanB512

JeanB51, 2018-07-30:
Ok I just hit the command. 
Let me know if you received the registers

Jean

jgyates, 2018-07-30:
Yep, I got it. Thanks Jean

JustApu, 2018-07-30:
@jgyates Will do. Had to wait for some parts to come in and now have to find the time to rebuild my monitor. It's still offline after the water got in and killed the first Pi Zero/MAX3232 and the USB serial adapter wasn't giving us reliable connectivity.

jgyates, 2018-08-02:
@JeanB51 

Question: Do you use propane or Natural Gas for fuel?

JeanB51, 2018-08-02:
I have propane. 
Although Genmon seems to think natural gas. 

Jean

jgyates, 2018-08-02:
ok. that makes sense. I think I found the register that controls that setting. You can change this from the settings page and it should keep the setting. If not the next update should correct that. Before the software made its best guess, going forward on liquid cooled models the software should be better at this.

closed 2018-08-06 by jgyates