jgyates/genmon #152: Quiet Mode Will Not Change

restauffer, 2018-07-30:
### Expected Behavior

On Maintenance page, when changing Quiet Mode (On or Off) after pushing "Set Exercise Time"  and answering OK to the "Set Exercise time to..." it is expected that the mode will change to the one chosen.

### Actual Behavior

On Maintenance page, when changing Quiet Mode (On or Off) after pushing "Set Exercise Time"  and answering OK to the "Set Exercise time to..." the Quiet mode does not change.

### Steps to Reproduce (including precondition)

  1 Go to Maintenance page.
  2. Toggle  "Quiet Mode" to the opposite of the current setting. 
  3. Press "OK"
  4. Go to any other page and then return to Maintenance page.
 
Quiet Mode will still be in original state.

It is possible to change the state of Quiet Mode from the generator control panel so the function is available and working on the generator

### Screenshot or Picutres relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model:  G0070421, Controller Evolution

- Generator Registers: 
Registers : 
    Num Regs : 244
    Not Changed : 186020
    Changed : 4366
    Total Changed : 0.02

    Base Registers : 
        0239 : 0000
        0238 : 0000
        0039 : 0000
        0038 : 0018
        0034 : 0009
        0037 : 2c88
        0036 : 0000
        0237 : 0000
        0033 : 0960
        0032 : 4000
        05f7 : 05d4
        05f6 : 05ce
        05f5 : 0000
        05f4 : 0000
        05f3 : 00c8
        05f2 : 4683
        05f1 : 0000
        023e : 0000
        023d : 0000
        05ed : 0064
        0235 : 0000
        023b : 0000
        023a : 0000
        003a : 0000
        003c : 0000
        003b : 0000
        05fa : 0000
        0059 : 0000
        0058 : 0000
        0053 : 0000
        0052 : 0000
        0057 : 0000
        0056 : 0000
        0055 : 0000
        0054 : 0000
        0208 : 03f4
        0209 : 0000
        005c : 0000
        005a : 0000
        005f : 0000
        005e : 0000
        005d : 0000
        020f : 0000
        020d : 0000
        020e : 0000
        020b : 0000
        020c : 0000
        020a : 0000
        000d : 0000
        000e : 1530
        000f : 071d
        024a : 0000
        000a : 0089
        000b : 0000
        000c : 0000
        0012 : 0000
        0242 : 0000
        0243 : 0000
        0006 : 0301
        0007 : 0000
        0246 : 0000
        0001 : 00000000
        0244 : 0000
        0245 : 0000
        0248 : 0000
        0249 : 0000
        0008 : 0000
        0009 : 00f5
        0005 : 1500
        002a : 6a77
        05ee : 00ce
        0258 : 0000
        0011 : 009c
        0010 : 0012
        0019 : 0011
        002b : 0005
        002c : 1500
        0241 : 0000
        002f : 0000
        002d : 0000
        002e : 0003
        0000 : 0009
        0247 : 0000
        0022 : 4683
        0020 : 0000
        0021 : 0000
        001f : 4aca
        001e : 018f
        001d : 0000
        001c : 0000
        001b : 47f0
        001a : 00c7
        025a : 0000
        01f4 : 333030323236323435
 
- Genmon Version: V1.9.36

jgyates, 2018-07-30:
@restauffer ,

A few of questions:

1) Are you showing quiet mode on or off in the web interface (your registers indicate that it is on)?
2) If you set quiet mode from the web interface and then refresh your browser do you see that it has changed? This will let me know if it is an issue with the java code (browser UI) or the python code.

restauffer, 2018-07-30:
1. The web interface is showing "on". It always shows what the register shows. If I change the mode from the generator panel, the register changes, and then the page reflects what the register says correct.

2. No. If I set it and refresh the browser it goes back to whatever it was set for before (and the register has not changed.)

Thanks for your help.

jgyates, 2018-07-30:
We have a few things to try to see what this issue could be:

1) try setting the quiet mode from the command line. For this we use the python program ClientInterface.py. You can run this from another system or from the pi console. [Here](https://github.com/jgyates/genmon/wiki/1----Software-Overview#clientinterfacepy) is the documentation for running ClientInterface.py

Once you run ClientInterface.py then use these commands:

     generator: setquiet=off 

then view the exercise state:

     generator: maint

change it again:

     generator: setquiet=on

then view the state:

     generator: maint

If this works then it is someting with the web interface code or browser. If it does not work then we need to look at your genmon.log log file. It is located on the pi in /var/log/genmon.log. You can attached this file to the thread or you can send me your log files with this command in your browser:

     http://MYIPADDRESS:8000/cmd/sendlogfiles

Change MYIPADDRESS to the ip address of your pi and change the port (8000) if you are not using the default.

restauffer, 2018-07-30:
ClientInterface.py changes did not change the register or the display in the web interface. 

cd /etc/genmon.conf

> I

 followed your directions and sent the log files. 

The large number of start and stops was due to working to get SSL/TLS Certs working (I succeeded finally) While on that topic, the comment in genmon.conf:

":For information on how to create your own
# certificate please see this site:
#     https://workaround.org/ispmail/jessie/create-certificate"

is no longer valid. The company referenced no longer gives free certs.

jgyates, 2018-07-30:
I don't see any thing obvious in your logs.

* Are you seeing any communications errors reported on the Monitor page?
* Have you edited /etc/genmon.conf manually and if so what settings have you changed?
*When you issued the "setquiet=yes" did you see the message ""Set Quiet Mode Command sent" returned?
* Have you tried a fresh reboot? I noticed in your logs there were some older serial errors and one error related to the genmon already being loaded. A reboot would clear up any issues related to multiple instances of the software loaded.
* Is there any other things that are not working or is it just quiet mode?

I have never see this issue before and I can not duplicate this so I am at a bit of a loss as to why this is occurring on your system.

restauffer, 2018-07-30:
Are you seeing any communications errors reported on the Monitor page? _No_
Have you edited /etc/genmon.conf manually and if so what settings have you changed? 

-   _uselegacysetexercise_ 

   

    _- _I  tried changing this to "true" to troubleshoot this problem. When it did not fix the problem I put it back to "False"_

_- enhancedexercise_

  _- set to True but the problem already existed when it was set to false. (Not sure if I manually edited this. Probably just did it from the web interface)_

- _useselfsignedcert  =_

- _keyfile  =_

_- certfile  =_  

  _-  I played with the cert settings quite a bit as I locked myself out  from the web interface several several times.__ 
  
*When you issued the "setquiet=yes" did you see the message ""Set Quiet Mode Command sent" returned? _Yes_

Have you tried a fresh reboot? I noticed in your logs there were some older serial errors and one error related to the genmon already being loaded. A reboot would clear up any issues related to multiple instances of the software loaded. 

-  _I have rebooted many times in the last days while working to get SSL certs working. The errors with Genmon already loaded occur when I do a startgenmon.py stop and start (Sorry if command name is wrong.  I am at work and can't SSH into the Pi_

Is there any other things that are not working or is it just quiet mode? Just Quiet mode. 

_- Everything  else works great._

I have never see this issue before and I can not duplicate this so I am at a bit of a loss as to why this is occurring on your system.

_If you suspect some sort of corruption in the genmon.conf file, I could replace it with a fresh copy, reconfig the app, and see what happens._

jgyates, 2018-07-30:
Replacing the genmon.conf file is a good idea.

the thing that is odd is that when you issue the command to set quiet mode we don't see any errors, which tells me that the modbus command is being sent to the controller and we get an acknowledge back from the controller. If this did not happend we would see an error logged to mymodbus.log, but we do not.

restauffer, 2018-07-30:
I will try replacing genmon.conf this evening.  Thanks for your help.

restauffer, 2018-07-31:
Good Evening,

I replaced /etc/genmon.conf with a untouched version from the genmon folder.  It did not make a difference.  I did note one thing. I stopped genmon, copied the file, and restarted genmon, inadvertenly as a foreground.  Since it was foreground I was able to watch the output.  Here is the output when I told it to change quiet mode from on to off:

127.0.0.1 - - [30/Jul/2018 20:53:55] "GET /cmd/setquiet?setquiet=Off HTTP/1.1" 200 -
127.0.0.1 - - [30/Jul/2018 20:53:55] "GET /cmd/setexercise?setexercise=Wednesday%2C21%3A00%2CWeekly HTTP/1.1" 200 -

Here it is when I told it to set it to on (it is already on):

27.0.0.1 - - [30/Jul/2018 20:57:15] "GET /cmd/setexercise?setexercise=Wednesday%2C21%3A00%2CWeekly HTTP/1.1" 200 -
127.0.0.1 - - [30/Jul/2018 20:57:15] "GET /cmd/setquiet?setquiet=On HTTP/1.1" 200 -

I don't know much about the commands but must say they look correct to me.Thought this might help you.

Any other ideas?  If we don't solve this it is not the end of the world. Setting quiet mode is not something I would often change and I still can do it from the generator control pad. It is just the mystery of why it does not work that keeps me looking at it.

Thanks for listening.

jgyates, 2018-07-31:
This output is correct. This is the web server displaying the commands it receives from the web app running in the browser. The test we ran with ClientInterface.py bypassed the web app and web server and went straight to genmon.py to issue the command. About the only think I can think of that would cause this (based on the info I have) would be that you have a bad block of memory or ROM in your controller or on the pi (probably more likely the controller since we are not seeing any other issues or errors in the log files), but that is just a guess.

One other thing you could try would be do download the repository again with this command into another directory like /home/pi/temp  :

     git clone https://github.com/jgyates/genmon.git

Then stop the copy running in /home/pi/genmon, then start the version in /home/pi/temp/genmon

This would rule out any issue with the files being corrupt. This is likely a long shot, but it is worth a try.

restauffer, 2018-07-31:
Did as you said and downloaded a whole new copy in another directory. Stopped old, started new...no change.

Bad memory is an interesting theory but if it is controller memory I would think it wouldn' work from the generator control panel either. Unless that panel is more directly interfaced and doesn't use modbus and its associated memory.  I need to look at the service manual and see if there is a way to reboot the generator controller.

Thanks for tyring to find the cause.

closed 2018-08-05 by jgyates

danorum, 2018-09-01:
I know this is a closed issue, but I needed to change my generator to not run in quiet mode (because of an issue with it not running in quiet mode during exercise). I have the same symptoms as above in that I can change it on the control panel, but not change it from genmon's UI. Once I change it on the control panel, the correct setting is displayed on gemon.  Mine is  a G0058750 with Nexus controller, air cooled.

jgyates, 2018-09-01:
@restauffer , Thanks for the data point!

danorum, 2018-09-01:
I have been playing around with this and it appears that whenever I set the exercise time via genmon, it always sets quiet mode on. If I change it on the generator, I can see it as off in genmon, but if I change the time again (and leave quiet mode off), it is set back to on.

jgyates, 2018-09-01:
@danorum 

The software will read the setting for quite mode (and exercise time) when you first show the Maintenance page in the web UI. If you change the setting on the generator while the Maintenance page is up and the change the exercise time, then you will see the behavior you are describing. 

Can you duplicate this issue in this way:

1) Check the state of quiet mode with with the web interface
2) Close all instances of the web interface
3) change quite mode on the generator
4) Check the state of quite mode in the web interface
5) Set the exercise time in the web interface 
6) Is quiet mode set as expected?

jgyates, 2019-04-30:
@restauffer 

I got another data point on this issue. See this thread:

https://github.com/jgyates/genmon/issues/255

Short story, I removed the ability to modify quiet mode from the web interface for Evo Air Cooled and all Nexus units.

restauffer, 2019-05-01:
Thanks for letting me know.

nkm8, 2021-05-16:
@jgyates I'm not sure if this should go in a new issue, but I have a new data point. I have a Nexus air cooled generator and observed the following:

1. Opened Genmon to change exercise time and saw quiet mode was currently set to off
2. Set exercise time, saw 'undefined' listed in the confirmation dialog
3. Exercise time was changed successfully, but quiet mode was now enabled
4. I confirmed this on the generator control panel
5. Set quiet mode back to off on the control panel
6. Checked Genmon to confirm quiet mode showed as off

I currently have no way to set quiet mode from Genmon, but it looks like setting the exercise time also forces quiet mode on. 

[image]
[image]
[image]

jgyates, 2021-05-16:
@nkm8 

Thanks for the data point. I checked in a change that should prevent the "undefined" in the confirmation dialog.  When I initially figured out how to set quiet mode (and setting the exercise time) I found that the Nexus controller operated differently than the Evolution controller. As a result they set the exercise time and date differently.  Also, setting quiet mode is not supported and disabled in the user interface for Nexus controllers since I do not know how to do that on Nexus controllers. Have said all of this, the behavior you are seeing is likely the result of the controller firmware resetting quiet mode every time the exercise time / date is set via modbus. I can add this to the known issues list so others will be aware of this. Let me know if you have any other questions.

jgyates, 2021-05-16:
This has been added to known issues (see number 14)

https://github.com/jgyates/genmon/wiki/Appendix-D-Known-Issues

nkm8, 2021-05-17:
Thanks @jgyates the detailed explanation makes perfect sense 👍