jgyates/genmon #154: Feature Request: BCC email

USAFPride, 2018-08-04:
It would be a nice feature to have the option to send messages to multiple email addresses via BCC.  I'm thinking the best place to make the option is a checkbox on the Notifications page and whether it should be a universal (all emails, easiest) or a separate choice for each address.  Currently I'm just changing to To line in `mymail.py` (line 258) to `msg['Bcc']`

jgyates, 2018-08-06:
Hi @USAFPride 

Let me think on this. You can already send to multiple addresses, it just does not use BCC. Is there a reason why you want to use BCC.

USAFPride, 2018-08-06:
@jgyates, I use email to send SMS notifications (most providers have an email option) and multiple addresses show up via SMS.  In addition, the most pressing version, I monitor my parents genset and it eliminates the "reply all" email of "did you see this?"

closed 2018-08-15 by jgyates