jgyates/genmon #155: Informational - integration with Home Assistant

lmamakos, 2018-08-05:
HI!  I'm a new genmon user, only a couple weeks now.  I'm using a Pi Zero W with a USB attached WiFi and external antenna.  Maybe some photos of the build later.

I recently did some work to get the open-source Home Automation software I use, called [Home Assistant](https://www.home-assistant.io/) to integrate with genmon.  I used the JSON interface in genserv.py as the means to get the good stuff out, and then some fancy sensor configuration in Home Assistant to parse out the JSON values I was interested in.  DIdn't have write one line of code or hack on anything in genmon to solve my problem.  See: https://community.home-assistant.io/t/monitor-your-generac-generator-with-home-assistant/62701 on the Home Assistant community forum where I put an example of this out there for others to use/learn from.

If you're looking to do more complex automation and integrate monitoring and control of lots of other types of things, you should seriously look at Home Assistant for that purpose.  It integrates with all sorts of devices/products and is easy to do your own thing.  And there's a bunch of UI options available, both as part of the basic distribution, to third-party contributions that target, e.g., tablet displays to hang on the wall.   Cool stuff, and even my wife likes having it in the house, controlling the lighting (mostly Z-Wave, some Sonoff Wi-Fi switches, some X-10 powerline modules connected off my alarm system, which it also talks to and can see zone states, etc.)

Anyway, back to genmon..

The only suggestion I'd have from going through this is to have some additional JSON fields with strictly numeric values, rather than formatted as "242 V", where I need to strip off the units before doing a conversion to a numeric value.  Happily, Home Assistant has a jinja2 template engine embedded in it, with some facilities for doing just that sort of hacking around.  E.g.,

`   value_template: '{{ states.sensor.genmon_monitor_json.attributes["Monitor"]["Platform Stats"]["CPU Temperature"] | regex_replace(" F","") | float | round(1) }}'`

Thanks for building this really useful software and sharing it.

proppilot, 2018-08-05:
This is very interesting to me.  My home automation is done via a Vera Plus and I have been noodling over ways to bring some monitor/control into that platform.  I had considered using a Z-wave I/O module and using the GPIO in/out programs in the repository to provide some basic start/stop and status .  I recently (like Thursday) even became aware of the Home Assistant project/platform and plan to play around with it as it integrates fairly nicely with Vera so I like what you have done here.  If I was aware of HA prior to my adoption a few weeks ago, I probably would have gone that route as I like playing with this stuff.  Thanks for posting the information.  I have been a Genmon user for just under a year and have really seen the contributions/additions and development accelerate quite rapidly in that time as it is gaining popularity.

lmamakos, 2018-08-05:
I happened across Home Assistant almost 2 years ago.  For probably 10 years or more, I've had a couple of dozen temperature sensors, and then later, power use sensors in my home.  And some X-10 PLC lighting (well, this goes back a few years! :-)    I hacked up all sorts of scripts to shove the interesting data into a database and then some ugly RRD-based graphs to present it.

I had built a very simple macro capability so that when this happened over here, turn that light on over there.  When I went looking for a better solution to integrate my sensors, I found Home Assistant and was overwhelmed and astonished at how capable and well-designed it is!  It nicely complemented the work I had done, and it was very easy to integrate with.  I really liked have a "neutral" component in the middle of things, rather than having to buy into one platform or another.

To illustrate - my Home Assistant has an experimental connection to my Elk alarm system.  I've been collaborating/supporting someone that's been working on this, and it will eventually get contributed into HA when it's mature enough.  In any case, I have access to all of the "zones" in the alarm system - wired sensors on doors and windows, and PIR motion detectors.  I've also got some Wi-Fi Sonoff switch modules, so cheap at less than $10 each!  And I have some Z-Wave switches for some lighting, as well as just a few X-10 PIR motion floodlights that are driven from the alarm system as well.

All these things get exposed as generic "entities" of various classes in HA, either lights or switches, or sensors.  So I have an "automation" that when the door off the kitchen opens when the sun is less than 15 degrees above the horizon, turn on the floor light in the enclosed porch.  What's remarkable is that the door opening is sensed by the alarm system, and the floor lamp is controlled by a Sonoff switch that's listening to an MQTT topic.  HA lets me effortlessly span all these different technologies that some of these "hub" systems might limit you to.

I've another automation that when either the motion detector on the front porch senses motion, or the Z-Wave lock on the front door is unlocked, to turn on the X-10 floodlights, turn on the Z-Wave porch light, etc.  Or a similar automation when HA senses "presence" because it can poll my wireless network and notice the MAC address of my phone connect when I pull up in front of the house in my car.

This is wonderful stuff!  And there's a thriving, active community of users and contributors.  See also the community web site at https://community.home-assistant.io/ to get sense of the activity going on.

If you're contemplating doing something clever based on genmon, rather than hacking up one-off scripts, it might be worth investing in a general purpose solution with HA.  It's easy to get started with a Raspberry Pi and a docker-based deployment for easy upgrades, call HASS.IO

jgyates, 2018-08-06:
@lmamakos 

This is great stuff. I will look into making a more exportable JSON formatted data as you describe. It sounds pretty straightforward, I just need to find the best way work it into the code.

baboss80, 2018-08-06:
I use SmartThings for home automation and would like to integrate my Pi/Genmon into it, not so much for control but for status reporting of the generator in the SmartThings app. I've just started looking into it and am not much of a developer but can muck my way through other's examples.

proppilot, 2018-08-06:
baboss80 a simple solution for you (and I am even considering) is the Fortrezz Mimo2+ z-wave module.  It has two input and two output channels.  You could hang it off your ST (or my Vera) and using the GPIO routine on this project, input two states such as READY and RUNNING and you could input the START/TRANSFER and STOP commands so you could initiate a start/transfer and stop from your ST.

Three would be perfect as I would go READY/RUNNING as inputs and START/START TRANSFER/STOP as outputs.  I am still noodling around for a z-wave i/o module with a few more inputs and am going to look seriously at the HA project as a means to do it as it looks like Vera and HA can chat.  The more I think about it I could probably use my Genmon Pi to host HA as well just for this specific functionality.

I anticipate some more late nights playing..... ;-)

USAFPride, 2018-08-09:
@lmamakos, I've managed to install your package into HA. While I'm a complete NOOB with HA and you are currently at an Alpha stage with your packages, I'm wondering if the best approach is to have Genmon have the ability to send the data to MQTT server?

lmamakos, 2018-08-10:
@USAFPride , I think it would be interesting to be able to push data via MQTT and get a more prompt response vs. polling for the current state.    I think that would be most appropriate if there was some mechanism inside of genmon that had some way to push updates out to clients.  Perhaps there is; I just am not familiar enough with the structure of genmon and the assorted libraries.

I started down the path of thinking about making MQTT-based sensors in Home Assistant, and I figure dI'd hack up some sort of daemon I could run to push out the MQTT messages.  I took a look at `gengpio.py` for instance, to see if I could adapt that structure, but ultimately it still polls for state changes.  Ideally, there'd be some sort of pubsub pattern inside of the main `genmon.py` script that polls the generator at whatever frequency that could be a way to hooking for push updates.

In any case,  I wasn't really sure what needed to be done and it seemed pretty easy to get started by polling remotely via the web interface.  This insulated me from any custom code on the RPi running `genmon`, as well as possibly any custom component in Home Assistant as well.  Ultimately, I wish I polled more frequently and I may try that using the current approach.

jgyates, 2018-08-10:
also, the /genmonlib/mynotif.py module uses  callbacks that could be used for a push type situation. At present it only notifies on state change (running, exercise, stopped, etc)

lmamakos, 2018-08-11:
So just for clarity, when one of the external application polls `genmon` (e.g, like the GPIO script, etc.), that's just retrieving the data out of state inside of `genmon`?  My concern was there being some connection between the individual polls, and causing more traffic on the modbus channel to the generator.

If my understanding is correct, then its probably reasonable to bang away with polls every second to detect changing state and then act on it.  While perhaps not quite as "elegant", the practical aspect of it is that specific activity is nicely isolated from the core of `genmon`.

jgyates, 2018-08-11:
@lmamakos ,

Correct. Calling into genmon as fast as the CPU / network will support will not cause additional modbus traffic. The modbus polling is in a separate thread so the only thing that will affect modbus traffic is if the CPU utilization is high and this would cause the modbus reads to be slowed. You can monitor the modbus packets per second and average transaction time on the Monitor page from the web UI. 

To give you an idea of throughput, if you have the web interface up the javascript calls into genserv then to genmon about once every 5 seconds to update the page and every 3 seconds to determine the base status. You could poll faster than this but you will eventually reach a point of diminishing returns (around 0.5 seconds would be my guess) since it takes time to read all of the registers.

lmamakos, 2018-08-11:
@jgyates ,
Thanks, that's good to know.  

I have some desire to spin a small board to piggyback on the Raspberry Pi GPIO connector, and include 3 or 4 thermocouple interfaces and an RS-232/TTL converter for the modbus connection as well.  I'd like to be able to monitor some internal temperatures, like the air coming in, the air exhaust, perhaps the oil temperature measured at the oil cooler.  As that will need a bit of software to manage, that might also be an opportunity to also include polling the genmon data, and publishing it all via MQTT.

That's nothing but a crazy idea at this point, so something for the back burner.  There's already some open-source work in this area that would be a great starting point: https://github.com/mikelawrence/RPi-pHat-Thermocouple

USAFPride, 2018-08-11:
@jgyates 

> I would think the cleanest method would be to have an external program poll genmon every 1- 2 seconds, change the data as needed, and push it to MQTT.

This makes sense.  A simple cron/python script is simple enough to be added. Is there currently a way to get all the "data" at once vs. issuing separate calls (eg. status, maint, outage)?  If not, could that be implemented?

jgyates, 2018-08-11:
I have something working now that imports any changed values from genmon (except the full alarm, start/stop, maintenance log) into a mqtt broker. I just need to put some finishing touches on it and do some more test. I should be able to check it in in a few day.

USAFPride, 2018-08-11:
@jgyates Awesome.  No rush on my part as I won't be able to test anything until Friday night at the earliest.

jgyates, 2018-08-12:
I have added a new program named genmqtt.py to the repository to support importing genmon data into a MQTT server. The program is described [here](https://github.com/jgyates/genmon/wiki/1----Software-Overview#genmqttpy-optional)

Let me know if there are any issues. I tested this with the MQTT broker from http://mosquitto.org/ and node-red.

proppilot, 2018-08-12:
Wow. this is getting wings.  There is a Plugin for the VERA to read MQTT data.  Would the RPi with Genmon running need a MQTT broker (like Mosquitto) running ?  This is new territory for me but super interesting and something I want (need !!) to implement between my Genmon Rpi and my Vera !

jgyates, 2018-08-12:
I am not too familiar with VERA, but a quick search turned up this site (https://github.com/jonferreira/vera-mqtt). This appears to be a MQTT client that can push data to a MQTT broker. This is essentially what the genmqtt.py program is doing (pushing to a MQTT server.  Some home automation programs have a MQTT broker underneath the hood or have the ability to read and post to one as a way of exchanging data. Home Assistant and OpenHab.org are two that I know of will support this but I am sure there are others. I use another pi to monitor my home energy usage from https://openenergymonitor.org/. This openenergy monitor has a MQTT backend and it also comes with node-red and openhab built in.

USAFPride, 2018-08-12:
@proppilot, a MQTT broker would be needed somewhere in your system.  I'd recommend not using the Genmon rPi as those tend to be located in less than ideal locations. A quick googlefu shows that VERA does not have a built in broker, so you will need another device, but it can read MQTT from a broker.  Hopefully you have another rPi in your house somewhere. 

@jgyates, thanks for the addition.  I'll try to test today, but don't know how in depth I can be for a week.

USAFPride, 2018-08-12:
the `clientid` field needs to be implemented. If not declared, it is random each time:
`Received PUBLISH from e8f608a7-b3a7-4875-bed3-10088b03b830`
`Sending PUBLISH to 52304f79-4ec0-43c4-91db-d4a9c4ee25ff`

jgyates, 2018-08-12:
I checked in a new version that has a static client id of "genmon".

USAFPride, 2018-08-12:
Much better.  Thanks!!!

lmamakos, 2018-08-13:
I'll need to use a `value_template` in the MQTT sensor to parse out the number so that my sensor is a _number_ and not a text string.  I feed all that sensor data into InfluxDB and Grafana so I can make pretty graphs and stuff..

I'm about to leave town on a business trip of a few days, so to avoid wrath from my wife about home automation stuff going weird, I'm in a "**Change Freeze**" until I return, when I'll be able to give this a try..

Thanks @jgyates for building this!

kalex1114, 2018-08-30:
Wow nice. I'm in the process of migrating from smartthings to home assistant

lmamakos, 2018-10-15:
If it might be helpful to others, here's most of my generator-specific configuration in Home Assistant.  I use  [Home Assistant "packages"](https://www.home-assistant.io/docs/configuration/packages/) to modularize my configuration, and this is all in one file in my system.   I have something like:

```yaml
homeassistant:
  packages: !include_dir_named packages
```
In my main configuration.yaml file, with this configuration as one file in the packages directory.

My MQTT topic scheme might not match exactly what others are using, but you'll get the idea.

[generator.yaml](https://github.com/jgyates/genmon/files/2477024/generator.yaml.txt)

jgyates, 2018-10-15:
@lmamakos 

Would this output meet your needs better?

    /generator/Status/Engine/Output Voltage/value/240
    /generator/Status/Engine/Output Voltage/type/"int"
    /generator/Status/Engine/Output Voltage/units/"V"

This would only apply to entries that return a float or integer. String values would remain the same.

I was considering making this and option in genmqtt.conf so you could choose the old way and the new way.

darkstar2002, 2018-10-15:
> @lmamakos
>
> Would this output meet your needs better?
>
> ```
> /generator/Status/Engine/Output Voltage/value/240
> /generator/Status/Engine/Output Voltage/type/"int"
> /generator/Status/Engine/Output [...]

@jgyates I think that would be great.

@lmamakos Unfortunately I wont be running home assistant but what you have done so far is enough to get me going on a mqtt to influxdb program.  I found a simple python one.  I just need to make some modifications to work with this setup.

lmamakos, 2018-10-15:
> @lmamakos
>
> Would this output meet your needs better?
>
> ```
> /generator/Status/Engine/Output Voltage/value/240
> /generator/Status/Engine/Output Voltage/type/"int"
> /generator/Status/Engine/Output [...]

I think that would simply matters.  I'm not sure I'd need to use the type or units.  Another option is to just combine those into one message with a JSON payload like:

MQTT topic: `generator/Status/Engine/Output Voltage`      with payload:    `{ "value": 240, "type": "int", "units": "V"}`

Note that here, the value in the JSON dictionary is an integer `240`, and not a string `"240"`  I think it might be more efficient to combine those into one message, rather than firing off a bunch of individual MQTT messages to the broker.  Of course, that's probably unnecessary optimization, since the client will just subscribe to the relevant `.../value` topic.

jgyates, 2018-10-15:
The JSON approach may be best since in my testing I have found that if you send an integer or float to the publish function it will convert it to a string. The docs back this up. Using a JSON string with the value an int or float would likely get around this issue.

jgyates, 2018-10-16:
I checked in an update that will return JSON string for numerical data based on the new option "JSON for Numerics" on the Add-On Page. Note that this only applies to the data on the Status page. The outage page numerical data is duplicated (except the log) on the status page. If you need the JSON strings for other data let me know and I can look at working that in.

darkstar2002, 2018-10-30:
Here is what I have for now.
[image]

kalex1114, 2018-10-30:
> Here is what I have for now.
> [image]

Nice. can you provide details on this?  I'm running grafana for my boiler already, would be nice to add this in there.

liltux, 2018-11-03:
is there a list on the mqtt broker that shows all of the topics being published to from genmon?

liltux, 2018-11-04:
so I have configured mqtt for a raspberry pi 2.  I can test it locally and works.  tried to set genmon pointing it to the mqtt broker I setup.  could not see anything being published.  Next, I configured an MQTT broker on the genmon raspberry pi.  pointed the genmon conf. to the newly configured broker and pointed my other raspberry pi 2 to that broker as well.  I could test on command line and see data going between the broker and my pi 2, but nothing from genmon.  I am wondering if there is a configuration step I may be missing with the broker.

jgyates, 2018-11-04:
@liltux 

For what it is worth, the MQTT broker I used for testing was already installed on the image I used from openenergymonitor.org (mosquitto). THis is running on a 2nd pi3.  I used red-node (also running on the openenergymonotry pi) to validate the topics submissions were received.

Are you getting any error messages in /var/log/genmqtt.log?

on my system the /etc/mosquitto/mosquitto.conf file looks like this:

    pid_file /var/run/mosquitto.pid

    persistence false
    persistence_location /var/lib/mosquitto/

    log_dest file /var/log/mosquitto/mosquitto.log

    include_dir /etc/mosquitto/conf.d

    allow_anonymous false
    password_file /etc/mosquitto/passwd

The directory /etc/mosquitto/conf.d just as a README file with this : 

    Any files placed in this directory that have a .conf ending will be loaded as 
    config files by the broker. Use this to make your local config.

liltux, 2018-11-05:
@jgyates , ok I will try using the main .conf file for my configuration settings.  I was able to subscribe and publish to the broker via command line to test broker.  And that was working.  But when I ran :
"mosquitto_sub -h (ip address) -t generator/# -u "username" -P "password""
Nothing displayed.  not being sure the topic subscription.  I see you used node-red to breakout the topics.  I will try that next.  I am trying to see if I can use a mobil app, through a cloud server, to sub data from broker that is holding data from genmon.  I would assume it is capable.  Though, most mqtt apps I am coming across are running websockets protocol, is genmon?  @anyone,  Can a broker run multiple protocols, has anyone tried or accomplished this?

liltux, 2018-12-04:
@jgyates , I know it has been a few days but now I am getting MQTT working.  I went through and found some of the software librarys were not loaded correctly.  I had to uninstall pyopenssl, install again.  When it finally installed successfully I was able to install chump and paho-mqtt successfully.  Now I am publishing to the broker, just trying to pinpoint the topics I want to subscribe to and what I can do with that.  Just wanted to let you know my problems were related to the librarys not being loaded correctly.

liltux, 2018-12-08:
@jgyates have you thought about an option for adding ssl/tls connection?  I am wanting to try and tweak your code, but my luck I will just break it.  Just curious.

jgyates, 2018-12-08:
Let me take a look, it should be easy enough to add.

jgyates, 2018-12-08:
are you using a single Certificate Authority file (i.e. ca.crt)? or do you have a a key file and cert file? or both (three files)?

liltux, 2018-12-08:
All three files are on my broker, yes.

jgyates, 2018-12-08:
According to this you need the three files on the broker but on the CA on the client.

http://www.steves-internet-guide.com/mosquitto-tls/

Just trying to get the options right...

liltux, 2018-12-08:
That's kind of what I gathered, the Certificate Authority, but wasn't quit sure exactly where to plug that into genmqtt.py.

liltux, 2018-12-08:
Of you would like I could add a broker user for you to test.

jgyates, 2018-12-08:
that would help. I don't have a cert at the moment.

jgyates, 2018-12-08:
I am going to force the port to 8883 (default SSL) if a cert is used and ignore the port option unless you think it would be better to let the user decide this.

jgyates, 2018-12-08:
I checked in a change that is ready to test. It adds two new options (available on the Add Ons page)

Full path to CA cert file
TLS version (default is 1)

Let me know how this goes. If there is an error it should log to /var/log/genmqtt.log

liltux, 2018-12-08:
@jgyates I tried without then with ca file and no connection.  Submitted logfiles, let me know if you recieved.

jgyates, 2018-12-08:
Got it, looks like the TLS version parameter is an int not a float. I checked in an update. Try it now. I did not bump the version info but clicking update will still work.

jgyates, 2018-12-08:
i.e version is still 1.11.31

liltux, 2018-12-08:
Still not connecting.  sent new logfiles.

jgyates, 2018-12-08:
I got your new logs. I checked in a new version. Let me know if this works.

liltux, 2018-12-09:
Let me look into the CA file.  I may need to fix my file. I will let you know.

jgyates, 2018-12-09:
The call to setup SSL/TLS looks like this:

tls_set(ca_certs=None, certfile=None, keyfile=None, cert_reqs=ssl.CERT_REQUIRED, tls_version=ssl.PROTOCOL_TLS, ciphers=None)

I am not 100% sure about the cert_reqs parameter. It should either be CERT_NONE, CERT_OPTIONAL, or CERT_REQUIRED (default is CERT_REQUIRED), depending on the server config. Here is ther SSL doc for the three options.

https://docs.python.org/3/library/ssl.html#ssl.CERT_NONE

jgyates, 2018-12-09:
I checked in an update that give you the options of require,none and optional

liltux, 2018-12-09:
@jgyates if it will help I followed this to setup and configure my MQTT broker:
https://www.digitalocean.com/community/tutorials/how-to-install-and-secure-the-mosquitto-mqtt-messaging-broker-on-ubuntu-16-04
So far no connection.

jgyates, 2018-12-09:
I will look at the api some more. I should probably get a cert and try to do some testing. I could use it with some other stuff I am running ..

liltux, 2018-12-09:
Here is my NodeRed Config...
[image]
[image]
[image]

kalex1114, 2019-04-17:
@lmamakos I'm using your configs and looks like some of your configs are going to be deprecated later on.

[image]

lmamakos, 2019-04-18:
Yes, time marches on..  Here's my present configuration, running with version 0.91.2 of Home Assistant.  Of course, you'll need to tweak the MQTT topic names because I happen to use some non-defaults to align with some existing conventions.

```
################################################
## generator sensors
################################################

sensor:
  - platform: mqtt
    state_topic: '19916/generator/Outage/Status'
    name: Genmon Outage Status

  - platform: mqtt
    state_topic: '19916/generator/Outage/Utility Voltage'
    name: Genmon Outage Utility Voltage
    value_template: '{{ value | regex_replace(" V", "") | float | round(1) }}'
    unit_of_measurement: "VAC"

  # status related sensors
  - platform: mqtt
    name: Genmon Generator Battery
    state_topic: '19916/generator/Status/Engine/Battery Voltage'
    unit_of_measurement: "V"
    value_template: '{{ value | regex_replace(" V", "") | float }}'

  - platform: mqtt
    name:  Genmon Generator RPM
    unit_of_measurement: "RPM"
    state_topic: '19916/generator/Status/Engine/RPM'
    value_template: '{{ value | int }}'

  - platform: mqtt
    name: Genmon Generator Frequency
    unit_of_measurement: "HZ"
    state_topic: '19916/generator/Status/Engine/Frequency'
    value_template: '{{ value | regex_replace(" Hz","") | float }}'

  - platform: mqtt
    name: Genmon Generator Output Voltage
    unit_of_measurement: "VAC"
    state_topic: '19916/generator/Status/Engine/Output Voltage'
    value_template: '{{ value | regex_replace(" V","") | float | round(1) }}'

  # current measrement is garbage on 22KW product, apparently
  - platform: mqtt
    name: Genmon Generator Output Current
    unit_of_measurement: "A"
    state_topic: '19916/generator/Status/Engine/Output Current'
    value_template: '{{ value | regex_replace(" A","") | float | round(1) }}'

  - platform: mqtt
    name: Genmon Generator Output Power
    unit_of_measurement: "W"
    state_topic: '19916/generator/Status/Engine/Output Power (Single Phase)'
    # this is normalized to watts from kilowatts
    value_template: '{{ value | regex_replace(" kW","") | float * 1000 | round(0) }}'

  - platform: mqtt
    name: Genmon Generator Last Alarm Log
    state_topic: '19916/generator/Status/Last Log Entries/Logs/Alarm Log'

  - platform: mqtt
    name: Genmon Generator Last Service Log
    state_topic: '19916/generator/Status/Last Log Entries/Logs/Service Log'

  - platform: mqtt
    name: Genmon Generator Last Action
    state_topic: '19916/generator/Status/Last Log Entries/Logs/Start Stop Log'

    # maint related sensors
[... 73 lines cut ...]
    action:
      - service: notify.pushover_louie
        data:
          message: "Utiliy Power Loss at {{  now().strftime('%Y-%m-%d %H:%M:%S') }}"
          title: "HASS"

# maybe also initiate a zwave heal network or something after power is restored to
# from standby generator after a ~45 second interruption?  Switch back is pretty
# quick, with only a couple hundred millisecond transfer time in the ATS..

#
  - alias: end power outage
    initial_state: true
    trigger:
      - platform: state
        entity_id: binary_sensor.genmon_power_outage
        from: 'on'
        to: 'off'
    action:
      - service: notify.pushover_louie
        data:
          message: "Transfer back to utiliy power at {{  now().strftime('%Y-%m-%d %H:%M:%S') }}"
          title: "HASS"

```

closed 2019-04-18 by lmamakos

GlennGoddard, 2020-05-22:
I know this is closed; but would adding the MQTT topics from the below link add what we need to Home Assistant.  I read above that the client_id was added but I can not find that in my MQTT topics.
https://www.home-assistant.io/docs/mqtt/discovery/

[13 more replies]