jgyates/genmon #156: Incorrect amperage and power usage reported on model G0071420 22KW/999 RESPONSE generator with evolution controller

lmamakos, 2018-08-06:
### Expected Behavior

More accurate reporting of current and power

### Actual Behavior

I noticed while the generator was running during its weekly Exercise, that incorrect values for current and voltage were reported.

### Steps to Reproduce (including precondition)

Seem to be reproduced during each exercise run on this generator.  Possibly I have a slightly differently configured Evolution controller for this slightly different generator.

### Screenshot or Picutres relating to the problem (if possible)

See below, dumps of JSON from genserv.py

### Your Environment

- Genereator Model: 
22kW air-cooled "Response" series generator, which appears to be a parallel product line that is sold through (at least) rural electric coops.  Here's some relevant information:

From the generator label:
```
Model      G0071420
Serial     3002841630
Item No    N/A
Prod Date  20180323
Volts      120/240       1 Phase
LPV AMPS   183.3/91.7    HZ 60
NG AMPS    162.5/81.3    RPM 3600
INSULATION CLASS H       1.0 PF
Controller P/N 0J8371C
Country of Origin US
DUTY RTG   EMERGENCY

```

From the evolution controller "Dealer" menu:
```
Firmware   V1.20
Hardware   V1.06
Bootloader V1.1
EEPROM     V1.04

Node    HZ   Volts
22KW    60   120/240
```

This appears to be similar to the air-cooled Guardian series.  There are some differences, which I'm not sure is due to this alternative product line, or subsequent versions of the generators in general.  There's no utility AC power outlet, and access to the compartment where the battery lives requires removing the side panel.

[image]
[image]

- Generator Registers: 
I captured these JSON blobs from genmon during the exercise run.
```
{
  "Registers": {
    "Num Regs": "244",
    "Not Changed": "15961889",
    "Changed": "354041",
    "Total Changed": "0.02",
    "Base Registers": [
      {
        "0239": "0000"
      },
      {
        "0238": "0000"
      },
      {
        "0039": "0000"
      },
      {
        "0038": "0015"
      },
      {
        "0034": "0008"
      },
      {
        "0037": "2a94"
      },
      {
        "0036": "0000"
      },
      {
        "0237": "0000"
      },
      {
        "0033": "0898"
      },
      {
        "0032": "14a4"
      },
      {
        "05f7": "060d"
      },
      {
        "05f6": "05d1"
      },
      {
        "05f5": "0007"
      },
      {
        "05f4": "0007"
      },
      {
        "05f3": "00c8"
      },
      {
        "05f2": "46a0"
      },
      {
        "05f1": "0000"
      },
      {
        "023e": "0000"
      },
      {
        "023d": "0000"
      },
      {
        "05ed": "0064"
      },
      {
        "0235": "0000"
      },
      {
        "023b": "0000"
      },
      {
        "023a": "0000"
      },
      {
        "003a": "0000"
      },
      {
        "003c": "0147"
      },
      {
        "003b": "1388"
      },
      {
        "05fa": "0005"
      },
      {
        "0059": "0000"
      },
      {
        "0058": "0000"
      },
      {
        "0053": "0000"
      },
      {
        "0052": "0000"
      },
      {
        "0057": "0000"
      },
      {
        "0056": "0000"
      },
      {
        "0055": "0000"
      },
      {
        "0054": "0000"
      },
      {
        "0208": "03f1"
      },
      {
        "0209": "0000"
      },
      {
        "005c": "0000"
      },
      {
        "005a": "0000"
      },
      {
        "005f": "0000"
      },
      {
        "005e": "0000"
      },
      {
        "005d": "0000"
      },
      {
        "020f": "0000"
      },
      {
        "020d": "0000"
      },
      {
        "020e": "0000"
      },
      {
        "020b": "0000"
      },
      {
        "020c": "0000"
      },
      {
        "020a": "0000"
      },
      {
        "000d": "0000"
      },
      {
        "000e": "1126"
      },
      {
        "000f": "0805"
      },
      {
        "024a": "0000"
      },
      {
        "000a": "0088"
      },
      {
        "000b": "0000"
      },
      {
        "000c": "0000"
      },
      {
        "0012": "009e"
      },
      {
        "0242": "0000"
      },
      {
        "0243": "0000"
      },
      {
        "0006": "0001"
      },
      {
        "0007": "0793"
      },
      {
        "0246": "0000"
      },
      {
        "0001": "00040000"
      },
      {
        "0244": "0000"
      },
      {
        "0245": "0000"
      },
      {
        "0248": "0000"
      },
      {
        "0249": "0000"
      },
      {
        "0008": "0020"
      },
      {
        "0009": "00f2"
      },
      {
        "0005": "1123"
      },
      {
        "002a": "6a78"
      },
      {
        "05ee": "036b"
      },
      {
        "0258": "0000"
      },
      {
        "0011": "009c"
      },
      {
        "0010": "0012"
      },
      {
        "0019": "0011"
      },
      {
        "002b": "0005"
      },
      {
        "002c": "1123"
      },
      {
        "0241": "0000"
      },
      {
        "002f": "0000"
      },
      {
        "002d": "0000"
      },
      {
        "002e": "0000"
      },
      {
        "0000": "0009"
      },
      {
        "0247": "0000"
      },
      {
        "0022": "46a0"
      },
      {
        "0020": "0000"
      },
      {
        "0021": "0000"
      },
      {
        "001f": "4ae7"
      },
      {
        "001e": "018f"
      },
      {
        "001d": "0000"
      },
      {
        "001c": "0000"
      },
      {
        "001b": "480d"
      },
      {
        "001a": "00c7"
      },
      {
        "025a": "0000"
      },
      {
        "01f4": "33303032383431363330"
      }
    ]
  }
}
```

The status response looked like this during the exercise run:
```
{
  "Status": {
    "Engine": {
      "Switch State": "Auto",
      "Engine State": "Exercising",
      "Battery Voltage": "13.60 V",
      "RPM": 1959,
      "Frequency": "32.00 Hz",
      "Output Voltage": "157 V",
      "Output Current": "91.00 A",
      "Output Power (Single Phase)": "14.29 kW",
      "Active Rotor Poles (Calculated)": 2
    },
    "Line State": {
      "Transfer Switch State": "",
      "Utility Voltage": "242 V",
      "Utility Voltage Max": "254V ",
      "Utility Voltage Min": "231V ",
      "Utility Threshold Voltage": "156 V"
    },
    "Last Log Entries": {
      "Logs": {
        "Alarm Log": "07/03/18 14:08:01 Charger Missing AC : Alarm Code: 0000",
        "Service Log": "",
        "Start Stop Log": "08/05/18 17:35:04 Running - Exercise "
      }
    },
    "Time": {
      "Monitor Time": "Sunday August 5, 2018 17:39:51",
      "Generator Time": "Sunday August 5, 2018 17:39"
    }
  }
}
```
The funky RPM and Frequency is correct, as the generator was running at the lower speed after the initial 3600 RPM start.  This scared the crap out of me the first time I saw it!  But these values match what was displayed on the Evolution controller, and what's expected in the manual.

It's the output power and output current that seems way out of whack.

What is curious, and maybe a clue is that the maintenance page in the JSON response (and the same web page) show "Unknown" for the model:
```
{
  "Maintenance": {
    "Model": "Unknown",
    "Generator Serial Number": "3002841630",
    "Controller": "Evolution, Air Cooled",
    "Nominal RPM": "3600",
    "Rated kW": "22",
    "Nominal Frequency": "60",
    "Fuel Type": "Propane",
    "Exercise": {
      "Exercise Time": "Weekly Sunday 17:35 Quiet Mode On"
    },
    "Service": {
      "Service A Due": "199 hrs or 07/02/2020 ",
      "Service B Due": "399 hrs or 07/02/2022 ",
      "Total Run Hours": "0 ",
      "Hardware Version": "V1.06",
      "Firmware Version": "V1.20"
    }
  }
}
```

- Genmon Version: 
            Generator Monitor Version : V1.9.36

jgyates, 2018-08-06:
Hi @lmamakos 

The software determines amperage in liquid cooled Evolution models by reading two modbus registers that is related to the output of the internal Current Transformers (CTs) in the generator. During installation of the the liquid cooled generator the CTs should be calibrated by the installer. If they are not or the calibration is off then the calculation made by the software will be off. To compensate for this there are two values in /etc/genmon.conf that can be modified to tweak the calculation of the current. The method to do this is described in item number 5 on (this)[https://github.com/jgyates/genmon/wiki/Appendix-4-Known-Issues] page.

Typically if your CTs were not calibrated you would measure the current using a clamp on type multimeter and determine the correct currentdivider and currentoffset values that are specific to the CTs in your generator. The CT calibration process is outlined in the diagnostic manual, but if you have the correct values for currentoffset and / or currentdivider then you don't need to calibrate your CTs to get the software working.

The model number is determined by looking up the serial number on the internet. This lookup is only performed if the model number is not valid (it is stored in /etc/genmon.conf once the lookup is performed) and only when the version of the software changes. Using your registers you submitted I was able to simulate the startup process and your lookup is performed correctly in my test. If you remove or change the "version=" entry in /etc/genmon.conf the lookup will be performed again (assuming your pi has internet access to perform the lookup)

Also, if QuietMode is enabled, then the generator will perform the Exercise Tests at a lower RPM (2000 instead of 3600) and Voltage Output. 

Let me know if you have any other questions.

lmamakos, 2018-08-06:
Shouldn't the current (and power) be zero during an exercise run since there's no load actually connected to the generator?

I actually have some power monitoring hardware in my load panel, so I have other means to really measure power consumption if I need to, it's just that this seemed to behave weirdly, given no load.

Next time my wife's not home, I'll force a switch to the generator and see how it behaves under load.  I can compare to the current measurement platform (a Brultech GreenEye Monitor thing) and see if the value might line up.  

I checked the end of my /etc/genmon.conf, and find:

```
# The following entries are written by genmon.py based on the generator
# settings, serial number and a one time lookup on the internet of the
# serial number. If you are not connected to the internet default
# settings will be used based values read from the generator. Some
# of these settings are editable via the web interface.
#    fueltype, model, nominalRPM, nominalKW, nominalFreq

# The following entries are written and maintained by genmon some of
# these are editable in via the web interface: version, autofeedback
version = V1.9.36
autofeedback = False
nominalfrequency = 60
nominalRPM = 3600
fueltype = Propane
nominalKW = 22
tanksize = 200
```
so genmon was able to update it once, but I don't see `model` in there.

jgyates, 2018-08-06:
The power is derived from the current reading.  Behind the scenes the software is  reading a register that changes with the current output. If the conversion is correct (based on calibration, currentdivider, etc, listed above). The output tracks with the output under load and the register also changes when exercising. As far as the expected output when exercising, after some research this is where I landed:

During exercise a dummy load is put on the generator, so current is produced, it is just not flowing to your house. It is my understanding that generac does this because it is not healthy for the engine to run without a load so the dummy load is present unless the transfer switch is activated.

To make the lookup occur you can remove or modify this line:

     version = V1.9.36

Then restart the software with this command:

     bash ./startgenmon.sh restart

This will force the software to do the lookup again.

Also, the results of the lookup should be logged to /var/log/genmon.log 

Let me know if you have any other questions.

lmamakos, 2018-08-06:
Yes, that seems to have done the trick.  Perhaps when I first powered it up, it wasn't connected to the wifi.. or something.  Now, at the tail end of /etc/genmon.conf I see:

```
autofeedback = False
nominalfrequency = 60
nominalRPM = 3600
fueltype = Propane
nominalKW = 22
tanksize = 200
version = V1.9.36
model = G0071420
```
which seems about right.

I wasn't aware of the dummy load; that makes sense.  I won't bother worrying about it until I actually do a transfer to the load in the house, and I can compare it to my other measurement means.

Thanks again for such a great software, and all the effort you've invested to figure how how all these variations happen to work!

lmamakos, 2018-08-06:
Speak of the devil, and he shall appear and make your power fail :-)

Just today I had a 15 minute outage, and got a chance to grab some data to compare the power usage as measured by my Brultech Green Eye Monitor, which has one channel connected to current transformers on the feed from the ATS, to the power reported by genmon.

This is even weirder than I expected.
[image]

Ok, so I get there might be some multiplicative factor difference.. looks like a factor of almost 4.  But there's the ramping behavior; maybe the generator has a filter to average/smooth the current measurements?

I'll have to maybe consider polling genmon more than once per minute for this data; the Brultech measurements are sampled every two seconds, so I have some reasonably high resolution data to look at.

I took a quick look on the generator control panel while it was running, and didn't see a display of what the power or current supplied by the generator is.  (Though frequency and RPM were in there..)  Did I miss that?  I took a quick look through the DEALER menu and didn't see it there.  It would have been interesting to see what it would have reported.

When the generator was installed, I fired up some large loads (electric clothes dryer, two electric wall ovens, maybe the water heater to about 15 or 16kW of load.  If the CT measurements in the generator were really uncalibrated by almost a factor of 4, that would have been 30kW and perhaps it would have shutdown or something? 

It feels like I'm looking for trouble here, as the generator did its job today.  But this kind of mystery is tough to let go.

MichaelB2018, 2018-08-06:
This behavior is consistent with what we have seen before. Somebody else on this forum may be able to explain things in more technical terms, but essentially the generator seems to have a minimum load that may be required to protect the equipment. It's about 20-30% of the rated output - so in your case about 6-10kW. If the load is lower than that, it will still show the value of the minimum load. After you exceed the minimum load, the genmon app seems reliable and scales nicely with actual load. See thread 112 for more detail. Also, the minimum load seems to warm up for up to an hour before it's stable. In between you see a consistent increase of the kW output. So your graph makes perfectly sense and matches what we have seen before.

jgyates, 2018-08-06:
Here is the link to issue 112 for reference:

https://github.com/jgyates/genmon/issues/112

I am wondering, if the 22kw does not have CTs external to the controller, why does it have a setting for Calibrate Current 1 and 2 in the dealer menu? and how does it detect over current? Could it be that there are CTs internal to the controller or they are calculating power some other way??

Based on the empirical data we have, we are reading a value from the controller (let's call it value x for the sake of discussion) that is related to, the total power or current from the controller, but not the actual power or current, hence the reason for the calculation to get current. Current Output, Power Output and Fuel Consumption are all derived off of this single "value x".

Out of all of the functionality in this software, this is the functionality that has the most uncertainty. More data points would help remove some of that uncertainty. If any one has validated the current output of their generator compared to what the software is reporting (other than @MichaelB2018 since he help a lot with the testing on this) please let us know in this thread. More data points will help!

Also, [here](https://github.com/jgyates/genmon/wiki/Appendix-4-Known-Issues) is the the page that describes know issues with the Generator Monitor. I tried to characterize the uncertainty in this on this page.

liltux, 2018-08-06:
So, no there are no Internal current transformers, they would all be external.  Generac introduced current transformers into the evolution air cooled product, but did not place them on the 22 KW.  The reason for this is the current transformers are not installed as a power monitoring application but as an electrical overload protection.  So as not to over current the stator windings and cause premature failure.  The 22kw model uses the same 999cc engine as the 20 KW but with a slightly different stator and minor engine compensations.  The trick here is that the engine in the 22 KW air cooled unit, minus any customer modifications, cannot over produce on the 22KW stator windings.  In short if you do not have enough engine or horsepower to overload the windings in a 22 KW stator.  Therefore the need of current transformers on the 22KW was not needed.  (got to save money where you can)  
The controller and the controller software is the same across all ranges.  The node ID is a small resistor bundle in the harness which allows the controller to know what unit it is on.  So there may still be a menu option to perform calibration but no transformers are there.  I imagine those inputs are just picking up electrical noise from the generator running.  

Now I will say that Generac tends to change things and not tell anyone.  If there are CT's on your generator, air cooled, they would be located in the battery compartment in the harness below the controller.  Usually somewhere near the fuel regulator and ct1 is around wire 11, ct2 is around wire 44.  If your 22kw unit does have current transformers please provide pictures, thank you.

By the way, Generac offers residential air cooled 3 day classes that discusses installation, theory or operation, and diagnostic.  No special requirements, its just not free.  They have them held across the country, and my company offers them as well.  (I teach them).  If interested email me liltux @ liltux.com.

lmamakos, 2018-08-06:
Thanks for the info.  As I have another means of measuring power, this isn't a huge issue for me.  I'll put it into the category of "working/broken as designed" and not worry about it.  Next time I have the covers off, I'll double check and look for evidence of the current transformers.  They ought not be too hard to find.  Maybe a project for this weekend, plus I'll try to get some better photos of my Raspberry Pi Zero W and associated cables and power supply while I'm at it.

The physical arrangement of my "Response" series generator is different from what I've seen in various YouTube videos and similar.  I have to remove the side-cover to access the compartment where the battery lives, and it's a bit of a pain to do so.  It also doesn't have the accessory AC power outlet on the side (nor the cover, for that matter).  I don't know if this is general 2018 change, or specific to the "Response" series vs. the "Guardian" series that it seems very similar to.

Thanks again to all for the background an explanation.  Quite a friendly, helpful and informative bunch that hang out here.

jgyates, 2018-08-12:
ok. Closing this issue. Free free to post to this thread if you want to look into this further.

closed 2018-08-12 by jgyates