jgyates/genmon #157: Start and transfer is not transferring

steve-bailey1, 2018-08-06:
### Expected Behavior

Under maintenance when clicking start and transfer ... expect to start run for a bit and transfer as it did in older versions. 

### Actual Behavior

Starts and runs but does not transfer. Stopping still does a cool down run. 

### Steps to Reproduce (including precondition)

Click start and transfer

### Your Environment

- Genereator Model: 20kw propane (sorry don’t have it in front of me at the moment)
- Genmon Version: latest as of 8/5 18:00 est

jgyates, 2018-08-06:
@steve-bailey1 

The way the software issues the remote commands is this: There are four unique commands (start, stop, startandtransfer, startexercise) (startexercise is not implemented in the web interface). Each of these is implemented with a specific modbus command. Once the command is sent the rest of the magic happens within the controller.  Once the command starttranser is issued the controller starts the generator then exercises a relay that signals the transfer switch to engage (line 23 to ground). I would look at the following:

1) make sure you are not using two wire start mode as this is not compatible with remote start. Sounds like this is not the case since normal start/ stop works for you.
2) Make sure you are not using a "smart" transfer switch as this is not compatible either. In a "smart" transfer switch, the switch  monitors the utility voltages and signals the generator to start and stop. This method will not work with the remote commands, but it sounds like this is not the case for you either if remote start and stop work.
3) Test your transfer switch by cutting the power at your switch box or by removing one of the two fuses in your transfer switch that protect the controller from the utility lines (Fuse F1 and F2). If you generator and transfer switch are working properly you should see your generator turn on, wait about 5 seconds, then the transfer switch should engage. If there is  a problem with the internal relay or so other issue keeping line 23 from going to ground then the transfer will never occur.

***Disclaimer: don't mess with your transfer switch unless you know what you are doing as it could hurt or kill you.***

The fact that is does a cool down is an indication that the command was issued correctly as the cool down does not occur on a normal remote start (i.e. start and no transfer). My guess is that your controller is not signaling your transfer switch. This could be a wiring issue or your controllers internal relay is dead.

Let me know if you have any other questions.

steve-bailey1, 2018-08-06:
Will definitely test it.  I was mostly wondering if something changed in the process since it worked a few months ago (last time I played with it) but you’re right ... I should check the functionality of the ats.

jgyates, 2018-08-06:
@steve-bailey1 , I am closing this for now. Feel free to comment in this thread if you have any other questions on this topic.

closed 2018-08-06 by jgyates