jgyates/genmon #158: Activating a username and password

tingraham1, 2018-08-06:
Just got Genmon setup a few days ago and it works awesome!  Only issue I am having is when I try to enable a username and password.  I turn on the "Optional - webserver security settings", leave on the "user self-signed certificate" and fill in the username and password boxes.  Once I save, the website stops responding as I would expect but I can't figure out a way to connect back to it.  I've used https://IPofgenmon, https://ipofgenmon:8000, http://ipofgenmon:8000 and nothing seems to work.  What am I doing wrong here?  I did make sure openssl was installed.

Any help would be greatly appreciated.  Thanks!!

jgyates, 2018-08-06:
The port used and some additional info relating to these settings is logged in /var/log/genserv.log

this may shed some light on the issue. https://IPofgenmon should be the correct address based on your description.

You can also send your log files to me with this URL:

      http://ipaddressofpi:8000/cmd/sendlogfiles

tingraham1, 2018-08-06:
Just sent log files.  It's saying something about address already in use and appears to be stuck in an endless restart of the service.

tingraham1, 2018-08-06:
From what I can tell, it is something related to the certificate.  The error SSL: CA_MD_TOO_WEAK goes away once switched back to http.

jgyates, 2018-08-06:
I did not receive your log files. You have to have outbound email configured for the log files to be sent to me.

If you have loaded the software multiple times without unloading it first you could see this situation with genserv.py. You may want to try rebooting your pi if you have not already.

jgyates, 2018-08-06:
One other thing, the message about address already in use is normal for that to occur once when applying settings from the web page. This has to do with the Flask library (the python web server) not releasing the port when it shuts down. So seeing this once for each restart is normal, seeing it happening multiple times in a row is not.

If the message " SSL: CA_MD_TOO_WEAK " is in your log file you may need to update your libraries.

This thread has some SSL related info that may be helpful:

https://github.com/jgyates/genmon/issues/121

jgyates, 2018-08-07:
the easiest way to send and email is to apply your settings, this will restart the program, which sends and email when the program starts. email errors or logged to /var/log/mymail.log

jgyates, 2018-08-12:
@tingraham1 

Have tried this:

     sudo pip install -U pyopenssl

as mentioned in the above referenced thread?

jgyates, 2018-08-15:
I will need to see your mymail.log and genserv.log files to diagnose your problem further. If you are not receiving email when you restart genmon (when the pi boots or when you apply settings on the web interface), then you email is not setup correctly. Once your email is setup you can use this command to send me your log files:

     http://ipaddressofpi:8000/cmd/sendlogfiles

If you don't want to setup your email, then you can copy and paste the contents of the files into this thread or attached them as files.

jgyates, 2018-08-27:
@tingraham1 

Are you  still having issues? I never received any log files so I am assuming you either do not have an issue any longer. I will close this issue however feel free to post to this thread if you still have questions.

closed 2018-08-27 by jgyates