jgyates/genmon #160: Adding serial number/hours to a new controller

sgtmaj27, 2018-08-18:
I have been using GenMon for over a year and have become quite dependent on it.  I recently had to replace the Nexus controller in my Generac 5870.  Obviously the new controller does not have my unit's serial number to report to GenMon and there is no way to program it into a Nexus controller.  So the serial number in the Maintenance screen is blank.  Also, there is no way to enter the total hours of the unit so the the controller obviously starts at zero.

What would be nice is if there were an entry in the Genmon.conf to enter a serial number (for units with replacement controllers) or default to the current method of obtaining it from the original controller.  

As for the total hours; another GenMon.conf entry would be nice to add previous total hours from an old controller which would then be added to the value received from the new controller.

I now track these values manually, however, I could see an operation that monitors several units needing the serial numbers off the units and the total hours handy to help in service monitoring.

Thanks for your great work!
Foster

jgyates, 2018-08-18:
This is a good idea. I will try to work this into the next update. 

For Evolution owners, the Evolution controller has the ability to set the run hours from the dealer menu, but it appears that the Nexus controller does not have this ability.

sgtmaj27, 2018-08-18:
Thanks, Jason.   A programmers work is never done!

jgyates, 2018-08-18:
This feature is now checked in. The genmon.conf settting for these are documented below :

     # (optional). If you controller has been replaced the serial number is not
     # present in replacement controllers. This option allows you to override a
     # missing serial number. This is a numeric value that must be ten digits.
     # Pad leading zeros if you serial number is less than 10 digits.
     serailnumberifmissing =

     # (optional). Nexus controllers do not have the ability to set the run hours.
     # If your controller has been replaced you can use this parameter to add hours
     # to the Run Hours reported by the controller. Note: The Evolution controller
     # has this functionality built into the dealer menu. Note: if this option is
     # used it will only modify the run hours displayed in the web interface.
     # modifying this value will not affect the controllers internal maintenance
     # counters. This value must be a numeric value.
     additionalrunhours =

Let me know if you have any questions

sgtmaj27, 2018-08-18:
Wow....that is some impressive response.

Thank you again for such a great project!

sgtmaj27, 2018-08-18:
Just loaded ver 1.9.39 and added the serial number and hours.  the program did not run and I received the following errors:
Starting genmon python scripts
Traceback (most recent call last):
  File "/home/pi/genmon/genmon.py", line 951, in <module>
    MyMonitor = Monitor(ConfigFilePath = ConfigFilePath)
  File "/home/pi/genmon/genmon.py", line 155, in __init__
    self.FatalError("Error opening controller device: " + str(e1))
  File "/home/pi/genmon/genmonlib/mycommon.py", line 93, in FatalError
    raise Exception(Message)
Exception: Error opening controller device: Missing config file or config file entries (evo/nexus): 'str' object has no attribute 'isnumeric'

If I commented out the serial number, the hours were added correctly.  I noticed that in the genmon.conf file  that serailnumberifmissing was spelled wrong.  When I changed it to serialnumberifmissing  the program will run but the new serial number is not listed.

sgtmaj27, 2018-08-18:
I changed "isnumeric" in generac_evolution.py line 3390 to "isdigit" The program runs with no errors and the new serial number is listed.  I am running Python 3 on my Raspberrypi.  I am not sure if thisc"troubleshooting" helps or not.

jgyates, 2018-08-18:
ok. I will take a look at this now. I tested this with a register file instead of a config file (my internal test setup) so I may have missed something.

jgyates, 2018-08-18:
ok. I checked in an update. @sgtmaj27 , you are correct. I added isnumeric instead of isdigit.  My initial method of testing did not check the read from the config file. I used a register file to test so I could simulate a controller with no serial number. 

Let me know if you have any other issues.

sgtmaj27, 2018-08-19:
Works perfectly....thanks!

jgyates, 2018-08-19:
Thanks for the feedback @sgtmaj27

closed 2018-08-19 by jgyates