jgyates/genmon #162: Logs & Outage UI elements don't work

madisdev, 2018-08-20:
### Expected Behavior

Clicking on Outages or Logs displays the data for those pages

### Actual Behavior

Clicking on the HTML elements doesn't do anything to the UI. Looking at the network, its making request and returning <h1>Unknown Command outage!</h1> or <h1>Unknown Command logs!</h1>

### Steps to Reproduce (including precondition)

Click on Outage or Logs in UI

This is a fresh install.

Commands work for ClientInterface.py

I saw this in the genserrv.log file 
2018-08-20 14:43:04,057 : Error in Process Command: dictionary update sequence element #0 has length 1; 2 is required

 Generator Monitor Version : V1.9.40

madisdev, 2018-08-20:
Here are the logs.
[genserv.log](https://github.com/jgyates/genmon/files/2303700/genserv.log)
[genmon.log](https://github.com/jgyates/genmon/files/2303701/genmon.log)

jgyates, 2018-08-20:
I don't known what the issue is yet, but let's try this to get another data point.

I checked in a version of genserv.py that has a little more debug info. Update the software with this command:

    sudo ./genmonmaint.sh update

Then restart with this command:

    ./startgenmon.sh restart

Then try to access via the web page and then send me your genserv.log file.

madisdev, 2018-08-20:
This is what is in log now:
2018-08-20 16:03:54,771 : Error in Process Command: dictionary update sequence element #0 has length 1; 2 is required : genserv.py:187

jgyates, 2018-08-20:
Looks like the error is on a function in the Flask module. 

What is the output of this command:

    sudo pip freeze | grep Flask

madisdev, 2018-08-20:
Traceback (most recent call last):
  File "/usr/bin/pip", line 9, in <module>
    load_entry_point('pip==1.5.6', 'console_scripts', 'pip')()
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 356, in load_entry_point

jgyates, 2018-08-20:
You appear to have something not right in your python libraries. The output should look like this:

    Flask==0.12.2

Did you install with the libraries with this command:

    bash ./genmonmaint.sh install

or did you install the libraries individually?

jgyates, 2018-08-20:
What is the output of this command:

     python -V

madisdev, 2018-08-20:
I used the install command. Should I try running that again. 

python -V 2.7.9

jgyates, 2018-08-20:
Yes. I would try that first. As it runs, look to see if there are any errors.

madisdev, 2018-08-20:
Yea I got errors on there. I think I might just do a clean wide of the SD card, I don't want to bother you anymore with issues on my pi install.

jgyates, 2018-08-20:
OK. I will close this for now. Feel free to post here if you have questions.

closed 2018-08-20 by jgyates