jgyates/genmon #164: Curious generator battery voltage pattern

lmamakos, 2018-08-24:
I've been collecting stats from my 22kW generator via genmon and MQTT, via Home Assistant, pushing it into InfluxDB and Grafana.  I see this interesting, periodic pattern in the battery voltage reported over the last couple days:

[image]

and over a longer period:

[image]

This is just the last couple of days, but it seems to be a daily pattern, where about every 23-24 hours, there's a sudden drop in the voltage.  (The small spike in middle of the lower voltage on 8/19 corresponds to the exercise cycle running.)

I'm wondering if there's some sort of battery test happening?  I don't think it's the charger cycling off and then a sudden drop, as there's a much more gradual drop in voltage that preceeds the sudden drop from about 13.5 volts to 12.8 volts, and then for about the next 5 hours, it remains at the lower voltage before jumping back up again (presumably when the charge cycle kicks in.)

The more data I have, the more questions it creates, apparently :-)

liltux, 2018-08-24:
Battery charge cycle of evolution air generator is 18 hrs and 26 hrs for liquid cooled.  Battery charge cycle starts when controller detects battery voltage drops below 12.6 volts.

jgyates, 2018-08-24:
From the diagnostic manual:

    
     During normal operation, the battery charger supplies all the power to the
     controller; the Generator battery is not used to supply power.
     The battery charger will begin its charge cycle when battery
     voltage drops below approximately 12.6V. The charger provides
     current directly to the battery dependant on temperature, and the
     battery is charged at the appropriate voltage level for 18 hours.
     At the end of the 18 hour charge period battery charge current
     is measured when the Generator is off. If battery charge current
     at the end of the 18 hour charge time is greater than a pre-set
     level, or the battery open-circuit voltage is less than approximately
     12.5V, an Inspect Battery warning is raised. If the engine cranks
     during the 18 hour charge period, then the 18 hour charge timer
     is restarted.
     At the end of the 18 hour charge period the charger does one of
     two things. If the temperature is less than approximately 40F
     the battery is continuously charged at a voltage of 14.1V (i.e. the
     charge voltage is changed from 14.6V to 14.1V after 18 hours). If
     the temperature is above approximately 40F then the charger will
     stop charging the battery after 18 hours.
     The battery has a similar role as that found in an automobile
     application. It sits doing nothing until it either self-discharges below
     12.6V or an engine crank occurs (i.e. such as occurs during the
     weekly exercise cycle). If either condition occurs the battery charge
     will begin its 18 hour charge cycle.

I assume that this behavior can change from what is documented with firmware updates.

lmamakos, 2018-08-24:
Thanks for the information, that's really helpful to explain what I see.   I had not considered a time-defined charge cycle like that, but it makes as much sense as anything else.

I wonder if the additional current drawn from the battery from the Raspberry Pi might cause a "Inspect Battery" warning at some point in the future.  Likely it would only be a bit further in advance of a actual "soft" battery.

It seems there's all sorts of cleverness at work inside that controller..

closed 2018-08-26 by jgyates

liltux, 2018-08-28:
Battery could be weak, low water, bad terminal connection?