jgyates/genmon #166: HTTPS communication

ToltingColtAcres, 2018-08-26:
More of a suggestion than an issue.

Can you remove (or comment out)

            HTTPPort = 443

in genserv.py ? 

This statement forces genmon to use port 443 for https communications, despite what may be specified as the http port in genmon.conf.

IMO there should be either:

1) a configuration parameter in genmon.conf, https_port, to allow users to specify which port they would like to have https requests serviced at, as the hard-coded default of 443 may be used by other things (like ngnix, apache, etc.), or

2) (the more appropriate option, IMO) simply have genmon use whatever the http_port parameter is to listen, regardless of the underlying protocol used (http or https). Since genmon will only communicate either http or https and not both at the same time like Apache, a second https_port parameter is really redundant.

In my installation, I opted to implement #2, and commented out the above line, so now my https requests are serviced at the default port 8000 specified in genmon.conf.

Unfortunately, this "breaks" my local copy from doing a git pull to retrieve updates :(

jgyates, 2018-08-26:
You are correct that the current implementation forces port 443 for HTTPS. I will look into changing this, I just want to make sure the UI logic matches the change and there are no side effects.

jgyates, 2018-08-26:
I checked in an update that should address this. I opted to not change the default behavior, but I added a new option in genmon.conf file to make this work as you suggested.

If you add the following to /etc/genmon.conf it will use the port of your choice if HTTPS is selected.

    https_port = 1234

Replace '1234' with the port of your choice. If this option is not present then the software will use port 443

ToltingColtAcres, 2018-08-26:
Thanks very much, I'll shelve my mod and re-pull.

closed 2018-08-26 by ToltingColtAcres