jgyates/genmon #168: Lost power and not showing output KW

zekyl314, 2018-08-29:
### Expected Behavior

During power outage, I would expect to see the Power Output.

### Actual Behavior

Nothing is showing, it shows zero.

### Screenshot or Pictures relating to the problem (if possible)
[image]

### Your Environment

- Genereator Model: Generac Evolution, Air Cooled, 17kW
- Generator Registers: I clicked send to developer button, did you get them?
- Genmon Version: V1.9.30

zekyl314, 2018-08-29:
I checked and the /etc/mymail.conf had email disabled for some reason.  I enabled it and restarted.  Please let me know if you got the logs and the registers.

jgyates, 2018-08-29:
I have not received anything yet. Did you receive an email when you restarted the software? If email is working properly you will receive two emails every time the software is started.

jgyates, 2018-08-29:
I don't have any data from you to base this one, however the likely problem is described on this page:

 https://github.com/jgyates/genmon/wiki/Appendix-4-Known-Issues

See item number 5

If you system was not calibrated during the install process the values read will be off. You can adjust how the output is calculated via the genmon.conf file. If the software detects the values are our of bounds (i.e. greater than the generator can output or negative values) it will report the values as zero.

zekyl314, 2018-08-29:
Please check now, I had an error in my SMTP config since I switched to G-Suite, I had to setup an APP password since I use 2FA for my email.

jgyates, 2018-08-29:
I received them. I will take a look.

jgyates, 2018-08-29:
I looked at your registers and log files. Since the generator is not running when the registers were recorded I can't tell if your issue is the calibration. Here are my recommendations:

1) if "Automated Feedback" is enabled on the settings page it will send and email to the developer (me) if the current reading is out of bounds. If you make sure this is enabled and start your generator, if I get an email then that is your issue.

2) Alternatively you can start your generator and send me another register dump while it is running

3) This is not related to your issue, but I noticed from you log file that you are overriding the auto-detect feature by using the "liquidcooled" and "evolutioncontroller" settings in /etc/genmon.conf. This is not causing any problem as you do have an air cooled evolution controller. If you comment out these entries in the /etc/genmon.conf file, the software will detect your controller based on the register values.

jgyates, 2018-08-29:
I added a new label in the issue tracker named "PowerDisplay". You can also review any previous issues that are related to this issue. In general, the way to overcome the calibration being off is to use the entries in /etc/genmon.conf named "currentdivider" and "currentoffset". See the [documentation](https://github.com/jgyates/genmon/wiki/1----Software-Overview#configuring-genmonpy) for details on these two entries

zekyl314, 2018-08-29:
okay, will do, thanks for the quick input.

zekyl314, 2018-08-29:
If I start the generator without transferring power, will you get the information you need if I send the registers?  I don't ever recall calibrating the generator.

jgyates, 2018-08-29:
yes, I will get the info needed, you do not need to transfer the power to do this test..

zekyl314, 2018-08-29:
Started and sent registers, please confirm you received them.

jgyates, 2018-08-29:
I did, you do have a negative reading from registers 003a and 003b, which are the current transformer registers so it does appear that your CTs have not been calibrated. To compensate for this you would need to measure the current on the E1 and E2 lines with a clip on type multimeter and modify the values of currentdivider and currentoffset until the combined values of E1 and E2 (from the meter reading) are close to the value report as the output current. Also you can view and graph the values of registes 003a and 003b from the display registers page (click icon in top right of the web interface).

zekyl314, 2018-08-29:
Thank you for the detailed information, I'll have to venture outside and see if I can get these readings and work on the adjustments.

lmamakos, 2018-08-30:
@zekyl314, or maybe, like in my case with a 22KW generator, you don't actually have current transformers installed?  Maybe take a look to see if they're actually present or not..  I see bogus data report for current measurements that are essentially unrelated to the actual load, likely because of noise or floating inputs.   See this other thread: https://github.com/jgyates/genmon/issues/156

zekyl314, 2018-08-31:
@lmamakos for the CT, are they in the generator or transfer switch?  I'm not 100% sure what I am looking for.

lmamakos, 2018-09-03:
I believe that the CT devices are in the generator.  Mine doesn't have them installed, so I'm not exactly sure what to tell you to look for :-)  I believe if you look at that thread I referenced,  at this response: https://github.com/jgyates/genmon/issues/156#issuecomment-410862244  it might be helpful in identifying the presence of a CT or not.

jgyates, 2018-09-16:
Closing this issue. Feel free to comment here if there are additional questions.

closed 2018-09-16 by jgyates