jgyates/genmon #169: Maximum kW Output only allows integer values

dprus, 2018-08-30:
I have a 22kW air cooled Generac. However, when running on Natural Gas it's only rated at 19.5kW per the manual. The settings in genmon don't allow me to input 19.5 since it requires an integer value.

I don't know what the downstream impact of this is, however. For now I've just input "20". Is this what's expected or is this a UI bug and it should allow me to enter a decimal?

Thanks.

jgyates, 2018-08-30:
The UI does limit this and internally genmon.py only looks at the integer portion. The downstream value is small. This value determines the limit of the kw graph and gauge in the UI. For 22kw NG this is all the value is used for. For fuel consumption calculation (not supported for NG) it is used  to determine the percentage of a full load the generator is under.  

For other generator models this value is used to to calculate a divisor for current output. 

In general I would set this value to 22kw, even if you do use NG and it is rated at 19.5, as this is the assumption that the software makes, but in your case it probably does not matter.

dprus, 2018-08-30:
ok cool. Thanks!

closed 2018-08-30 by dprus