jgyates/genmon #171: Trouble Stopping Generator During Outage

beschaaflh, 2018-08-31:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/Appendix-6---Serial-Troubleshooting

If you need to send you logs to the developer, if you email is setup and working properly you can issue this command in your browser:

     http://MYIPADDRESS:8000/cmd/sendlogfiles

Replace MYIPADDRESS to the IP address of you pi and port 8000 if you are using a port other than the default.

You can submit your registers from the Monitor page on the web interface.

### Expected Behavior

Turn off/stop generator

### Actual Behavior

Had a two-day power outage at our lake house.  Generator started immediately.  Tried to remotely stop the generator through web interface (initially on another pi on the network, then on the pi zero connected to the generator).  Also started ClientInterface on the pi zero and tried command generator:setremote=stop.  Generator didn't stop running.  I've been running genmon for about six months and it works great.  I recall remotely starting and stopping the generator from the web interface when I initially set it up.  Any idea why I couldn't stop the generator now?  Power was restored earlier today.  Thanks for your help!

BTW -- I reset the pi zero last last night and the outage log only shows the time from the restart until power was restored  

### Steps to Reproduce (including precondition)

{Please write here}

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Genereator Model: Generac 11KW Air Cooled
- Generator Registers:
- Genmon Version: 1.9.7

jgyates, 2018-08-31:
@beschaaflh 

Unfortunately, to my knowledge, the only way to stop a generator that is running due to an automatic start from an outage is by pressing the off button on the front panel. The remote commands will stop the generator that has been started with remote commands, but they will not change the controller from auto to manual mode or stop a generator during an outage. The remote commands as implemented in the controller firmware just don't function this way.

beschaaflh, 2018-09-01:
Ah -- at least I wasn't doing anything wrong.  Thanks for your help!

closed 2018-09-01 by beschaaflh