jgyates/genmon #172: Lost access to web app

Cleo1, 2018-09-04:
I have everything working. I was in the web app and enabled the webserver thinking I would expose to the internet. Assuming thats how it works? I changed the port to 8030 added a username and password. I have now lost access to the web app.  I still have access to the pi. Just no genmon app. What am I doing wrong?

jgyates, 2018-09-04:
If you are using the web interface to change the settings, when you use "Web Server Security Settings" the web server will use HTTPS by default (notice the port option is removed if "Web Server Security" setting is enabled)

    https://mypiaddress

If you use HTTPS, port 443 is assumed by your browser so this would be the same address:

     https://mypiaddress:443

If you want override the default port 443 when using secure settings by using HTTPS with a custom port number (starting with version 1.9.40) you will need to uncomment and modify the https_port value in the /etc/genmon.conf file.

Also, the log file /var/log/genserv.log will show the port and web security settings used once the software is restarted.

Cleo1, 2018-09-04:
Yes I am using web interface to make changes. genserv.log shows:
2018-09-04 08:32:34,715 : Starting /home/pi/genmon/genserv.py, Port:443, Secure HTTP: True, SelfSignedCert: True
2018-09-04 08:32:34,788 : Error in app.run:[Errno 98] Address already in use : genserv.py:961
2018-09-04 08:32:36,792 : Script: /bin/bash /home/pi/genmon/startgenmon.sh restart
2018-09-04 08:32:42,167 : Starting /home/pi/genmon/genserv.py, Port:443, Secure HTTP: True, SelfSignedCert: True
2018-09-04 08:32:45,447 : Error in app.run:[SSL: CA_MD_TOO_WEAK] ca md too weak (_ssl.c:2699) : genserv.py:961
2018-09-04 08:32:47,450 : Script: /bin/bash /home/pi/genmon/startgenmon.sh restart
2018-09-04 08:32:52,826 : Starting /home/pi/genmon/genserv.py, Port:443, Secure HTTP: True, SelfSignedCert: True
2018-09-04 08:32:52,900 : Error in app.run:[Errno 98] Address already in use : genserv.py:961
2018-09-04 08:32:54,903 : Script: /bin/bash /home/pi/genmon/startgenmon.sh restart

jgyates, 2018-09-04:
what URL are you using in your browser?

You log shows that you should use 443, so you should use 

    https://myipaddress

Cleo1, 2018-09-04:
I am using https://192.168.XXX.XXX. I have tried all combinations with and without ports. Receive site cannot be reached. I can ping the IP and remote into the Pi without issues. Can I just copy over the genmon.config file to get it going again. Hurricane on its way to my area.

jgyates, 2018-09-04:
This thread has some discussion on resolving the SSL: CA_MD_TOO_WEAK issue:

https://github.com/jgyates/genmon/issues/121

Cleo1, 2018-09-04:
Thanks. usehttps = False was already set.

jgyates, 2018-09-04:
This should correct the SSL error:

    sudo apt-get install python-pip
    sudo apt-get install build-essential libssl-dev libffi-dev python-dev
    sudo pip install -U cryptography     
    sudo pip install pyopenssl

Let me know if you still have any questions. It is not clear from your response that you are still having an issue.

Cleo1, 2018-09-04:
Sorry. Yes still having the issue. I will try the above commands. Stupid question. I'm new to linux. To save to genmon.conf do I need to only stop genmon using ./startgenmon.sh start

Cleo1, 2018-09-04:
Sorry meant ./startgenmon.sh stop

jgyates, 2018-09-04:
Yes, to stop genmon type:

    ./startgenmon.sh stop

To modify /etc/genmon.conf do the following:

     sudo nano /etc/genmon.conf

Then make you edits, then type "Ctrl+x" to exit the nano editor, you will be prompted to save you changes.

Cleo1, 2018-09-04:
I ran the SSL commands above. Started genmon. Still no access to the web app using https//192.168.XXX.XXX. Started genmon. Receive the following over and over

Stopping genmon python scripts
Starting genmon python scripts
OK : Auto, Off - Ready
 OK - Init complete.
Stopping genmon python scripts
Starting genmon python scripts
OK : Auto, Off - Ready
 OK - Init complete.
Stopping genmon python scripts
Starting genmon python scripts
OK : Auto, Off - Ready
 OK - Init complete.
Stopping genmon python scripts
Starting genmon python scripts
OK : Auto, Off - Ready
 OK - Init complete.
Stopping genmon python scripts
Starting genmon python scripts

jgyates, 2018-09-17:
@Cleo1 

There is more discussion on this topic on this thread: https://github.com/jgyates/genmon/issues/176