jgyates/genmon #174: Nagios XI Monitoring Setup

jonathanpisarczyk, 2018-09-13:
Question

I'm trying to get my PI and genmon setup with Nagios XI monitoring.  I uploaded the plugin to Nagios XI "check_monitor_system.py".  However I don't see the "service" showing.  Any help would be appreciated? 

[image]

jgyates, 2018-09-13:
you will need to define the plugin as a service for the specific system (the raspberry pi) that Nagios is monitoring.

The plugin is just a python script that Naigos calls if directed to do so. There are many plugins in the same directory for a given Nagios install. You can test the plugin outside of Nagios:

     python check_monitor_system.py -H  YOURIPADDRESS

If the system is working, it should return this:

     OK : Auto, Off - Ready

I will not be much help with Nagios XI as I run an earlier version of Nagios. For Nagios 3 (the one that comes with the debian / raspbian repositories) you have to modify files in the /etc/nagios3 directory to enable specific services. 

Here is what I did for Nagios3:

In the /etc/nagios3/commands.cfg file I added this:

     define command {
         command_name check_generator_system_status
         command_line $USER1$/check_monitor_system.py -H $HOSTADDRESS$
     }

then in the services file I added (located in /etc/nagios3/conf.d/ directory) 

    define service{
        use                                  generic-service
        host_name                      rpi.generator.local
       service_description         Generator Monitor System
       check_command             check_generator_system_status
     }

Note: rpi.generator.local is the name that is specific to my nagios install so you will need to change that to match your setup.

In NagiosXI there is likely an easier way than modifying config files, but this is how it is done for Nagios3.

jonathanpisarczyk, 2018-09-14:
Hello @jgyates 

Your tips above pointed me in the right direction.  I was able to get it working with Nagios XI.  I created a command using the gui.  Using VI in the config files didn't work... 

My command /usr/bin/python $USER1$/check_monitor_system.py -H $HOSTADDRESS$

I then created a new service using the gui.  

[image]

[image]

Thanks again

jonathanpisarczyk, 2018-09-14:
Here's where I created the command:

[image]