jgyates/genmon #175: generator not responding

jboardman1, 2018-09-14:
Im having trouble with the program

Both Serialtest and serial config pass

 when I plugin to the generator I just get 

Monitor :

        Generator Monitor Stats :
            Monitor Health : Not receiving data. 
            Controller : Nexus, Air Cooled
            Run time : Generator Monitor running for 0:42:14.
            Generator Monitor Version : V1.7.9

        Serial Stats :
            Packet Count : M: 843, S: 0
            CRC Errors : 0 
            CRC Percent Errors : 0.00
            Packet Timeouts : 842
            Discarded Bytes : 0
            Comm Restarts : 0
            Packets Per Second : 0.33

What is the next step in troubleshooting?

jgyates, 2018-09-14:
this page has info on serial troubleshooting:

https://github.com/jgyates/genmon/wiki/Appendix-6---Serial-Troubleshooting

"If this test passes but you still can not receive data with the Generator Monitor, then try swapping your transmit and receive connections in your cable (not the converter board). Note: the diagram for the cable in the repository has wiring for both a DTE and DCE serial connector. The wiring is different for a DTE vs a DCE connector. Another thing that could be wrong if you fail this test is a bad connection in your molex connector."

Also make sure you have your /etc/genmon.conf file updated with the correct serial port you are using. The genmon.conf file must be copied from the genmon directory to the /etc directory unless you choose to have the genmonmaint.sh program do it during install.

Let me know if you still have issues.

jboardman1, 2018-09-14:
this issue is resolved, the pins needed to be reversed sorry for the trouble.

Awesome program thank you!

closed 2018-09-14 by jboardman1

jgyates, 2018-09-14:
No worries, glad it is working.