jgyates/genmon #176: Enabling HTTPS

thepfef, 2018-09-15:
Genmon works fine in basic setup using HTTP. When changing to HTTPS = True in genmon.conf and also use  useselfsignedcert = True for simplicity. Also set username and password.
Manually start genmon and then genserv.

In the text window where I started genmon, I get the following:

pi@raspberrypi:~/genmon $ sudo python genmon.py
Traceback (most recent call last):
  File "genmon.py", line 955, in <module>
    time.sleep(0.01)
TypeError: Close() takes exactly 1 argument (3 given)
pi@raspberrypi:~/genmon $ ^C
pi@raspberrypi:~/genmon $ 

In the text window where I started genserv, I get the following; genmon continually stops and restarts as I get blasted with emails until I stop both programs.

pi@raspberrypi:~/genmon $ Starting genmon python scripts
OK : Auto, Off - Ready
 OK - Init complete.
Stopping genmon python scripts
Starting genmon python scripts
OK : Auto, Off - Ready
 OK - Init complete.
Stopping genmon python scripts
Starting genmon python scripts
OK : Auto, Off - Ready
 OK - Init complete.
Stopping genmon python scripts
Starting genmon python scripts

I have installed all the programs/libraries as directed with no errors.

sudo apt-get install python-pip
sudo apt-get install build-essential libssl-dev libffi-dev python-dev
sudo pip install -U cryptography     
sudo pip install pyopenssl

Any ideas?

jgyates, 2018-09-15:
the errors you listed first are normal when the program is restarted. The app is closed with a kill command and relaunched.

It is also normal that console messages are printed to the screen if the console window is open.

I would need to see your registers and log files to give any additional insight. You can send the log registers to me by entering this URL: 

     http://IPADDRESSOFYOURPI:8000/cmd/sendlogfiles

Replacing IPADDRESSOFYOURPI with the actual IP address of your raspberry pi. The above command will gather the genmon specific log files and email them to me, but like the "Send Registers" button, this requires your outgoing email be working. 

Based on the output above (repeated starting and stopping) it looks like something is wrong with the Flask app. It may be the crypto libraries. I read that the default crypo algorithm changed and this may be affecting Flask, but I have not validated this therory. 

The log files that are relevant are:

/var/log/genmon.log
/var/log/genserv.log   
/var/log/myserial.log 
/var/log/mymodbus.log 
/var/log/mymail.log 

The first two (genmon.log and genserv.log) are likely the ones we want to start with. If you prefer you can attach the files to this thread (attached files is not an option in the mobile version of the github site).

jgyates, 2018-09-15:
Note: the  http://IPADDRESSOFYOURPI:8000/cmd/sendlogfiles command assumes HTTP not HTTPS. Since you HTTPS is not working you would need to change back to HTTP to send this command. It will work with HTTPS, but there appears to be another issue keeping that mode from working.

thepfef, 2018-09-15:
Tried the sendlogfiles. got a screen back that said "log files submitted"

jgyates, 2018-09-15:
That is expected, however I have not received them. Do you see the email is your sent folder?

jgyates, 2018-09-15:
what does your /etc/mymail.log file show?

jgyates, 2018-09-15:
sorry the mail log file should be /var/log/mymail.log

jgyates, 2018-09-15:
As another data point, what is the output of these two commands:

   pip show pyopenssl

and 

   pip show cryptography

My system shows this:

    pip show pyopenssl
    Name: pyOpenSSL
    Version: 18.0.0
    Summary: Python wrapper module around the OpenSSL library
    Home-page: https://pyopenssl.org/
    Author: Hynek Schlawack
    Author-email: hs@ox.cx
    License: Apache License, Version 2.0
    Location: /usr/local/lib/python2.7/dist-packages 
    Requires: cryptography, six

    pip show cryptography
    Name: cryptography
    Version: 2.3.1
    Summary: cryptography is a package which provides cryptographic recipes and primitives to Python developers.
    Home-page: https://github.com/pyca/cryptography
    Author: The cryptography developers
    Author-email: cryptography-dev@python.org
    License: BSD or Apache License, Version 2.0
    Location: /usr/local/lib/python2.7/dist-packages
    Requires: enum34, asn1crypto, idna, cffi, six, ipaddress

thepfef, 2018-09-17:
my system is at the same revision as yours in previous post.

following entries in genserv.log file.

this line is logged when HTTPS is selected as true and selfsigned is true. (wanted to start with a basic secure login setup and work from there)

2018-09-17 12:36:35,403 : Starting genserv.py, Port:443, Secure HTTP: True, SelfSignedCert: True
2018-09-17 12:36:48,083 : Error in app.run: [SSL: CA_MD_TOO_WEAK] ca md too weak (_ssl.c:2699) : genserv.py:879

jgyates, 2018-09-17:
What version of Flask do you have installed? These commands will show this:

    pip show Flask
    pip show Werkzeug

my install shows this:

    pi@genmon:~/genmon $ pip show Flask
    Name: Flask
    Version: 1.0.2
    Summary: A simple framework for building complex web applications.
    Home-page: https://www.palletsprojects.com/p/flask/
    Author: Armin Ronacher
    Author-email: armin.ronacher@active-4.com
    License: BSD
    Location: /usr/local/lib/python2.7/dist-packages
    Requires: click, Werkzeug, itsdangerous, Jinja2
    pi@genmon:~/genmon $ pip show Werkzeug
    Name: Werkzeug
    Version: 0.14.1
    Summary: The comprehensive WSGI web application library.
    Home-page: https://www.palletsprojects.org/p/werkzeug/
    Author: Armin Ronacher
    Author-email: armin.ronacher@active-4.com
    License: BSD
    Location: /usr/local/lib/python2.7/dist-packages
    Requires: 

I know there is an issue with the openssl library changing a encryption variable and depending on what versions of the library (this could vary depending if it was installed via apt-get or pip) were installed it will complain.

this post has some details: https://projects.bentasker.co.uk/jira_projects/browse/LOC-28.html

thepfef, 2018-09-17:
same versions

jgyates, 2018-09-17:
are you running Raspbian 9?

jgyates, 2018-09-17:
you and type this to get the OS version:

   cat /etc/os-release

    pi@genmon:~/genmon $ cat /etc/os-release
    PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
    NAME="Raspbian GNU/Linux"
    VERSION_ID="9"
    VERSION="9 (stretch)"
    ID=raspbian
    ID_LIKE=debian
    HOME_URL="http://www.raspbian.org/"
    SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
    BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

thepfef, 2018-09-17:
yes same version

jgyates, 2018-09-17:
I rested HTTS, it will load, there is one issue with usernane / password that I am working on now.

closed 2018-09-17 by jgyates

thepfef, 2018-09-17:
updates did the trick for me. it now works with simple https and username and password.

thepfef, 2018-09-17:
looks like second time i pulled up the webpage it just let me in without a username and password

jgyates, 2018-09-17:
I just checked in a update related to the HTTS username / password.

jgyates, 2018-09-17:
your browser will keep session info. If you close your browser it should ask for the credentials .