jgyates/genmon #178: Upgrate to 1.10 killed my connection

jcherney, 2018-09-16:
Upgraded to 1.10 tonight, and could not reconnect. I restarted the PI by disconnecting/reconnecting the power but no change. I can ping the ip address so the wifi is working. 

Thoughts??

jgyates, 2018-09-16:
I would log in with ssh from another system and look at the log files. If you run windows you may have to install a ssh client. This also assumes you have ssh enabled on the pi. It is a very useful tool for remotely logging in to a linux system. 

Nothing related to networking changed so my guess is that something may have gone wrong reloading the software so the web server is not loaded.

If you don't have ssh enabled on the pi then you can enabled by running 

   sudo raspi-config

Then select Interfacing Options from the menu.

jgyates, 2018-09-16:
Windows SSH clients: https://www.ssh.com/ssh/client

it is built in on a Mac and linux

jcherney, 2018-09-16:
I have putty installed.

jgyates, 2018-09-16:
look at these three files:

/var/log/genmon.log
/var/log/genserv.log
/var/log/genloader.log

You can display them to the console by typing 

     cat /var/log/genmon.log

jcherney, 2018-09-16:
The genloader seems to be showing an error. Error check system readiness. Exiting.

jcherney, 2018-09-16:
Also, Error in CHeck system: global name "line" is not defined : genloader.py:117

jcherney, 2018-09-16:
Trying to rerun the logs and it's showing permission denied.

jgyates, 2018-09-16:
I checked in an updated genloader.py that corrected the error line 117.

the genloader program checks to see if the required libraries are installed. The bug was in the line that logs the error to file.

I would recommend this:

update to the latest with this command line:

    sudo bash ./genmonmaint.sh update

Then restart the software with this command.

    sudo bash ./startgenmon.sh restart

One the restart, genmonloader.py  should attempt to install the missing library, then it will start genmon.

Let me know if you have any problems.

jcherney, 2018-09-16:
the update commant shows no such file or directory

jcherney, 2018-09-16:
command

jcherney, 2018-09-16:
Looking at your code (and I am no coder!), should it read .......updategenmon instead of update?

jgyates, 2018-09-16:
you you first log in with ssh you need to change directories to the genmon directory:

    cd genmon
    sudo bash ./genmonmaint.sh update
    sudo bash ./startgenmon.sh restart

I made and assumption that this was know since this is where all of the project files are located. Let me know if you still have an issue.

jcherney, 2018-09-16:
Thanks. Sorry, I only know enough to get myself in trouble! I’ll do this when I get home.

jcherney, 2018-09-17:
Got 2 errors while loading:  warning required library pyowm not installed attempting to install....
and   same with pytz not installed. Not sure if they installed or not.

jgyates, 2018-09-17:
if you did not get another warning about the library failing to install then everything went as expected. The warning is just to let you know that the libraries were missing and an attempt is made to correct this.

Are you able to load the software and view the web interface now? if not does the file /var/log/genserv.log have any errors?

jcherney, 2018-09-17:
The web interface is now loading. Its acting a bit strangely though. Seems to take forever to load (it does load), and won't load at all on my Iphone as it did before. Not sure what this has to do w the original problem if at all.

jcherney, 2018-09-17:
Thank you for the update!

jcherney, 2018-09-17:
After I typed the above, it's now loading on my phone. Don't know why the holdup, but its working. Thanks for your help!

jgyates, 2018-09-17:
ok. No problem, glad it is working. I will close this issue.

closed 2018-09-17 by jgyates