jgyates/genmon #181: odd behavior

diverbill, 2018-09-21:
Hello, 
First let me say this is an awesome piece of software! I'm experiencing several small issues and i have been reading and figured i would reach out. Please let me know if i missed something along the way.

Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/Appendix-6---Serial-Troubleshooting
i was until i added the ground wire - this has been resolved (great documentation)
If you need to send you logs to the developer, if you email is setup and working properly you can issue this command in your browser:

     http://MYIPADDRESS:8000/cmd/sendlogfiles
result - Unknown Command sendlogfiles!
Replace MYIPADDRESS to the IP address of you pi and port 8000 if you are using a port other than the default.

You can submit your registers from the Monitor page on the web interface.
-sent
### Expected Behavior
issue 1 click open log file & outage file via web ui prest the requested information 
issue 2 fuel reading
also getting an error on pytz - saw documentation this is no longer needed.
### Actual Behavior
issue 1 - nothing - the page does not load - i see requests in the http log but nothing is ever displayed to the page.
issue 2 - I have a fuel pressure reading of 93 (tank is !98-99% full) it holds 32 US Gallons. 
     I reset the logs and it shows 0 on the dashboard. 
### Steps to Reproduce (including precondition)
unsure how to reproduce this is my only pi and generator - I installed and also just ran an update both issue remain - no log file (however it shows in ~/genmon/kwlog.txt

fule level - i reset the power/fuel logs and it still shows 0% remaining while the fuel pressure sensor reads 93

### Screenshot or Pictures relating to the problem (if possible)
screenshot of clicking on the logs
[image]

### Your Environment

- Generator Model: RD020 Generac Protector 20KW Diesel (evolution 1.0 fw 1.39)
- Generator Registers: {Please write here} - unsure how to provide this
[Generator Monitor.pdf](https://github.com/jgyates/genmon/files/2403554/Generator.Monitor.pdf)

- Genmon Version: V1.10.2
- pi 3b running on POE with splitter with wire 5 on the seral to wire 2 on the controller for common ground.

jgyates, 2018-09-21:
Hi @diverbill 

Have you set the Tank Size on the settings page to 32? You log file has a lot of the same error in so if you can let me know this then it will tell me the direction to go to get your problems resolved.

Jason

jgyates, 2018-09-21:
Also, this page has some info on fuel monitoring and power reporting:

https://github.com/jgyates/genmon/wiki/Appendix-8-Monitoring-Fuel-and-Power-Usage

Let me know how your Tank Size is set...

diverbill, 2018-09-21:
Hi @jgyates 

I have added it in the setting page as well as in /etc/genmon.conf & /home/pi/genmon/genmon.conf
[image]
[image]

Also attaching Serial Stats just as an FYI
[image]

-Bill

jgyates, 2018-09-21:
OK. BTW, if you add it via the web interface you should be set. The web interface updates the /etc/genmon.conf file. The file /genmon/genmon.conf file is just a copy and has not effect on the settings. Ideally you should just need to make changes from the web interface since if you add it to /etc/genmon.conf and it already exist you would have two settings.

If the tanks size setting is new, then send me your log files again with the Now that you have the tank size can you send me your log files again with the  http://MYIPADDRESS:8000/cmd/sendlogfiles URL.

closed 2018-09-21 by jgyates

diverbill, 2018-09-21:
I have added and restarted the app - it does not make any difference. 
I also verified i only have one entry within genmon.conf
[image]

I also just tried to send the logs and got this message
[image]

also just wondering about the log tab - does this take time to load?

thanks,
-Bill

jgyates, 2018-09-21:
Sorry, instead of:
   
    sudo bash genmonmaint.sh updatelibraries

type:

     sudo bash genmonmaint.sh updatelibs
updatelibs

diverbill, 2018-09-21:
ok here is the step by step
pi@raspberrypi:~/genmon $ sudo bash genmonmaint.sh update
This script will update genmon to the latest version from the github repository. This script requires internet access. If you have modified any files in the genmon directory, they will be overwritten. Configuration files in the /etc directory will not be overritten.   Continue? (y/n)  y
Updating genmon...
HEAD is now at 2efcd1c updated for future changes.
pi@raspberrypi:~/genmon $ sudo bash genmonmaint.sh updatelibraries
No valid command given. Specify 'update', 'install' or 'updatelibs' on command line
pi@raspberrypi:~/genmon $ sudo bash genmonmaint.sh updatelibs

pi@raspberrypi:~/genmon $ sudo bash genmonmaint.sh updatelibs
Traceback (most recent call last):
  File "/usr/bin/pip", line 9, in <module>
    load_entry_point('pip==1.5.6', 'console_scripts', 'pip')()
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 356, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 2476, in load_entry_point
    return ep.load()
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 2190, in load
    ['__name__'])
  File "/usr/lib/python2.7/dist-packages/pip/__init__.py", line 74, in <module>
    from pip.vcs import git, mercurial, subversion, bazaar  # noqa
  File "/usr/lib/python2.7/dist-packages/pip/vcs/mercurial.py", line 9, in <module>
    from pip.download import path_to_url
  File "/usr/lib/python2.7/dist-packages/pip/download.py", line 25, in <module>
    from requests.compat import IncompleteRead
ImportError: cannot import name IncompleteRead
Traceback (most recent call last):
  File "/usr/bin/pip", line 9, in <module>
    load_entry_point('pip==1.5.6', 'console_scripts', 'pip')()
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 356, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 2476, in load_entry_point
    return ep.load()
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 2190, in load
    ['__name__'])
  File "/usr/lib/python2.7/dist-packages/pip/__init__.py", line 74, in <module>
    from pip.vcs import git, mercurial, subversion, bazaar  # noqa
  File "/usr/lib/python2.7/dist-packages/pip/vcs/mercurial.py", line 9, in <module>
    from pip.download import path_to_url
  File "/usr/lib/python2.7/dist-packages/pip/download.py", line 25, in <module>
    from requests.compat import IncompleteRead
ImportError: cannot import name IncompleteRead
Traceback (most recent call last):
  File "/usr/bin/pip", line 9, in <module>
    load_entry_point('pip==1.5.6', 'console_scripts', 'pip')()
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 356, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 2476, in load_entry_point
    return ep.load()
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 2190, in load
    ['__name__'])
  File "/usr/lib/python2.7/dist-packages/pip/__init__.py", line 74, in <module>
    from pip.vcs import git, mercurial, subversion, bazaar  # noqa
  File "/usr/lib/python2.7/dist-packages/pip/vcs/mercurial.py", line 9, in <module>
    from pip.download import path_to_url
  File "/usr/lib/python2.7/dist-packages/pip/download.py", line 25, in <module>
    from requests.compat import IncompleteRead
ImportError: cannot import name IncompleteRead
Traceback (most recent call last):
  File "/usr/bin/pip", line 9, in <module>
    load_entry_point('pip==1.5.6', 'console_scripts', 'pip')()
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 356, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 2476, in load_entry_point
    return ep.load()
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 2190, in load
    ['__name__'])
  File "/usr/lib/python2.7/dist-packages/pip/__init__.py", line 74, in <module>
    from pip.vcs import git, mercurial, subversion, bazaar  # noqa
  File "/usr/lib/python2.7/dist-packages/pip/vcs/mercurial.py", line 9, in <module>
[... 122 lines cut ...]

Stopping....
Stopping gensyslog.py
Stopping gensms_modem.py
Stopping gensms.py
Stopping genpushover.py
Stopping genmqtt.py
Stopping genlog.py
Stopping gengpioin.py
Stopping gengpio.py
Stopping genserv.py
Stopping genmon.py
Starting....
Starting /home/pi/genmon/genmon.py
Starting /home/pi/genmon/genserv.py
pi@raspberrypi:~/genmon $ WARNING: System Initializing.  Not receiving data. : Auto, Off - Ready
 OK - Init complete.
 * Running on http://0.0.0.0:8000/
192.168.0.93 - - [20/Sep/2018 22:27:22] "GET /cmd/registers_json HTTP/1.1" 200 -
192.168.0.93 - - [20/Sep/2018 22:27:23] "GET /cmd/maint_json HTTP/1.1" 200 -
192.168.0.88 - - [20/Sep/2018 22:27:23] "GET /cmd/gui_status_json HTTP/1.1" 200 -
192.168.0.93 - - [20/Sep/2018 22:27:24] "GET /cmd/gui_status_json HTTP/1.1" 200 -
192.168.0.88 - - [20/Sep/2018 22:27:25] "GET /cmd/gui_status_json HTTP/1.1" 200 -
192.168.0.88 - - [20/Sep/2018 22:27:25] "GET /cmd/status_json HTTP/1.1" 200 -
192.168.0.88 - - [20/Sep/2018 22:27:25] "GET /cmd/registers_json HTTP/1.1" 200 -

diverbill, 2018-09-21:
I did make all the required changes via the webpage after it started clean. 
I will also send another set of logs along.

jgyates, 2018-09-21:
Looks like there is something wrong with your libraries. Let's try to reinstall the libraries with this command:

First stop the software:

     bash ./startgenmon.sh stop

Then run the install again:

     bash ./genmonmaint.sh install 

When this is complete then restart:

     bash ./startgenmon.sh restart

jgyates, 2018-09-21:
Also, I checked in an update that should give us some additional info if you still get the errors after the re-install.

Once the above is complete you can update from the web interface on the Monitor page.

diverbill, 2018-09-21:
Hi @jgyates 
While i was running the requested commands python did the same thing again so I started digging more into the PI. I found another issue, it looks like i am running low on space. I am going to order a larger SD card and reinstall everything. I will update once this is complete. 
Thank you for the support.
-Bill

jgyates, 2018-09-21:
ok. Make sure you get a SD card that is at least class 10. The speed of a class 10 is recommended for a raspberry pi.