jgyates/genmon #182: Informational: RS232 surge suppressors

ToltingColtAcres, 2018-09-21:
Been using Genmon for about a month now. 

Early last week, we had bad thunderstorms in the area and apparently had a lightening strike on a pole nearby. The surge travelled down the cable internet coax and fried the cable modem (physically had a hole burn through one of the chips on the cable modem motherboard). 

From the cable modem, the surge travelled though Ethernet to a switch, which had three other devices connected to it... 2 Pi 3B+’s (one running GenMon, one running Pi-Star, and one of my UHD TVs). All three devices were toast, both Pi’s fried and the TV lost the main circuit board (replacement: $50 on eBay).

Once I replaced the Pis, I couldn’t get Genmon to start.... No interaction with the modbus port on the back of the evolution controller. I reconnected my wireless display and it could not communicate with the controller either (flashes yellow/red). Seems the surge also took out the RS232 communications port on the evolution controller. Controller otherwise works okay (e.g. control panel is fine, exercise is working, etc.)

Generac tech was out here today to do diagnostics... No resolution... ordering a replacement evolution controller from Generac, will be here in 5-14 days.

So... raises an interesting question... anyone have any experience with RS232 surge suppressors? I’d hate for this to happen again once I get the controller replaced.

jgyates, 2018-09-21:
@ToltingColtAcres 

I don't have a data point on the new Evolution controller 2.0 yet, so I don't know that much about it at this point.

Regarding surge protection. For the cable modem, I use two of these:

https://smile.amazon.com/gp/product/B003AVRLZI/ref=oh_aui_search_detailpage?ie=UTF8&psc=1

with two of these cables:

https://smile.amazon.com/gp/product/B000067SCD/ref=oh_aui_search_detailpage?ie=UTF8&psc=1

This allows me to optically isolate the ethernet from my modem from the rest of my network.

I don't use RS-232 surge protection, but something like this should do the trick:

https://smile.amazon.com/Powered-Serial-Electric-Isolator-protect/dp/B01EACRB0K/ref=sr_1_2?s=electronics&ie=UTF8&qid=1537494267&sr=1-2&keywords=rs-232+surge

I lost a controller this year to lightning along with some other electronics. My router and ethernet was fine, but I had to replace the generator controller and an outdoor security camera.

ToltingColtAcres, 2018-09-21:
I was looking at that DTECH unit earlier this evening.

I may pick up a few of these as well, to prevent any surges from traveling down the Ethernet line:

https://www.amazon.com/gp/product/B000BKUSS8

Lastly, I’ve been toying with the idea of buying one of these:

https://www.amazon.com/gp/product/B07DNWM62H

And just run it inside the generator, rather than positioning a Pi inside. It has an SMA connector for a wireless antenna, so with a small amount of drilling and a 3’ SMA cable, I could connect an external WiFi antenna to my generator to connect this.

Then I could communicate via Modbus TCP to the unit to retrieve data from the generator and run genmon anywhere on my network. Of course I’d have to fork the repo and change a lot of the code to do this... not sure I really want to go down that path... but I do think it would give the software a great deal more flexibility. If you have any interest, I’d be willing to buy a unit for you as well.

jgyates, 2018-09-21:
I will be happy to take a look at it. If you email me at generatormonitor.software@gmail.com I will send you my address.

jgyates, 2018-09-21:
One issue that may come up is the power for this converter. It looks like it is powered from a wall wart type power supply. the outlet that is powering the converter would need to be on during a power outage. As long as you are doing that it should be OK. There will need to be some changes to the software to support TCP or UDP unless you run one at each end.

ToltingColtAcres, 2018-09-21:
power is a barrel-style connector, 5-36vdc, so in principle it should run off the battery. So a serial connection to the evolution controller, a wifi antenna (either internal or external, depending on circumstances) and the power/battery connector would be the only things necessary.

I ordered one for myself, I'll drop you an email.

At work, I use a .NET modbus library to talk to my generators' registers... its a TCP or UDP connection to port 502. When my controller is repaired, I'm going to try it at home w/ one of these.

Having GenMon talk to "remote" generators via modbus-tcp/udp would be huge, IMO.

jgyates, 2018-10-04:
Serial over IP is supported, here are more details:

https://github.com/jgyates/genmon/wiki/Appendix-6----Serial-over-IP

closed 2018-10-04 by jgyates