jgyates/genmon #183: Can I run MobileLink Wifi and Genmon RS-232 at the same time?

jimland58, 2018-09-22:
Have a 22kw Generac (7043) with Evolution 2.0 Controller with MobileLink Wifi.  The MobileLink is plugged into the 8 pin Port 1 connector on the controller.   It appears the MobileLink uses pins 1,2,5,6 and the GenMon RS-232 connection is thru pins 7,8.   My question is can I have both the MobileLink and the GenMon connected at the same time and they will both work?

jgyates, 2018-09-22:
Hi @jimland58 

Since Evolution 2.0 is so new, I don't have a data point on compatibility. In short, I don't know that it works with 2.0, much less if it will work at the same time as MobileLink WiFi.  This thread has some limited 2.0 discussion (https://github.com/jgyates/genmon/issues/149), but you are the first person that I know of who actually has one installed that has commented in this forum.

If you do decide to try this let me know and I can help you.

jimland58, 2018-09-23:
Wanted to pass on the good news that it appears I am able to communicate with the Evolution 2 controller with your software.  I clicked on Submit Registers button to send you the registers.   On first glance, it appears the Status, Maintenance, Monitor, Notifications, Settings and Register View pages are working properly.  When I click on the Outage or Logs links, nothing happens.   I will play with is some more so I can learn more about it and then I will try and see if I can have MobileLink and GenMon working at the same time.

I will keep you updated on my progress and I will also update this progress and future progress on your Issues page.

Thanks and again, GREAT job on this Project!

Jim

From: jgyates 
Sent: Saturday, September 22, 2018 12:15 PM
To: jgyates/genmon 
Cc: jimland58 ; Mention 
Subject: Re: [jgyates/genmon] Can I run MobileLink Wifi and Genmon RS-232 at the same time? (#183)

Hi @jimland58

Since Evolution 2.0 is so new, I don't have a data point on compatibility. In short, I don't know that it works with 2.0, much less if it will work at the same time as MobileLink WiFi. This thread has some limited 2.0 discussion (#149), but you are the first person that I know of who actually has one installed that has commented in this forum.

If you do decide to try this let me know and I can help you.

—

---
This email has been checked for viruses by AVG.
https://www.avg.com

jgyates, 2018-09-23:
@jimland58 

Can you validate that your logs in the controller look like this:

    Alarm Log : 
        09/21/18 15:07:46 ESTOP Pressed : Alarm Code: 2800
        08/27/18 17:14:07 Low Battery : Alarm Code: 0000
        03/28/18 19:40:46 Low Battery : Alarm Code: 0000
        03/28/18 19:42:26 Low Battery : Alarm Code: 0000
        01/01/17 00:00:02 ESTOP Pressed : Alarm Code: 2800
        01/01/17 00:00:02 ESTOP Pressed : Alarm Code: 2800

    Service Log : 

    Start Stop Log : 
        09/23/18 08:45:02 Switched Off 
        09/22/18 09:17:04 Stopped - Auto 
        09/22/18 09:16:46 Switched Off 
        09/22/18 09:15:49 Running - Manual 
        09/21/18 15:09:41 Stopped - Auto 
        09/21/18 14:46:35 Switched Off 
        09/18/18 12:36:29 Stopped - Auto 
        09/18/18 12:30:42 Running - Exercise 
        09/15/18 16:37:12 Stopped - Auto 
        09/15/18 16:37:04 Switched Off 
        09/15/18 16:33:22 Running - Manual 
        09/15/18 16:33:14 Switched Off 
        09/14/18 14:26:08 Stopped - Auto 
        09/14/18 14:25:32 Switched Off 
        09/14/18 14:24:25 Running - Manual 
        09/14/18 14:24:17 Switched Off 
        09/11/18 12:36:22 Stopped - Auto 
        09/11/18 12:30:34 Running - Exercise 
        09/07/18 05:48:28 Stopped - Auto 
        09/07/18 05:48:22 Switched Off 
        09/04/18 12:36:14 Stopped - Auto 
        09/04/18 12:30:27 Running - Exercise 
        08/29/18 19:14:57 Stopped - Auto 
        08/29/18 19:13:09 Switched Off 
        08/28/18 12:36:06 Stopped - Auto 
        08/28/18 12:30:19 Running - Exercise 
        08/28/18 10:00:10 Stopped - Auto 
        08/28/18 09:39:28 Running - Utility Loss 
        08/28/18 09:28:44 Stopped - Auto 
        08/28/18 09:28:37 Switched Off 
        08/28/18 09:22:52 Running - Manual

jgyates, 2018-09-23:
Also, when you have a moment can you enter this URL

http://YOURIPADDRESS:8000/cmd/sendlogfiles

Replace YOURIPADDRESS with the IP address of your pi. This will send me your log files, there may be some info in there that will help..

jgyates, 2018-09-23:
I just checked in an update with some of the things I have found. You log and outage page may be a separate issue, will need to see your log files (see above for how to submit those). You can update the software from the Monitor page.

jimland58, 2018-09-23:
Sent it and this is what I get:  
Unknown Command sendlogfiles!

From: jgyates 
Sent: Sunday, September 23, 2018 10:42 AM
To: jgyates/genmon 
Cc: jimland58 ; Mention 
Subject: Re: [jgyates/genmon] Can I run MobileLink Wifi and Genmon RS-232 at the same time? (#183)

Also, when you have a moment can you enter this URL

http://YOURIPADDRESS:8000/cmd/sendlogfiles

Replace YOURIPADDRESS with the IP address of your pi. This will send me your log files, there may be some info in there that will help..

—

---
This email has been checked for viruses by AVG.
https://www.avg.com

jgyates, 2018-09-23:
Not for Evo 1.0 air cooled. Transfer switch status only shows up on Evo Liquid Cooled units. We will have to do some additional debugging to see if there are things that the 2.0 controller displays that the 1.0 does not.

Do the log and outage page show up?

jimland58, 2018-09-23:
Here is what my memory usage looks like:

Filesystem                 1K-blocks            Used                 Available         Use%         Mounted on
/dev/root                   28896496             4264572        23140984          16%                 /
devtmpfs                      468148               0                        468148               0%                 /dev
tmpfs                             472756               0                        472756               0%                 /dev/shm
tmpfs                             472756               12508                460248               3%                 /run
tmpfs                               5120                   4                          5116               1%                     /run/lock
tmpfs                             472756               0                        472756               0%                 /sys/fs/cgroup
/dev/mmcblk0p6         66528           21421                     45108              33%                 /boot
tmpfs                              94552               0                             94552               0%                 /run/user/1000
/dev/mmcblk0p5         30701             455                         27953               2%             /media/pi/SETTINGS

Doesn’t appear to be low on disk space?????

From: jgyates 
Sent: Sunday, September 23, 2018 11:22 AM
To: jgyates/genmon 
Cc: jimland58 ; Mention 
Subject: Re: [jgyates/genmon] Can I run MobileLink Wifi and Genmon RS-232 at the same time? (#183)

I got your log files. You system is showing some errors that I have seen before when the system is low on disk space. You can check your disk space by issueing the "df" command on the console:

 pi@genmon:~ $ df
 Filesystem     1K-blocks    Used Available Use% Mounted on
 /dev/root       29833596 1559960  27029140   6% /
 devtmpfs          495480       0    495480   0% /dev
 tmpfs             500088       0    500088   0% /dev/shm
 tmpfs             500088   31620    468468   7% /run
 tmpfs               5120       4      5116   1% /run/lock
 tmpfs             500088       0    500088   0% /sys/fs/cgroup
 /dev/mmcblk0p1     43539   22500     21039  52% /boot
 tmpfs             100016       0    100016   0% /run/user/1000
The above is what my system shows when I type "df". If the % use is very hight then you may need to free up disk space or get a larger SD card. Let me know how the test goes.

—

---
This email has been checked for viruses by AVG.
https://www.avg.com

jimland58, 2018-09-23:
No, with the updated GenMon software V1.10.4, the Outage and Logs page still don’t work.

From: jgyates 
Sent: Sunday, September 23, 2018 11:27 AM
To: jgyates/genmon 
Cc: jimland58 ; Mention 
Subject: Re: [jgyates/genmon] Can I run MobileLink Wifi and Genmon RS-232 at the same time? (#183)

Not for Evo 1.0 air cooled. Transfer switch status only shows up on Evo Liquid Cooled units. We will have to do some additional debugging to see if there are things that the 2.0 controller displays that the 1.0 does not.

Do the log and outage page show up?

—

---
This email has been checked for viruses by AVG.
https://www.avg.com

jgyates, 2018-09-23:
ok. Let's try this from the genmon directory:

     sudo bash ./genmonmaint.sh install 

This will reinstall all of the libraries needed. You can select No when prompted to copy the config files or make genmon start at boot if you have already done this.

Once that is finished then type:

     sudo bash ./genmonmaint.sh updatelibs

this will update the libraries to the latest version.

Then restart the software:

     sudo bash ./startgenmon.sh restart

Then go the to web interface and see if you have logs or an outage page.

jimland58, 2018-09-23:
On the PuTTY console, I’m getting these type messages coming across every second or two:

192.168.1.132 - - [23/Sep/2018 11:36:46] "GET /cmd/registers_json HTTP/1.1" 200 -
192.168.1.132 - - [23/Sep/2018 11:36:48] "GET /cmd/gui_status_json HTTP/1.1" 200 -
192.168.1.132 - - [23/Sep/2018 11:36:51] "GET /cmd/gui_status_json HTTP/1.1" 200 -
192.168.1.132 - - [23/Sep/2018 11:36:51] "GET /cmd/monitor_json HTTP/1.1" 200 -
192.168.1.132 - - [23/Sep/2018 11:36:51] "GET /cmd/registers_json HTTP/1.1" 200 -
192.168.1.132 - - [23/Sep/2018 11:36:54] "GET /cmd/gui_status_json HTTP/1.1" 200 -

Is there a way to stop and start this logging to the console?

BTW,  I was having a problem with losing the wifi connection to the Pi3 when there was inactivity (just several minutes) on the wifi.   Reading thru forums on the Pi it seems others have seen this.  As long as GenMon is running and it’s sending data continuously thru the wifi port is hasn't stopped working yet.  Any suggestions on how to fix this? 

Thanks,
Jim 

---
This email has been checked for viruses by AVG.
https://www.avg.com

jimland58, 2018-09-23:
IT WORKS!!!!!!!   THANKS!!!!!

Did want to pass on when I ran the sudo bash ./genmonmaint.sh updatelibs  command I did see a few errors come across the screen.  Here they are:

Compiling /tmp/pip-build-3a79F9/Jinja2/jinja2/asyncfilters.py ...
  File "/tmp/pip-build-3a79F9/Jinja2/jinja2/asyncfilters.py", line 7
    async def auto_to_seq(value):
            ^
SyntaxError: invalid syntax

Compiling /tmp/pip-build-3a79F9/Jinja2/jinja2/asyncsupport.py ...
  File "/tmp/pip-build-3a79F9/Jinja2/jinja2/asyncsupport.py", line 22
    async def concat_async(async_gen):
            ^
SyntaxError: invalid syntax

  Found existing installation: Werkzeug 0.9.6

Running setup.py install for MarkupSafe
  building 'markupsafe._speedups' extension
  arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-security -fPIC -I/usr/include/python2.7 -c markupsafe/_speedups.c -o build/temp.linux-armv7l-2.7/markupsafe/_speedups.o
  arm-linux-gnueabihf-gcc -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-z,relro -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-security -Wl,-z,relro -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-security build/temp.linux-armv7l-2.7/markupsafe/_speedups.o -o build/lib.linux-armv7l-2.7/markupsafe/_speedups.so
Could not find .egg-info directory in install record for MarkupSafe>=0.23 from https://files.pythonhosted.org/packages/4d/de/32d741db316d8fdb7680822dd37001ef7a448255de9699ab4bfcbdf4172b/MarkupSafe-1.0.tar.gz#sha256=a6be69091dac236ea9c6bc7d012beab42010fa914c459791d627dad4910eb665 in /usr/lib/python2.7/dist-packages (from Jinja2>=2.10->Flask)

From: jgyates 
Sent: Sunday, September 23, 2018 11:37 AM
To: jgyates/genmon 
Cc: jimland58 ; Mention 
Subject: Re: [jgyates/genmon] Can I run MobileLink Wifi and Genmon RS-232 at the same time? (#183)

ok. Let's try this from the genmon directory:

 sudo bash ./genmonmaint.sh install 
This will reinstall all of the libraries needed. You can select No when prompted to copy the config files or make genmon start at boot if you have already done this.

Once that is finished then type:

 sudo bash ./genmonmaint.sh updatelibs
this will update the libraries to the latest version.

Then restart the software:

 sudo bash ./startgenmon.sh restart
Then go the to web interface and see if you have logs or an outage page.

—

---
This email has been checked for viruses by AVG.
https://www.avg.com

jgyates, 2018-09-23:
You can check the file /var/log/genserv.log to see if you have any errors. Werkzeug is related to flask so if you see any odd behavior with the web interface we an look into it, but otherwise I would ignore the errors on updatelibs.

One thing we ca do to make sure we don't have any other outstanding issues is this:

Let it run for a few minutes, and click on every web page, then send your log files again.

On the Maintenance page you should see some info about your generator, if you want to validate this info is correct, that would be helpful. Also making sure your logs displayed in the software match the test in the logs that you view from the controller screen would be useful also. These things will let me know if there are any issue that are not causing errors ,but report incorrect data.

jgyates, 2018-09-23:
you latest logs look good. Only one issue related to your incoming mail settings (see /var/log/mymail.log). This is likely due to the fact that you have an email folder in your settings that does not exist on your inbox. If you don't want to use email to send commands to the generator you can disable incoming email on the settings page and this error will go away. If you want this functionality you can create a mail folder named Generator and another named Generator/Processed (these are the default values, but you can change the names if you like) .

Let me know if you have a chance to validate the info on the web interface with the data from the controller screen.

jimland58, 2018-09-23:
Maintenance Page:

Generator Menu vs GenMon,  everything matched except for:

GenMon shows a Service A at 199 hrs or 8/26/2020 and Service B at 399 hrs or 8/26/2022
The Generator Menu under Sub-Menu/Maint shows only 1 service schedule that says Next Maint 199 RNHRS on 8/27/2019 and then shows Inspect Battery 199 RNHRS on 8/27/2019.  So it matches the Service A except for the date.  
If I try scrolling up or down I don’t see anything else so it doesn’t show a Service B, 399 hrs schedule 
Also, on GenMon the Total Run Hours shows 0 and Generator Menu shows Run Hrs as 0.8 so I’m sure it’s just a round off in GenMon

Log Page:

GenMon Alarm Log matched the Generator Alarm Log exactly except for the description on Alarm Code: 2800.  GenMon Alarm Log says “ESTOP Pressed” and Generator Alarm Log  say “Auxiliary Shutdown”.  

Service Log matched

GenMon Start Stop Log matched the Generator Run Log exactly except for the descriptions of the logs.  GenMon calls it “Start Stop Log” and the Generator Menu calls it “Run Log”

Other
Also noticed a couple other data points as I was flipping thru the Generator Menu that you might consider including.  
PROTECTION HOURS: 643
CHARGING STATUS:  CHARGING
UTIL RECOVERY VOLTS: 190.
AMBIENT TEMPERATURE: 98

From: jgyates 
Sent: Sunday, September 23, 2018 11:57 AM
To: jgyates/genmon 
Cc: jimland58 ; Mention 
Subject: Re: [jgyates/genmon] Can I run MobileLink Wifi and Genmon RS-232 at the same time? (#183)

You can check the file /var/log/genserv.log to see if you have any errors. Werkzeug is related to flask so if you see any odd behavior with the web interface we an look into it, but otherwise I would ignore the errors on updatelibs.

One thing we ca do to make sure we don't have any other outstanding issues is this:

Let it run for a few minutes, and click on every web page, then send your log files again.

On the Maintenance page you should see some info about your generator, if you want to validate this info is correct, that would be helpful. Also making sure your logs displayed in the software match the test in the logs that you view from the controller screen would be useful also. These things will let me know if there are any issue that are not causing errors ,but report incorrect data.

—

---
This email has been checked for viruses by AVG.
https://www.avg.com

jimland58, 2018-09-23:
I decided at this time to disable the Incoming Email Commands Processing so I turned if off on the Setting Page.  Thanks!

From: jgyates 
Sent: Sunday, September 23, 2018 12:33 PM
To: jgyates/genmon 
Cc: jimland58 ; Mention 
Subject: Re: [jgyates/genmon] Can I run MobileLink Wifi and Genmon RS-232 at the same time? (#183)

you latest logs look good. Only one issue related to your incoming mail settings (see /var/log/mymail.log). This is likely due to the fact that you have an email folder in your settings that does not exist on your inbox. If you don't want to use email to send commands to the generator you can disable incoming email on the settings page and this error will go away. If you want this functionality you can create a mail folder named Generator and another named Generator/Processed (these are the default values, but you can change the names if you like) .

Let me know if you have a chance to validate the info on the web interface with the data from the controller screen.

—

---
This email has been checked for viruses by AVG.
https://www.avg.com

jgyates, 2018-09-23:
Can you look on the registers page at register 0033, it should be labeled "Unknown Sensor (EvoAC)". Does this value ever change (you can see the history of the registers, but the history is only kept if the web interface is open, you don't have to be on the registers page). Does this decimal value correspond to the ambient temp?

Regarding Hours of Protection, Ambient Temp, Utility Recovery volts, and Charging Status, I have attempted to look for these values before on the Evolution Air Cooled 1.0 controller and I was unable to find them. I have found what the controller exports via modbus varies depending on the controller / firmware variants. The different variants are : Nexus Air Cooled, Nexus Liquid Cooled, Evolution Air Cooled, Evolution Liquid Cooled, and now Evolution 2.0 Air Cooled. To data, Evolution Liquid Cooled exports the most data, but it looks like Evolution 2.0 may be a the one with the second most exported data.

I was able to find the UTIL RECOVERY VOLTS and will post this update shortly, It is labeled Pickup Voltage on the outage page (again some controllers call this pickup voltage, others call it Utility Recovery Voltage, but is is the same thing, the voltage that the controller should assume the utility voltage is restored from a previous outage). It looks like it is in the same modbus register as the Evolution Liquid Cooled Controller.  The Hours of Protection is exported by the Evo Liquid Cooled, but at first glance I don't see it in the Evo 2.0.

The data of service due in the other controllers, I had to add one day to the value read from the controller to get it to match up, looks like I don't have to do that with Evo 2.0. This fix will be checked in shortly also.

I have changed the name of the "Start / Stop Log" to the "Run Log". This is just what I called it, but it should probably match the naming convention used by the controller to be consistent. This change will be checked in shortly.

Regarding the "ESTOP Pressed" vs "Auxiliary Shutdown", this is the error text for error code 2800, it all means the same thing, some generac generators call it Emergency Stop and but the Evo 2.0 looks like it is callin it Auxillary Stop. Unless you have an emergency stop button you will likely not see this again (some odd error messages can show up during the generator installation depending on how the controller is powered up so the instance you have in your log now you should be able to ignore). Short story, I will likely leave this labeled that way for now.

Regarding the Engine Run Hours. The Evolution Controller supports reading run hours and run minutes via the modbus. At first glance run minutes does not appear to be exported on Evo 2.0.

Having said all of this, it looks like we have the basic support up and running on Evo 2.0, however we can take this a step farther if you are willing to do some simple tests on your end. These test involve editing /etc/genmon.conf and making this change:

Change this:
    
    enabledebug = False

To 

    enabledebug = True

this will enable an additional thread in genmon that will read more registers and look for values that change. These values are stored internally and emailed to me. Once the test runs for around 30 min the debug thread will sleep doing nothing. We need to run this test when the system is idle, and again when the generator is running. It does not have to run the engine time, about 10 min should do it.

Another thing we can to that may help determine if the battery charging status is exported is to leave the web interface up in your browser for 24 hours. The register page will color registers that have changed in red (recent changes are deep red , and the red changes to a light red as time passes, black denotes no change). If you hover your mouse over the register you can see a graph of the register over a 10 min, 1 hour and 24 hour range). You can also export the register graphs to a PDF. You can email these to me after it has ran for 24 hours. In the Evolution Liquid Cooled Controller the status of the relay that controller the battery charger is exported via a modbus register, it is not exported in Evo 1.0 Air Cooled. The battery charging cycle last about 18 hours and restarts the 18 hour count if the generator starts so this is a test we would need to run over a period of time that the generator is not going to be operating.

Let me know if you have any questions.

jgyates, 2018-09-23:
I checked in one final change. If you enable the "Display Unknown Sensors" on the settings page, a new section of data will show up on the Status page titled "Unsupported Sensors". this is where I display values that I am not sure about. There is an alternate method for determining the battery charging status tha works on Evo LC, but not others that I have found. It monitors the current transfomer for the battery charger to determine if the charger is active. I put this and two candidates for ambient temp in the "Unsupported Sensors" section for Evo 2.0 controllers.

jimland58, 2018-09-23:
Thanks for all the work you are doing to make this work for the Evolution 2.0 Controller!

Here’s feedback on your previous questions/comments:

- Looked at register 0033 and it is at Decimal 100 and looked at the history and it hasn’t changed.  I believe the ambient temperature is actually on register 05ed, the same location where the EvoLC ambient temperature is located.  I looked at the history on this register and it is moving with the ambient and matches exactly what I read on the Dealer Menu a few minutes ago which is 92 degrees.

- With the latest update to GenMon V1.10.15, I do see the Utility Pickup Voltage: 190.

- The Hours of Protection around 6 pm CST today was 649 hours so if you are trying to find the register look for something around that number and climbing.   

-  The Date of Service A is not only off by a day but also the year was incorrect.   GenMon showed 8/26/2020 and the Generator Menu showed 8/27/2019

-  Regarding Alarm Code 2800,  my unit has two On/Off type switches that are used as emergency shutdown switches.  One is located inside the unit and the other on the outside and both are labeled as Auxiliary Shutdown and either of these switches appears to be what triggers this alarm code.

- I’ve set the enabledebug to True and enabled Automated Feedback on the Settings page.

- For the Battery Charging, I do see that register 05ee, labeled Battery Charger CT (EvoLC) is red and has been changing.

- I have enabled the Display Unknown Sensors on the Settings Page.

From: jgyates 
Sent: Sunday, September 23, 2018 5:41 PM
To: jgyates/genmon 
Cc: jimland58 ; Mention 
Subject: Re: [jgyates/genmon] Can I run MobileLink Wifi and Genmon RS-232 at the same time? (#183)

I checked in one final change. If you enable the "Display Unknown Sensors" on the settings page, a new section of data will show up on the Status page titled "Unsupported Sensors". this is where I display values that I am not sure about. There is an alternate method for determining the battery charging status tha works on Evo LC, but not others that I have found. It monitors the current transfomer for the battery charger to determine if the charger is active. I put this and two candidates for ambient temp in the "Unsupported Sensors" section for Evo 2.0 controllers.

—

---
This email has been checked for viruses by AVG.
https://www.avg.com

jimland58, 2018-09-23:
Once you are ready for me to run the generator to gather more data, let me know if I need to do anything before starting it.
Thanks,
Jim

---
This email has been checked for viruses by AVG.
https://www.avg.com

jgyates, 2018-09-23:
ok. I will. I assume the Ambient Temp is in Farenheit?

jimland58, 2018-09-24:
Yes, it is in Farenheit

jgyates, 2018-09-24:
I have enough data now for the case where the generator is not running. It does not look like the hours of protection is exported via modbus. When debug is enabled, software polls the first 8000 mod bus registers looking for changes. Nothing looks like a value incrementing once every hour.

Give me a day to digest the info I have and then we can try the debug test again with the generator running. For now you can set enabledebug = False in /etc/genmon.conf and restart.

jimland58, 2018-09-24:
Thanks for checking for the Hours of Protection register.  Not a big deal not knowing this one.

I've disabled the debug so I'll report back tomorrow and see how it goes over 24 hrs.

jimland58, 2018-09-24:
I just saw where the webpage stopped working and when I ran genloader is said genmon was already running but started genserv.  Genserv log file showed this error:

2018-09-24 15:31:30,230 : Error in app.run: [Errno 5] Input/output error : genserv.py:896

jgyates, 2018-09-24:
OK. I will add this to the list. This is related to the Flask library not shutting down properly. The longer term fix is to make flask use apache, but I have not gotten around to that yet. Using flask does have it's limitations when used in a stand alone web server, but the plus side it it is very simple to setup and reliable once it is working.

jimland58, 2018-09-25:
Updated to V1.10.7 and everything seems to be working as configured with my 22kW Generac generator with the Evolution 2.0 controller.  It did it's weekly exercise today for 5 minutes and I got all the text messages reporting this happening and I also saw the Voltage, Frequency and RPM indicators on the Status webpage responding.

Thanks for developing, sharing and providing outstanding support on this fantastic project!

jimland58, 2018-09-30:
More good news!  I connected the MobileLink WiFi and Gemon which both use the same 8 pin port on the Evolution 2.0 controller (different PIN numbers) and they both work at the same time. 

Next is to test Genmon while performing a Utility Undervoltage transfer to  check that it reads all real time values from the generator (I.e. volts, frequency, kW, rpm’s, etc.)

jgyates, 2018-09-30:
@jimland58 

That is good new. Thanks for the update on Evo 2.0. Let me know how the tests go.

jgyates, 2018-10-04:
Closing this issue, feel free to comment here if you have any additional data.

closed 2018-10-04 by jgyates

jonathanpisarczyk, 2019-01-26:
> More good news! I connected the MobileLink WiFi and Gemon which both use the same 8 pin port on the Evolution 2.0 controller (different PIN numbers) and they both work at the same time.
>
> Next is to [...]

Hey @jimland58 can you send a picture of how you connected both MobileLink and Genmon?  I'd like to do the same thing.

waynewarburton, 2019-12-06:
I'm just fabricating a Raspberry Pi system for my 20KW Generac ad came across this string.
To clarify, "MobileLink uses pins 1,2,5,6 and the GenMon RS-232 connection is thru pins 7,8".  Using this information with the Evolution 2.0 Controller, we could just cut the existing manufacturer's wires connected to terminals #7 and #8 and extend/reroute them to our serial port (#2 and #3 on the RS232)?

jgyates, 2019-12-06:
Yes, as long as you also use pin 2 (ground). This page has the pinout: https://github.com/jgyates/genmon/wiki/3.1--Making-a-Cable

waynewarburton, 2019-12-06:
If I'm using the generator's battery to power the Pi, should be able to get away without using the pin (2) gnd?

jgyates, 2019-12-06:
I do not know for sure, but I would guess no. The answer depends on how the USB battery power supply is wired internally (are the two grounds isolated across the internal transformer?) and how the controllers RS-232 circuit is designed (is the RS-232 isolated from the battery ground?). Since I do not know the answer to these two questions I use the ground on the controller and power the pi from the battery.  In short, you could try, but if you see comm errors then you then add the ground from the connector.

waynewarburton, 2019-12-06:
Thanks for the speedy response!

deh2025, 2024-02-09:
I am thinking of doing same on new 26kw Generac, allowing Mobile Link and Genmon to run simultaneously.  I am planning to use a Pi 3B+ in conjunction with a HAT and Molex cable fabricated by PintSize.Me. Shoutout to PintSize.Me for their excellent support!  I am  considering remoting the Pi in the house given the generator's close proximity to house.  I have both power and signal at the location power enters the house.  If I understand the previous threads correctly, I should be able to port signal and ground from the generator to the Pi via DB-9 to RS-232 to RS-442 then back RS-442 to RS-232 to TTL via a Cat 5e or better shielded cable.  Unless there is a simpler method is available.  Any input would be greatly appreciated!

skipfire, 2024-02-09:
If you are looking for an off-the-shelf option I (PintSize) also sell an RJ45 Splitter/Adapter designed specifically for the use case you describe. It lets you use Ethernet cables to do distance runs, but to be clear it is not a network device, just uses network cables.  It functions as a splitter to let you keep MobileLink and also adapts the 8-pin Molex to 8-pin RJ45 and back again.  https://pintsize.me/store/ols/products/generator-splitterinjector/v/GSI-RJ4-SPL

deh2025, 2024-02-09:
Hi Jon,

Thank you for reaching out. You have been an excellent resource.

So that I am clear, I am attempting route the signal to both Mobile Link
and the Pi/Genmon. I had not realized until this morning that the cable
included in yesterday's purchase is actually a splitter, which was great
news.

My preference is to remove the Pi and HAT from the harshest environmental
conditions and place them in my lower garage where power enters the house
and there is also allow for access to an eero Pro for an ethernet
connection. A simple and eloquent solution would be to simply run a
Cat5e/6e between the generator and the Pi/HAT. If the product in the link
included your email facilitates this configuration - - I am all in and will
follow-up with another order. Please let me know if there is any
misunderstanding on my part.

Thank you, Dean

skipfire, 2024-02-09:
@deh2025 the RJ45 splitter/adapter has a splitter that goes in the generator, similar to the one you have except the GenMon side has an RJ45 plug instead of a Molex plug.  This would keep MobileLink at your generator (with the installs I've seen it's not easily removed or leaves a big hole if it is).  Then it also comes with an RJ45 ejector that has an RJ45 port on one end and an 8-pin Molex on the other to get it back to the Molex that the HAT accepts.  The RJ45 Splitter/Adapter is expected to have power on the other end and should not be powered over the Molex in this case due to the extra power loss from the longer wire run.

I also have 3 units of a HAT designed for use away from the generator that includes an opto-isolator to completely segregate the electrical of the generator from the electrical of the Pi.  The HAT is quite a bit different and hasn't had a full production run because I haven't seen interest in the product, it has very different fan support, no 1-Wire temperature support, and has a single channel analog input (such as for a tank monitor) tied to an ADC1115 chip.  If you are interested in swapping out what you just got feel free to contact me directly and we can work something out.

k5map, 2024-05-18:
Has anyone in the SW or South experienced a heat issue with the RPi being installed inside the generator enclosure?

skipfire, 2024-05-18:
@k5map I'm outside New Orleans and have no problem with mine inside or attached to the outside the enclosure.  The only time it overheated was after Hurricane Ida when shut down after 6 days and then the lack of airflow through the engine let the radiant heat of the engine spike the air temp, opened the lid for an hour and it was fine.

k5map, 2024-05-18:
@skipfire... ok that's great news... one additional question; do you have your RPi in case or just mounted with open air flow?

skipfire, 2024-05-18:
When inside the enclosure I had an acrylic base plate with magnets just holding it onto a metal plate under the controller.  Right now I have a plastic electrical box on the back of the generator to facilitate easy access when I need to test something.  The enclosure is a big faraday cage so moving it on the outside gained about 20db of WiFi strength.

k5map, 2024-05-18:
@skipfire... much thanks again for answering my questions.