jgyates/genmon #184: Options for sending SMS Text Messages to Cell Phone

jimland58, 2018-09-24:
I want to send SMS text messages to my cell phone when the generator changes state.  I set up a trial Twilio account and used the gensms program to do this and it works great.  However, after the trial account on Twilio expires, I will have to pay $1/month + $0.0075 per text message sent.  Why couldn't I send my text messages free by emailing my text messages using the mymail program.  It seems I just need to set up the email_recipient to be "xxxxxxxxxx@txt.att.net" with xxxxxxxxxx being my cell phone # to send messages to my ATT cell phone.   Won't that work?

jimland58, 2018-09-24:
I should state that I don't need the capability to email remote commands to the generator (have that disabled) so don't need to the capability to email anything back to the genmon program.

shpegel, 2018-09-24:
Al phone providers I've used usually provide a way to send SMS text messages via email. In other words, you send an email and it ends up as a text on your phone. You will have to look it up for your provider as there is no one way to do this. But all are similar in that they simply use some variation of your phone number and the phone provider  email to SMS server. These often  work in reverse too which is kind of nice.