jgyates/genmon #186: Secure Internet Access to Genmon Webpage

jimland58, 2018-09-25:
I have setup my router for port forwarding to Port 8000 on my Raspberry Pi running Genmon so I could access the Genmon server webpage from the Internet.  However, I'm concerned with the security of this Internet Connection.  Is there a way to require a logon access to get to the webpage or at a minimum, hide or prevent any setting changes from the webpage when reaching it from the WAN?

liltux, 2018-09-25:
I wouldn't use port forwarding.  I use a VPN to access things at home. Openvpn works great on raspberry pi.  What I use is a VPN server I subscribe to and control on the internet.  All my devices connect securely to the server.  Then all devices connected to server can access each other, securely.  There is a login page and security settings on genmon too, but I have yet to setup.

jgyates, 2018-09-25:
I agree with @liltux , a VPN is a much more secure solution. 

The "Optional - Webserver Security Settings" will ebable HTTPS and once that is enabled you can set a user name and password. Enabling HTTPS will change the port from 8000 to the HTTPS standard of 443.

nah2323, 2018-11-17:
I don’t know if I would call this specifically related, however I set up admin and user logins to the genmon web interface and now I can’t get the web interface to load.  :8000 or :443.  Help?