jgyates/genmon #187: Web page is very slow generating- or non- existent.

jcherney, 2018-09-26:
Since  last 2 updates (and I’m not suggesting it was the updates that caused this), my system is taking an extraordinarily long time to generate the web page. Sometimes it never generates. PI3 temps seems to be reasonable (prox 115f) and CPU is under 3% utilization. Wi-Fi hasn’t changed strength. 

Once it does generate everything seems fine.

Any thoughts or suggestion as to why this is happening? PI3 going bad? microsd going bad? Operator going bad?!!!

Thanks.

jgyates, 2018-09-26:
Could be network related? If your wifi or router has high traffic it could cause this. In the last two updates, nothing related to the web interface has been updated so I am not sure why this is happening on your system. My setup is working the same. You could check your logs to see if you see any errors.

Your wifi performance is related to signal strength, signal quality, and traffic load on your network.

jcherney, 2018-09-26:
Very little traffic at all. I'd say average or below. I'll check the logs

jcherney, 2018-09-26:
Please take a look at my closed issue #178. It looks like it started during the problem I was having upgrading. Oddly coincidental.

jgyates, 2018-09-26:
Another note: your web performance is also a factor of the system running the web browser.

If you are on a Pi3, your serial status should look like this:
[image]

jcherney, 2018-09-26:
No errors found. And yes, when it does load, it looks just like that.

jgyates, 2018-09-26:
Have you tried this on multiple systems? Is the behavior consistent across multiple systems and browsers?

jcherney, 2018-09-26:
Pings bounce between 5 and 23 ms. I'll try looking at the restart stats and see what I can find. I'll reload my browsers when I can clean up the tabs..! Thanks.

jcherney, 2018-09-26:
Broken Pipe?

[image]

jgyates, 2018-09-26:
Looks like you found the problem. This is a trace back from an exception within the python SocketServer library. Flask uses this internally, so my guess is that there is something not correct with your libraries. There are two options, start digging in your libraries and try to find the problem, or start with a clean install of the operating system, then reinstall the software. If you choose the latter, use the install scripts to setup everything. 

Install raspbian, enable ssh, set the time zone / keyboard layout / wifi settings., then

From the /home/pi directory:

      sudo apt-get update
      sudo apt-get dist-upgrade 
      sudo apt-get install git .  
      git clone https://github.com/jgyates/genmon.git
      cd genmon
       bash ./genmonmaint.sh install
       cd OtherApps
       sudo python serialconfig.py -e

Then restart with your cables connected.

jcherney, 2018-09-26:
Is it possible to just overwrite the libraries?

jcherney, 2018-09-27:
Hi- I used a backup image of my system and reloaded it. The webpage generates but there is no data presented. I've run serialtest.py and it comes up clean. But when the webpate monitor is accessed, it shows no packets being received. Not sure where to go from here (or if I should start another thread for this).

jcherney, 2018-09-27:
[image]

jcherney, 2018-09-27:
[image]

jcherney, 2018-09-27:
Sorry- Loopback cleared ok, not serialtest

[image]

jgyates, 2018-09-27:
I am not sure what you mean by " Loopback cleared ok, not serialtest".  serialtest.py is a loopback test. Does your serialtest.py work with the transmit and receive looped back. Please review this page for the sequence to troubleshoot serial problems:

https://github.com/jgyates/genmon/wiki/Appendix-6---Serial-Troubleshooting

jgyates, 2018-10-04:
I assume this issue is resolved. Feel free to comment in this thread if it is not.

closed 2018-10-04 by jgyates