jgyates/genmon #188: install fail on 2 libraries

liltux, 2018-09-28:
@jgyates 
I was performing a fresh install, using: "sudo bash genmonmaint.sh install".
Goes through all prompts works fine until it gets to two program installs with pip.
sudo pip install chump
sudo pip install paho-mqtt
I am not using mqtt and do not believe I require chump to do the basic monitoring. 
This was on a rpi 2b board.  Both had install/compile fails.  Nothing else failed.

jgyates, 2018-09-28:
That is odd, never seen that before. You are correct that you don't need those to load genmon or genserv. So you should be fine.

There is code in genloader that tries to load missing libraries when loading, but it will will not prevent anything from working with genmon / genserv, it might just dealy your starting by a few seconds.

Is this on raspbian 9? I wonder why it is failing. What was the reason given by pip (you should be able to type 

sudo pip install chump
sudo pip install paho-mqtt

on the command line to reproduce.

liltux, 2018-09-29:
that is some good thought...
I know i am running the 6-27-18 image, but i have not done a dist-upgrade recently if that maybe causing and issue.  Sorry I did not save the errors, and I will not be back at the office till monday.  I did however just did a rpi3 over the weekend and earlier this week and it did fine no errors, but it was a fresh image and install.  I will try a few things and report on monday.

jgyates, 2018-10-04:
Going to close this for now, feel free to comment if you have any other data.

closed 2018-10-04 by jgyates