jgyates/genmon #189: Live Register View Errors

shpegel, 2018-09-30:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

If you need to send you logs to the developer, if you email is setup and working properly you can issue this command in your browser:

     http://MYIPADDRESS:8000/cmd/sendlogfiles

Replace MYIPADDRESS to the IP address of you pi and port 8000 if you are using a port other than the default.

You can submit your registers from the Monitor page on the web interface.

### Expected Behavior

No Items flagged red in register view

### Actual Behavior

4 items flagged red in register view, utility voltage, battery voltage, generator time, battery charger CT

### Steps to Reproduce (including precondition)

Continuous. Not sure when this started as never checked register view before two versions ago. At that time version upgrade caused red banner error at bottom of screen. However, no apparent reason for error readouts as all flagged items are working as expected and no errors flagged locally at generator. Red banner would not clear until I tuned on Auto Feedback in Settings page and saved. Has not returned since even if I turn off Auto Feedback. 

### Screenshot or Pictures relating to the problem (if possible)

[image]
{Please write here}

### Your Environment
[Generator Monitor.pdf](https://github.com/jgyates/genmon/files/2431574/Generator.Monitor.pdf)

- Genereator Model: G0070370 (2016) Hardware V1.06, Firmware V1.20
[Generator Monitor.pdf](https://github.com/jgyates/genmon/files/2431571/Generator.Monitor.pdf)

- Generator Registers: Submitted
- Genmon Version: 1.10.9

jgyates, 2018-09-30:
This is not an error, it is the normal operation.

Normal unchanged registers are grey background with black letters. In the live register view, when a register changes, it is marked red with black letters. The changed registers will change color over time 
 to grey background with a red letters. The purpose of this is to let you know what changed recently.

You should notice your generator time register (HH:MM for hours and minute) change every minute.

closed 2018-09-30 by jgyates

shpegel, 2018-09-30:
Jay,

Thank you for the support. Never had occasion to look at the Live register
screen previously. Only looked at it after bottom banner turned red for no
apparent reason after an update to Genmon. As mentioned previously, that
issue is now gone.

Many thanks,

Howard

jgyates, 2018-09-30:
No worries, thanks for the feedback.