jgyates/genmon #190: Non-Issue Generator Support Question

liltux, 2018-10-05:
Has anyone set up a genmon on a PowerPact unit??  I just learned that they are supported by MobilLink, so there should be a serial communication port.  @jgyates, would you be interested in adding support for another controller type?  The PowerPact is a bare minimum controller.  I had someone ask if it is supported on the ebay page for a prebuilt component.  I don't have a PowerPact at my disposal to send you register information or log files but maybe someone on here might see this post and be interested in giving it a shot.

jgyates, 2018-10-05:
Sure, I would be happy to add support for the PowerPact

liltux, 2018-10-09:
@jgyates, I have recieved pictures of the cable connecting the mobillink to the powerpact unit.  It appears it still uses the same 8-pin molex and pins 7 and 8 for serial communication.
[image]
[image]

jgyates, 2018-11-05:
PowerPact is now supported

closed 2018-11-05 by jgyates