jgyates/genmon #191: Serial fails, suspect lack of ground - using cell phone 110VAC charger for power

wotseurba, 2018-10-07:
New hardware and Genmon installation.  I have located the Pi indoors, and pulled a shielded twisted pair communications cable to the generator, and to the Molex connection.  Powering the pi from a standard wall outlet power supply.  

serial loopback test is fine.  However running serial test through to the generator controller fails.

I see that a ground from the generator is recommended, which I assume by using the motorcycle power converter brings the ground to the usb power connection of the pi.

Given I am using a standard wall power supply (2-prong w/o ground) how do you recommend that I tie the generator ground to the pi?

jgyates, 2018-10-07:
If you use a wall wart type charger there is not common ground so you will need to connect the ground from the controller molex to the ground on the rs-232 connector or on the ground on the pi expansion header.

the motorcycle power converter makes the pi on the same ground as the battery (which is the same ground as the controller). Powering from the molex connector puts the pi on the same ground as the controller. In short you have to connect the ground from the generator to the ground on the pi or the RS-232 signal levels will not have the same reference, so +/-12V (RS-232 voltage levels) on one system will appear as some unknown voltage (not +/-12V) to the other system.

wotseurba, 2018-10-07:
Wow.  Thanks for the speedy reply.  Which pin on the Molex at the controller is ground?  And what pin on the DB-9 RS232 is ground?  I have extra conductors on my communications cable, so inter connection is easy.

jgyates, 2018-10-07:
pin 5 is typically ground on a DB-9, pin 2 is ground on the molex. A diagram of the molex is here:

https://github.com/jgyates/genmon/wiki/3.1--Making-a-Cable

wotseurba, 2018-10-07:
No dice.   brought pin 2 from controller Molex to pin 5 of DB-9

pi@raspberrypi:~/Genmon/OtherApps $ sudo python serialtest.py

Loopback testing for serial port /dev/serial0...

Serial port opened
write data: sent test string
waiting to received data....
FAILED: Sent data does not match receive. Received 0 bytes

As I previously said, a loopback of pin 2 - pin 3 of DB-9 is successful.

I also tried swapping Tx & Rx same failure.

Stumped

wotseurba, 2018-10-07:
Perhaps I've made a wrong assumption that the serialtest when connected to the generator controller will function and test communication.  Is this a correct assumption?

I performed a loopback test at the DB-9 with success.  I then figured I should repeat the test after being connected to the generator controller, before moving forward.

Genmon is loaded, although I need to copy mymail.txt to /etc before I move forward.

I ran serialconfig -c to be sure everything is configured properly.

pi@raspberrypi:~/Genmon/OtherApps $ sudo python serialconfig.py -c
Checking : BT service disabled: OK
Checking : Serial console command line removed: OK
Checking : Is enable UART in boot config: OK
Checking : Is BT overlay disabled: OK
Checking : Serial console service disabled: OK

Serial port settings are OK. A reboot is needed if changes were made.

liltux, 2018-10-07:
When I have comm issues I start at the pins on board, then at the two wires that connect to my ttl to 232 converter, then at the output of the 232, and finally at the molex connector pins 7 and 8.  If all loopback are successful I then plug in to controller.  If no communication then, I then swap my tx and rx wires.  Also if you have a multi meter or dvom, and with everything de-energized you can unplug your ground wire at the pi and measure continuity between the ground pin in molex and to your pi ground wire to verify a good ground.

jgyates, 2018-10-07:
Running serialtest while connected to the controller is expected to fail.
Now that you have validated the serialtest.py test, does the generator monitor software communicate with the controller? To validate, connect to you controller then try to pull up the web page once the software has loaded.

jgyates, 2018-10-08:
The output of the web server looks like this:

192.168.11.13 - - [07/Oct/2018 21:12:52] "GET /cmd/status_json 

In this example, the ip address 192.168.11.13 is the system requesting the page (i.e. the IP address of the computer with your browser). This is the expected output, the /cmd/status_json part will change depending on what the javascript in the browser is requesting.

If you look on the Monitor page of the web interface it will show serial communications statistics. If you see the “not receiving data” message, the Monitor page will give more details as to the specific errors. My guess is that you have many timeouts, which means that you are not receiving anything form the generator. Try swapping your tx and rx lines and reload the software and web page.

wotseurba, 2018-10-13:
OK, went back to working through this.

Back to basics, ran: bash ./startgenmon.sh stop, then all 3 loopback test modes successfully

I then decided to jumper the return ground from the generator Molex pin 2 (previously also connected to DB-9 pin 5) to a ground pin on the GPIO.  Genmon communication immediately began working.

So, for those not installing the Raspberry PI inside the generator, and are using a wall plug power supply, you must bring the controller ground to the GPIO ground.

jgyates, 2018-10-13:
@wotseurba 

I am glad it is working for you. I will close this issue but feel free post here if you have related issues.

closed 2018-10-13 by jgyates