jgyates/genmon #192: Update to 1.14 started multiple instances of genmon and genserv

liltux, 2018-10-12:
@jgyates , I just did an update on our shop generator genmon over vpn via the web ui.  And I got the normal notifications when the software restarted,  but then they kept coming and coming and coming.  I ssh'd into the pi and issued "sudo bash startgenmon.sh stop", and it scrolled down the list of stopping all python programs.  But I was still getting updates.  Issued ps -e | grep python, and it list 3 python programs running still.  So I issued startgenmon.sh stop again, and it finally stopped.  I then did a reboot and everything is fine.  Not quit sure what happened, it was crazy.  32 alerts before I stopped it.

jimland58, 2018-10-12:
I also upgraded to V1.11.4 and had the same thing happen, got 60 alerts in about 10 min.  Had to power down since I couldn't get thru SSH/PuTTY which happens quite often for some reason.

jgyates, 2018-10-12:
If someone could submit their logs it may offer some clues.

jgyates, 2018-10-12:
@jimland58 and @liltux 

Are the multiple alerts you are seeing the alert that the program is starting and the initial status email? Or is the software reporting there is an alarm?

jgyates, 2018-10-12:
I checked in an update that is my best guess for a fix with the data I have.

The issue *i think* is related to restarting flask. I have code that will restart flask if flask generates an address already in use error or if it generates and I/O error. The I/O error is a relatively new addition. I removed the restart on I/O error. If flask ever generates this error it will not restart.

closed 2018-10-12 by jgyates

jgyates, 2018-10-12:
@liltux , I got your logs, looks like my guess was correct. I/O error on flask restart. I had the code set to wait 2 seconds then try a restart but it looks like flask is not wanting to let go of some resource. The behavior now will be that if an I/O error occurs within flask on restart then the web interface will not load. You will have to reload it manually. 

the longer term fix is to use apache instead of the flask development server, which I have not gotten around to implementing. There may be some trade offs in going this path. If we do then the apache server will not be restarted for a simple upgrade or settings change, but changing port numbers or web security settings may require a restart of apache.

liltux, 2018-10-12:
@jgyates , Sounds good.  Yes, the alerts were just notifications, no alarms.  After the first few I figured something was looping through and got a terminal and stopped the shennanigans.  
 If you use apache, the folks over at Let's Encrypt have a great tool for anyone to create ssl/tls certificates for using https.  Worth looking into if planning on making Genmon accessible out to the internet.  I have a pi I use for playing around with webserver stuff and webpage making.  I have it setup with a Let's Encrypt certificate and only accepts https connections.  Easy to setup and secure.  Just fyi.

jgyates, 2018-10-13:
@liltux 

I was able to find a way to reproduce the I/O error that flask was throwing. If you started or restarted the software from the console, then logged out of the console and restarted via the web app (software update or settings change), then the flask library would not reload immediately. This is new behavior that is related to the way I am loading the programs with genloader.py instead of just calling a script. To workaround, I changed the way I am loading the programs in genloader.py and redirecting the output of genserv.py to /dev/null. As a result, the output that flask normally prints will not show up on the console and we will should not see the I/O error.

Let me know if you see the problem again.

lmamakos, 2018-10-15:
Thanks for the change to redirect the output of genserv.py to /dev/null; perhaps all the various programs started up should have that as normal behavior?  I've always been uneasy starting/stopping all the processes from an SSH session, and then wonder what happens when I log out, and the standard input/output/error file descriptors stop working, and/or HUP signals might get delivered, etc.

Seems like there ought to be some standard library function to "daemonize" a process, to this end.

Also, if you're still thinking about adding in a full web server, I'd urge you to take a look at nginx rather than apache.  I think it's higher performance (which isn't really an issue here), and has a smaller footprint when installed.  I note that a number of embedded systems tend to use nginx or lighttpd as an alternative to apache.

jgyates, 2018-10-15:
@lmamakos 

I will take a look at nginx and lighthttpd when I get to this. There are several trade offs in making this change so I am happy to look at alternatives to apache.

Regarding the redirect, the other program's output is routed through the logging library when writing to the console output so the abandoned stream is handled by the logging module. The Flask library does not do this.