jgyates/genmon #194: Genmon email questions

hokie21, 2018-10-14:
I see that genmon starts to open an SMTP connection with my mail server every two minutes and then disconnects. Is this expected behavior?

Oct 13 20:34:02 gator sendmail[1248]: STARTTLS=server, relay=generac [192.168.0.31] (may be forged), version=TLSv1/SSLv3, verify=NO, cipher=DHE-RSA-AES256-SHA, bits=256/256
Oct 13 20:34:02 gator sendmail[1248]: w9E1Y101001248: generac [192.168.0.31] (may be forged) did not issue MAIL/EXPN/VRFY/ETRN during connection to MTA
Oct 13 20:36:03 gator sendmail[1275]: STARTTLS=server, relay=generac [192.168.0.31] (may be forged), version=TLSv1/SSLv3, verify=NO, cipher=DHE-RSA-AES256-SHA, bits=256/256
Oct 13 20:36:03 gator sendmail[1275]: w9E1a2Iv001275: generac [192.168.0.31] (may be forged) did not issue MAIL/EXPN/VRFY/ETRN during connection to MTA

Is there any easy way to cause genmon to send a test email to the configured email addresses?

jgyates, 2018-10-14:
email is normally sent on these occasions:

1) when . the software is stared there are typically 2 emails sent.  the first is just and email that lets you know the software is starting, the second is the initial status of the generator 
2) If there is a status change on the generator, this includes alarms, warnings, or the generator switch changes state.

Other emails can be sent to the developer if you have "Automated Feedback" enabled . and if the software encounters a condition in the generator that is unknown.

The fact that this is repeating every two minutes is a good indicator that your email is enabled, but not configured properly. If errors occur in the email send or receive process, the errrors are logged to /var/log/mymail.log. The code will retry on errors, some of the errors retry at at 2 minute interval. 

If you don't want to use email you can disable incoming and outgoing email on the settings page.

The easiest way to test email is to just start the software. If you get an email, then it is working. If you don't then I would look at the file /var/log/mymail.log (the errors are time stamped so you should be able to match that up with your aboe log.

This file handles all email:

https://github.com/jgyates/genmon/blob/master/genmonlib/mymail.py

You can search it for the number 120 (i.e. 120 seconds or 2 min) and that is likely where it is failing. 

Let me know if you have any questions.

buschauer, 2018-10-14:
see attached. Had a bug in genmon.css that came up when I compiled it. 
So that's why I attached that again. Rest is now all good from my side - 
I'm fine if you want to publish it.

Michael

/* BEGIN RESET*/
html, body, applet, object, iframe,
h1, h2, h3, h4, h5, h6, p, blockquote, pre,
a, abbr, acronym, address, big, cite, code,
del, dfn, em, img, ins, kbd, q, s, samp,
strike, strong, sub, sup, tt, var,
b, u, i, center,
dl, dt, dd, ol, ul, li,
fieldset, form, label, legend,
table, caption, td, tr, 
article, aside, canvas, details, embed,
figure, figcaption, footer, header, hgroup,
menu, nav, output, ruby, section, summary,
time, mark, audio, video {
    margin: 0;
    padding: 0;
    border: 0;
    font-size: 100%;
    font: inherit;
}
/* BEGINNING: HTML 5 reset for older browsers */
body {
    line-height: 1;
}
ol, ul {
    list-style: none;
}
blockquote, q {
    quotes: none;
}
blockquote:before, blockquote:after,
q:before, q:after {
    content: none;
}
table {
    border-collapse: collapse;
    border-spacing: 0;
}
/* END: HTML 5 reset for older browsers */
input, select {
    font-size: 100%;
    font: inherit;
    vertical-align: middle;
    margin:0;
    padding:.15em;
    border:1px solid #999;
    outline:0;
    border-radius:3.01px;
    box-shadow:inset 0 1px 2px rgba(0,0,0,0.15);
    -webkit-border-radius:3.01px;
    -webkit-box-shadow:inset 0 1px 2px rgba(0,0,0,0.15);
}
body {
    font-size: 18px;
    font-weight: bold;
    font-family: Arial;
    background-color: white;
    margin: 0;
    /*    overflow: hidden; */
    height:100%;
}
html {
    height:100%;
    /* overflow: hidden; */
[... 603 lines cut ...]
    font-size: 15px;
    height: 30px;
}

#footer.alert {
    background-color: #ffcccc;
}

#myheader td {
    vertical-align: middle;
}

#footer td {
    vertical-align: middle;
}

#allnotifications td {
    padding: 3px;
}

#allsettings td {
    padding: 3px;
}

.ui-helper-hidden{display:none}.ui-helper-hidden-accessible{border:0;clip:rect(0 0 0 0);height:1px;margin:-1px;overflow:hidden;padding:0;position:absolute;width:1px}.ui-helper-reset{margin:0;padding:0;border:0;outline:0;line-height:1.3;text-decoration:none;font-size:100%;list-style:none}.ui-helper-clearfix:before{content:"";display:table;border-collapse:collapse}.ui-helper-clearfix:after{content:"";display:table;border-collapse:collapse;clear:both}.ui-helper-clearfix{min-height:0}.ui-helper-zfix{width:100%;height:100%;top:0;left:0;position:absolute;opacity:0;filter:Alpha(Opacity=0)}.ui-front{z-index:100}.ui-state-disabled{cursor:default!important}.ui-icon{display:block;text-indent:-99999px;overflow:hidden;background-repeat:no-repeat}.ui-widget-overlay{position:fixed;top:0;left:0;width:100%;height:100%}.ui-accordion .ui-accordion-header{display:block;cursor:pointer;position:relative;margin:2px 0 0;padding:.5em .5em .5em .7em;min-height:0;font-size:100%}.ui-accordion .ui-accordion-icons,.ui-accordion .ui-accordion-icons .ui-accordion-icons{padding-left:2.2em}.ui-accordion .ui-accordion-header .ui-accordion-header-icon{position:absolute;left:.5em;top:50%;margin-top:-8px}.ui-accordion .ui-accordion-content{padding:1em 2.2em;border-top:0;overflow:auto}.ui-autocomplete{position:absolute;top:0;left:0;cursor:default}.ui-button{display:inline-block;position:relative;padding:0;line-height:normal;margin-right:.1em;cursor:pointer;vertical-align:middle;text-align:center;overflow:visible;text-decoration:none}.ui-button:link,.ui-button:visited,.ui-button:hover,.ui-button:active{text-decoration:none}.ui-button-icon-only{width:2.2em}button.ui-button-icon-only{width:2.4em}.ui-button-icons-only{width:3.4em}button.ui-button-icons-only{width:3.7em}.ui-button .ui-button-text{display:block;line-height:normal}.ui-button-text-only .ui-button-text{padding:.4em 1em}.ui-button-icon-only .ui-button-text,.ui-button-icons-only .ui-button-text{padding:.4em;text-indent:-9999999px}.ui-button-text-icon-primary .ui-button-text{padding:.4em 1em .4em 2.1em}.ui-button-text-icon-secondary .ui-button-text{padding:.4em 2.1em .4em 1em}.ui-button-text-icons .ui-button-text{padding:.4em 2.1em .4em 1em;padding-left:2.1em;padding-right:2.1em}input.ui-button{padding:.4em 1em}.ui-button-icon-only .ui-icon,.ui-button-text-icon-primary .ui-icon,.ui-button-text-icon-secondary .ui-icon,.ui-button-text-icons .ui-icon,.ui-button-icons-only .ui-icon{position:absolute;top:50%;margin-top:-8px}.ui-button-icon-only .ui-icon{left:50%;margin-left:-8px}.ui-button-text-icon-primary .ui-button-icon-primary,.ui-button-text-icons .ui-button-icon-primary,.ui-button-icons-only .ui-button-icon-primary{left:.5em}.ui-button-text-icon-secondary .ui-button-icon-secondary,.ui-button-text-icons .ui-button-icon-secondary,.ui-button-icons-only .ui-button-icon-secondary{right:.5em}.ui-buttonset{margin-right:7px}.ui-buttonset .ui-button{margin-left:0;margin-right:-.3em}input.ui-button::-moz-focus-inner{border:0;padding:0}button.ui-button::-moz-focus-inner{border:0;padding:0}.ui-datepicker{width:17em;padding:.2em .2em 0;display:none}.ui-datepicker .ui-datepicker-header{position:relative;padding:.2em 0}.ui-datepicker .ui-datepicker-prev,.ui-datepicker .ui-datepicker-next{position:absolute;top:2px;width:1.8em;height:1.8em}.ui-datepicker .ui-datepicker-prev-hover,.ui-datepicker .ui-datepicker-next-hover{top:1px}.ui-datepicker .ui-datepicker-prev{left:2px}.ui-datepicker .ui-datepicker-next{right:2px}.ui-datepicker .ui-datepicker-prev-hover{left:1px}.ui-datepicker .ui-datepicker-next-hover{right:1px}.u [... cut ...] e-block;padding:2px 0 0;border-left:1px solid #439646;-webkit-border-radius:0 2px 2px 0;-moz-border-radius:0 2px 2px 0;border-radius:0 2px 2px 0;-webkit-box-sizing:border-box;-moz-box-sizing:border-box;box-sizing:border-box}.selectize-control.plugin-remove_button [data-value] .remove:hover{background:rgba(0,0,0,0.05)}.selectize-control.plugin-remove_button [data-value].active .remove{border-left-color:#00578d}.selectize-control.plugin-remove_button .disabled [data-value] .remove:hover{background