jgyates/genmon #196: Random Email Messages from Evo 2.0

jimland58, 2018-10-18:
### Expected Behavior

Only get email messages if actual change in Engine, Switch or Alarm status

### Actual Behavior

Got 2 emails back to back (with 10 or so seconds) indicating a state change but there was no actual change in the Engine, Switch or Alarm status,  Seen this happen once or twice before.  

### Steps to Reproduce (including precondition)

Random,  Unit is in Auto

### Screenshot or Pictures relating to the problem (if possible)

HERE ARE THE 2 EMAILS I GOT WITHIN 12 SECONDS OF EACH OTHER
**1ST RANDOM EMAIL:**

Time: 08:10:33 PM
Date: Wed 17-Oct-2018

NOTE: This message is a notice that the state of the generator has changed. The system is not in alarm.

Status : 

    Engine : 
        Switch State : System in Alarm
        Engine State : Off - Ready
        Battery Voltage : 13.60 V
        RPM : 0
        Frequency : 0.00 Hz
        Output Voltage : 0 V
        Output Current : 0.00 A
        Output Power (Single Phase) : 0 kW
        Active Rotor Poles (Calculated) : 0

        Unsupported Sensors : 
            Raw RPM Sensor : 0
            Frequency (Calculated) : 0.0 Hz
            Battery Charger Sensor : 1.08
            Battery Status (Sensor) : Charging
            Unknown Sesnor 33 : 65236
            Unsupported Sensor 1 : 0.00
            Unsupported Sensor 2 : 65236
            Unsupported Sensor 3 : -1
            Unsupported Sensor 4 : 0

    Line State : 
        Utility Voltage : 245 V
        Utility Voltage Max : 250V 
        Utility Voltage Min : 235V 
        Utility Threshold Voltage : 156 V

    Last Log Entries : 

        Logs : 
            Alarm Log : 09/21/18 15:07:46 ESTOP Pressed : Alarm Code: 2800
            Service Log : 
            Run Log : 10/16/18 12:35:56 Stopped - Auto 

    Time : 
        Monitor Time : Wednesday October 17, 2018 20:10:31
        Generator Time : Wednesday October 17, 2018 20:10

No Alarms: 0001:00003f3d

**2ND RANDOM EMAIL:** 

Time: 08:10:45 PM
Date: Wed 17-Oct-2018

NOTE: This message is a notice that the state of the generator has changed. The system is not in alarm.

Status : 

    Engine : 
        Switch State : Auto
        Engine State : Off - Ready
        Battery Voltage : 13.60 V
        RPM : 0
        Frequency : 0.00 Hz
        Output Voltage : 0 V
        Output Current : 0.00 A
        Output Power (Single Phase) : 0 kW
        Active Rotor Poles (Calculated) : 0

        Unsupported Sensors : 
            Raw RPM Sensor : 0
            Frequency (Calculated) : 0.0 Hz
            Battery Charger Sensor : 0.95
            Battery Status (Sensor) : Charging
            Unknown Sesnor 33 : 65236
            Unsupported Sensor 1 : 0.00
            Unsupported Sensor 2 : 65236
            Unsupported Sensor 3 : -1
            Unsupported Sensor 4 : 0

    Line State : 
        Utility Voltage : 245 V
        Utility Voltage Max : 250V 
        Utility Voltage Min : 235V 
        Utility Threshold Voltage : 156 V

    Last Log Entries : 

        Logs : 
            Alarm Log : 09/21/18 15:07:46 ESTOP Pressed : Alarm Code: 2800
            Service Log : 
            Run Log : 10/16/18 12:35:56 Stopped - Auto 

    Time : 
        Monitor Time : Wednesday October 17, 2018 20:10:42
        Generator Time : Wednesday October 17, 2018 20:10

No Alarms: 0001:00000000

IMMEDIATELY CHECKED REGISTER 0001  HISTORY AND IT SHOWED HEX:00000000 WITH NO CHANGE DURING THE PREVIOUS 10 MINUTES

ALARM LOG DIDN'T SHOW ANYTHING WENT INTO ALARM

ONLY THING I SEE CHANGED BETWEEN THE DATA REPORTED IN THE TWO EMAILS IS THE TIME, SWITCH STATE,  BATTERY CHARGE SENSOR AND THE No Alarms: DATA WHICH I DON'T UNDERSTAND

### Your Environment

- Generator Model: G0070432
- Generator Registers: N/A
- Genmon Version: V1.11.6

jgyates, 2018-10-18:
Hi @jimland58 

this is interesting. this may be a new status from the Evolution 2.0. Do you notice any new log entries? the lower half of register 0001 is the switch state and alarm or warning info. Did you notice if your firmware was updated? If you are connected via the web this may be a status for that condition (this is just a guess). Also, can you look .at the monitor page. Do you notice any communication errors under Serial Stats?

Genmon will report a status change if register 0001 ever changes. Another thing you can do is enable "Automated Feedback" in on the settings menu. If you have outgoing email enabled and "Automated Feedback" enabled it will send and email to me with the current registers for review if there are any unknown data read from the controller, like in this case.

jgyates, 2018-10-18:
Also, the web app only shows history as far back as the web app is running (i.e. you have to have the web page up to get history). the monitor does not save history, but the web app does.

jgyates, 2018-10-18:
Is your firmware version reporting anything other than 1.05 (see the Maintenance page for firmware version)

jimland58, 2018-10-18:
No new entries in the Alarm, Service or Run logs.

Don't think the Firmware on the controller has updated.  The Maintenance page shows Firmware Version: V1.05.

Checked the Serial Stats and 0 errors on CRC, Modbus exceptions, Validation Errors, Discarded Bytes and Comm Restarts.  It did show 19 packet timeouts with 0.00% packet timeout % errors.  Packet count was at 9984539 and counting.  64.27 packets/second and average transaction time was 0.0307 sec

I did have the web app running when this happened and when I checked the 10 minute history immediately after it happened, register 0001 showed no change.  

I have turned on the Automated Feedback.

What do the numbers mean behind the No Alarms: parameter in the email?

jgyates, 2018-10-18:
The numbers are the contents of register 0001. Can you send me your log files? You can send them with v1.11.6 by entering this command in your browser. youripaddress:8000/cmd/sendlogfiles

liltux, 2018-10-18:
Im not sure what you might find, but i do know some residential dealers were complaining, on EVO2 units with the wifi Mobillink, that they were getting random check/inspect battery alarms.  When the dealers would go on site to inspect they could find no fault history and the batteries were good.  this may be similar response? just fyi, interested what you might find.

jimland58, 2018-10-18:
Thanks!  Just sent the log files.  

So was does the 3f3d mean in Register 1 (No Alarms: 0001:00003f3d)?

jgyates, 2018-10-18:
I don't know. I have never seen this value from this register before. Normally what happens when the software runs into an unknown value in  register 0001 is that, if the automated feedback is on I will get an email with the register values and I can either look at the generator logs (from the registers sent) or the state of the generator and determine the value. In this case I don't have any information other than the register apparently changed value for some reason and changed back. Normally, in an alarm condition the controller sets the alarm code in the lower half of 0001 and the value stays there until the  alarm is reset by the operator, an entry is created in the alarm log and, on Evolution the E-Code (the E-Code is the code is the published error code value that is displayed on the controller LCD, for example 2700 or 2701). The value is reg 0001 in an alarm is an code that is internal to the firmware so it is not the same as the E-Code.

If this occurs again see if you have the time out errors again. The controller may be busy doing something related to the built in mobile link and the timeout on modbus and the unknown value of 0001 may be related.

My best guess at the moment is that this is a bug in the firmware since the Evolution 2.0 is so new, but this is only a guess.

jimland58, 2018-10-19:
Thanks.  What I don't understand was that the web app was running when these emails came across and so within a couple minutes of getting the emails I checked the 10 minute history on Reg 0001 and it didn't show any change, it stayed at 0.  So is it possible for genmon.py to see a state change in Reg 0001 and send out the emails but that genserv.py/web app not pickup a change in Reg 0001 with the 10 minute history?

jgyates, 2018-10-19:
The app genmon.py reads the registers over the serail link via the modbus protocol and saves the values. If and alarm condition is detected or an unknown value and email is sent. This is the core function of genmon.py and since it is modbus the software must poll the generator, therefore the responsiveness is a factor of how fast the software can read the registers. This code has been optimized to read register 0001 at a higher rate than other registers.

The app genserv.py is the web server, it makes a request from genmon.py when a web page is loaded. The web app (javascript which runs in the browser, not on the pi) makes requests for data to genserv.py (via TCP/IP) which then requests  the data from genmon.py. Since the web app can only has to run with other apps on whatever system the browser is running it makes request about every 3 seconds. 

The short answer is that genmon.py is much more responsive than the web app so the register history (which is maintained by the web app) is really for debugging and diagnosing issues, but it has it's limits.

jimland58, 2018-10-20:
Got another random email on 10/19 @ 15:23:18 that showed Register 0001 as: 62620000 and then another email 11 seconds later showing Register 0001 going back to 00000000.  Wasn't home to check the serial stats when it happened but looking at it now I still show no errors.  I do see where the Packet Timeout increased by 1 from 19 to 20 sometime since I last checked it 3 days ago.  

I looked back at a random one I got back on 10/15 and it showed Register 0001 as: 20200000 and again 10 seconds later cleared.  

I don't see how packet timeouts could cause this and if it was data that was corrupted during the transmission because of noise I would think that it would  show CRC errors but not getting any so it does sound like a bug in the Evo 2.0 firmware ???

jgyates, 2018-12-05:
@jimland58 

I am closing this issue for now. If you have any other data on this feel free to post it to this thread.

closed 2018-12-05 by jgyates

jgyates, 2019-06-26:
FYI, the random email messages can be disabled now. See this:

https://github.com/jgyates/genmon/issues/281