jgyates/genmon #197: wifi extension cable question

ts4s, 2018-10-19:
Does anyone know the name to the wifi extension cable being used in the attached photo?
[image]

Tried to find the original thread with it to see if it is listed. Anyways, I'm looking to do the same with my wifi connection. Currently I have a usb extension cable hanging out of my generator, so the wifi module is expose to the weather. Really want to fix this before winter sets in.

I'm thinking it's this but not sure on the connectors:
https://www.amazon.com/dp/B07DLSQQFW/ref=sspa_dk_hqp_detail_aax_0?psc=1

Thanks!

jgyates, 2018-10-19:
The connectors are called SMA connectors. I got these and an external antenna at www.data-alliance.net

https://www.data-alliance.net/sma-extension-cables-4-inch-to-6-feet/

There are a few variations of connectors 

RP-SMA - reverse polarity SMA
NP-SMA - normal polarity SMA

Both come in male and female varieties.

https://en.wikipedia.org/wiki/SMA_connector

It has been a while since I was using the antenna (switched to Ethernet) so I don't remember the exact one I used. If you use the RP and need the NP then your antenna will basically do nothing for you as the ground and signal will be reversed.

ts4s, 2018-10-19:
cool, so I should be fine with the normal polarity sma... Thank jgyates!

closed 2018-10-19 by ts4s

jgyates, 2018-10-19:
possibly, I dont remember if the wifi adapter is NP or RP. You could just get both cables (I think they are inexpensive) and try each one.

ts4s, 2018-10-19:
just completed my order with data-alliance, good prices and reasonable ship cost :-)

lmamakos, 2018-10-20:
The [USB WiFi dongle ](https://cgi.ebay.com/ws/eBayISAPI.dll?ViewItemVersion&item=311736220053&view=all&tid=855824660021) that I used had an RP-SMA connector arrangement.  I think for Wi-Fi products, the RP-SMA variation is most commonly seen due to the fiction that such an uncommon connector would make it "impossible" for the end-user to use a different antenna than the one that the FCC Part 15 device was tested against.  Which is pretty funny these days, because trying to find an normal SMA jumper (like for your GPS antenna or similar) is more difficult than the RP variation.

ts4s, 2018-10-28:
the RP-SMA connector is what I needed for my setup. I was able to get it install and clean up my setup. 

[image] Outside Antenna view
[image] Inside Antenna View
[image] Pi view
[image] Command strips work great for holding the Pi.

jgyates, 2018-10-29:
Thanks for the pics!