jgyates/genmon #198: Evo 2.0 was working, now not working after upgrade

beschaaflh, 2018-10-20:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

If you need to send you logs to the developer, if you email is setup and working properly you can issue this command in your browser:

     http://MYIPADDRESS:8000/cmd/sendlogfiles

Replace MYIPADDRESS to the IP address of you pi and port 8000 if you are using a port other than the default.

You can submit your registers from the Monitor page on the web interface.

### Expected Behavior

### Actual Behavior

Set up genmon on new Evo 2.0 11 KW air cooled generator two weeks ago; worked great; then connected cable to use WiFi.  Generac system upgrade by wifi on 10/15.  Now getting no data on genmon.  Repeated timeout on maintenance tab. Retried loopback test, and that works fine. Has anyone done a system upgrade and had a similar issue?

### Steps to Reproduce (including precondition)

{Please write here}

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: Generac 11kW Air cooled
- Generator Registers: {Please write here}
- Genmon Version: v1.11.8

jgyates, 2018-10-20:
What is your new firmware version?

liltux, 2018-10-20:
That was the first thing that came to my mind, but I checked Generac Service website and it shows no new firmware.  I can only find firmware 1.05 for Evo2.0, which has been the current firmware for some time now.  I am interested as well if the firmware has changed or not.

jgyates, 2018-10-20:
I am pretty sure @jimland58 is using 1.05 and his is working.

Have you tried restarting the software?

jimland58, 2018-10-20:
That’s correct.  Running 1.05 and works.

beschaaflh, 2018-10-20:
Thanks for the replies.  I was searching how to find firmware version directly from the controller keypad -- I've been through the menus and can't see this.  Based on documentation, it appears I would see version if I did a firmware upgrade by USB.  I went back to the Mobile Link website and there's a diagnostic screen that shows a firmware version of 1.5.58 (by luck I have a screenshot of the diagnostic screen when the generator was installed 3 weeks ago and the version was 1.3.54).

I've also done several resets of the software and even swapped the RX and TX.  Everything was working a week ago, so don't think the cable or setup is the issue.  Again, really appreciate your help.  Please let me know if you have any ideas what more I could try.

jgyates, 2018-10-20:
You should be able to see the version on the dealer -> display menu. To get to the dealer menu, from the top menu enter Up, Up, ESC, DN, Up, ESC, Up, ENTER, then you can enter the dealer menu.

Are you running the built in WiFi at the same time as genmon? I know that @jimland58 is doing this, but  there may be some additional caveats that we have not figured out. You could try to disconnect the wifi cable and see if that will let the controller communicate with genmon again. There could be some dependencies with the two serial outputs (for example which one was communicating first.

beschaaflh, 2018-10-20:
Thanks again.  I confirmed I have 1.05 firmware.  I've disabled wifi and disconnected the cable, but still no data to genmon.

beschaaflh, 2018-10-21:
Yes, I did the loopback test at the molex connector at the end of the cable (repeated this morning).  Unfortunately no traffic LEDs on my RS-232 converter.  I have a working genmon setup on my generator at our weekend place, so I'm planning to take this one up there next weekend and confirm it works on the Evo 1 generator.

beschaaflh, 2018-10-28:
An update.  I tried the new raspberry pi at our lake house this weekend and it works fine on the evo 1 generator.  I reconnected to the evo 2 generator at home, and I'm still getting packet timeouts and  don't seem to get any data.  I have wifi turned off on the evo 2.  As a reminder, about a month ago when evo 2 generator was installed the genmon worked.  I connected wifi a couple weeks later, and the mobile link website says there was a system update (apparently to 1.05).  Since then, the genmon software doesn't seem to work.

jgyates, 2018-10-29:
Hmm, I am guessing there is a minor version being used by generac. I know of some people using v1.05 but your Genlink snapshot shows 1.5.58. I am guessing the "58" part of this is what is significant. 

I have a few ideas that we could try if you are interested. We could see if the modbus address has changed or the serial rate. If generac is going the route of blocking all third party monitoring then this will effect companies like omnimetrix also. 

Of course another thing that is possible is that there is something wrong with he serial port on the controller. I have seen this before (I had to have a controller replaced and the one they were going to put in had a defective serial port). Maybe we can get another data point from another Evo2.0 user...

jimland58, 2018-10-29:
I'm running Genmon on Evo 2.0 with firmware 1.05 and also with MobileLink wifi running.  I updated the firmware when it was first installed about 6 weeks ago but haven't checked if there was another firmware update available.  Based on what I'm hearing now, I will not update until this problem is figured out.

jimland58, 2018-10-29:
Opened the Dealer Menu on the Evo 2.0 controller and it shows Firmware V1.05, Hardware V1.00, WIFI FW 01.05.58 and WIFI HW 01.00.    I'm running Genmon V1.11.6 and everything works as expected.

jgyates, 2018-10-29:
That is good info too. 

Interesting, there appears to be a piece we are missing as to why @beschaaflh is having an issue...I don't have any immediate ideas, I will think on this...

beschaaflh, 2018-10-30:
Thanks.  Appreciate both of your help on this.  I confirmed I have Firmware V1.05 and Hardware V1.0. I also have Boot Loader V0.7 and EEPROM V1.03.  The good news is this doesn't appear to be an House with the firmware versions.  Maybe it's an issue with the controller.  I did a bit of research if there's a way to "reboot" the controller.  Doesn't appear to be a simple process.  Think I'll reconnect the wifi .

jgyates, 2018-10-30:
I don't know about Evo 2.0 but for mine (Evo Liquid Cooled), if you unplug the connectors on the back of the controller and plug them back in it will power cycle the controller. All of your setting should remain. the Air cooled units get their config from a resistor pack in the generator harness wiring.

beschaaflh, 2018-10-31:
Thanks again @jgyates.  I'm out of town until this weekend but I'll try resetting the controller when I'm home.  I've also ordered a RS-232 adapter with RX/TX leds -- so hopefully can confirm if I'm getting data from the controller.

beschaaflh, 2018-11-19:
Good news!  I restarted the controller and we're back up and running!  Thanks @jgyates and @jimland58 for your help!

closed 2018-11-19 by beschaaflh

jgyates, 2018-11-19:
Thanks for the update. I am glad it is working now.

jonathanpisarczyk, 2019-06-12:
@beschaaflh which wire harness did you unplug on the back of the controller?  My firmware updated last night night and I lost contact with Generator. 
[image]

jonathanpisarczyk, 2019-06-12:
Another picture: 
[image]

jwtaylor310, 2019-06-12:
I encountered the same problem with my generator last night. Resetting the controller fixed it for me. Another thread here lists this as the procedure to reset the controller:

_Pull the F1 fuse - which is the fuse on top, and disconnect the small 2 wire white connector under the controller - it has about a 12 inch cable with a white sheath around it, this is the battery charge power supply. By removing the fuse and this cable connection the controller will be removed of all power supply. Wait a few seconds and install the fuse first and then connect the battery charge cable._

I didn't find that thread until after doing a reset using a different procedure:

1. Set controller to "Off".
2. Remove the battery cable from the negative battery post.
3.   Briefly (30 seconds or so) turn off the main disconnect in the transfer panel. This will disconnect the power for your house. The controller display should be blank. Turn the main disconnect back on.
4. Reconnect the battery cable.

Either of these procedures should work. When the controller restarts it will ask you to confirm the language and time and will re-establish the WiFi connection. Once that is done, be sure to press the "Auto" button to return to normal operation.

Good luck!

RRoyceus, 2019-08-09:
I had a similar problem.  Genmon had been working perfectly for months.  Last weeks we had a short loss of power; it only lasted 11 seconds. 12 seconds later I got a message that the monitor was not communicating with the controller.  Three minutes later communications was restored.  Nine minutes later communication was lost for good.  

I later tried the loop back test on my Pi, then at the generator.  Both passed.  Then I reset the controller. Still no communications. I finally ordered and replaced the RS232 serial port to TTl converter.  It finally works again.  I will be keep a backup serial converter in the future.  They cost $6.69 with free Prime shipping on Amazon