jgyates/genmon #199: No communication - all packets time out

wrvolz, 2018-10-22:
New install. Checked serial port with serialtest.py and that worked. All I'm getting is packet time outs. No connection. See Monitor page below. Restarted genmon, no change.

One question - you mentioned ground. I'm powering the PI from POE. So to get common ground as you mention, I connected Pin 2 on the Molex to Pin 5 on the DB9 connector. Was that correct? The Molex pin out also mentions a RS232 ground on Pin 4.
Thanks Bill

### Screenshot or Pictures relating to the problem (if possible)
[image]

{Please write here}

### Your Environment

- Generator Model: generac 22KW
- Generator Registers: can't read them.
- Genmon Version: 1.9.37

jgyates, 2018-10-22:
You mentioned that you tested with serialtest.py, did you run this with the molex pins connected as your loopback and with the cable attached?

This page has details on serial troubleshooting. If you have validated this data then I would check your ground. 

The molex ground pin is for Evolution 2.0 (the one . If you have an evolution 2.0 controller you can use this pin. It is probably a better bet to connect the ground from the molex (pin 2) to the raspberry pi header ground pin. If you have a well made RS-232 converter the ground on the 9 pin connector (pin 5) will be tied to the ground on that connects to the pi on the other side of the converter board so if that is the case then pin 5 or the pi would work.

wrvolz, 2018-10-23:
I have an Evolution 1.0 controller. I verified that the serial port works from the end of the cable. I checked the ground from the PI to the end of the cable and it's 0 ohms. I checked the resistance from battery ground to the PI and it's about 8K ohms. possibly since the generator is on house current. Actually it was alternating between infinity and 8K about once a second. I'm sure I'm using pin 2 on the molex - it's on the opposite side as the tx/rx pins and 2nd from the opposite end.
I also tried swapping pins 7 and 8 in case I have that wired wrong - same thing. Note the the serial test won't show if it's backwards, I think. Any ideas?
Thanks for your help.

jgyates, 2018-10-23:
a photo of your cable would help.

wrvolz, 2018-10-23:
It's kind of hard to see from a photo. The molex has 4 yellow wires on pin 5-8 and black on pins 1-4. And they squeeze together where they go into the case so you can't follow them. Here's the molex end:

[image]

The black wires that aren;t used are pulled aside and you can see the ground wire in pin 2. The straight yellow wire is on pin 8, the more bend on on pin 7. These will be the longer and shorter of the two in the next photo.
[image]
 You can the shorted going to pin 2 on the 9 pin, and the longer going to pin 3. The ground wire goes to pin 5. 

Now in your figure on making a cable or DCe and DTE that shows the wiring from the pin to the molex, is that a male or female molex?  The holes make it look like a male, but the latch on the side makes it look like a female. I wired it as if it's a male, but if it's female, I've got 5/6 swapped with 7/8 and ground is wrong too.

wrvolz, 2018-10-23:
I don't know what I did, but it suddenly started to work. Didn't change any wiring. Just traced that I actually had the wires correct. Started motor and did not get any CRC errors so that's good, unless the noise increases with generator load. I must have some loose wire somewhere.

wrvolz, 2018-10-23:
What do I need to do to get mail working. I entered parameters in both mymail.conf and the settings page. Is there a way to test mail?

Also, I started the generator remotely, but I could not shut it off. The "stop generator' was kind of greyed out green and the button seemed disabled. The generator was in Auto mode. I had to go and shut it off manually. Not that I would use this, but just something that doesn't seem to be working. Thanks

jgyates, 2018-10-23:
if you email is working, you will get an email sent to the notify email address when the program is started (or restarted, for example when you apply setting it is restarted).

Email errors are logged to the file /var/log/mymail.log 

editing the mymail.conf file in the directory where the program files reside (i.e. /home/pi/genmon) will do nothing. All *.conf are copied to the /etc directory and the /etc/ directory is where the programs read the files. It would not recommend editing the files unless you are confident in your settings. I recommend using the web interface to modify your settings. Setup your email on the settings page and the email address to send notices to on the Notifications page.

Most email providers have additional security enabled and depending on your provider. Gmail for example has an option to use "less secure apps" which must be enabled if you do not use a gmail approved email app (https://support.google.com/accounts/answer/6010255?hl=en). If you have two factor authentication enabled you would need to create a one time password. A common solution to work around these limits imposed by email providers is to create a dedicated email address using free email providers like gmail and enable the required security settings to allow an third part app to send email. This allows you to keep your security settings high for your personal email and lower them for single use email that can be dedicated to things like a generator monitor.

Regarding the remote start / stop commands. According to your earlier post your are running version 1.9.37. The current version is 1.11..10. I would update to the latest version before we look into any issues you are describing. This can be done from the Monitor page before version V1.11.7 and on the About page starting with version 1.11.7.

jgyates, 2018-10-24:
@wrvolz , closing this issue for now since your main problem is resolved. Feel free to comment here if you have any other questions.

closed 2018-10-24 by jgyates