jgyates/genmon #200: Pi Zero--disconnected from server messages and high CPU

hedgemybets, 2018-10-27:
### Expected Behavior

Web UI opens showing Status, Maintenance, Outage, Logs, Monitor screens with current data

### Actual Behavior

Web UI shows Status gauges, but all values zero. Maintenance, Outage, Logs, and Monitor screens no longer display at all. Notifications, settings, add-ons, and about screens load quickly and display normally.

"Disconnected from server" count on bottom left continues to climb, even when on the screens that do work.

CPU utilization of genmon is <30% before logging in to the UI, but stays between 85% and 95% after logging in.

### Steps to Reproduce (including precondition)

Installed software, finished connecting everything and bring the system up on October 17. All has worked reasonably well before last nite. Also I did upgrade on the 20th. Occasionally viewing on iPhone has not been responsive, but Chrome on PC seems to work fine. Last nite we had 6 hour power outage, the first since installation, so I had been watching the kW output graph and all was working. This morning, did not see data on the graph, so I click on 1 day view, and nothing happens. In troubleshooting, there are no errors in any of the logs, have stopped/started with the startgenmon script, rebooted the Pi Zero, and also upgraded to the latest version. Using "top" via ssh, the only thing I notice is that python CPU utilization spikes after logging in to the web UI, and even closing the browser does not help. The PID with the high CPU is the genmon process.

The only other clue to provide is that my Internet service went down sometime after the power went out and this morning I was initially accessing the UI from my LAN without any Internet service, but now Internet service is back, and the problem remains, even after restarts, reboots, and upgrade.

### Screenshot or Pictures relating to the problem (if possible)

[image]

### Your Environment

- Generator Model: 22KW/999 Guardian with model RTSW200A3 transfer switch
- Generator Registers: can't see these, sent from UI, but not sure if this is working.
- Genmon Version:  V1.11.14

jgyates, 2018-10-27:
The disconnected from server messages means that the web app in your browser can not communicate with the web server running on the pi. 

Can you ping the pi IP address? If you can not, then this is your problem. You have lost connection to your pi. Either your pi is down or your pi lost the wifi connection.

In order for the registers to be submitted your email and internet connection have to be working (I have not received your registers or logs).

Based on the info that you have provided my best guess is that your pi has lost it's internet connection.

jgyates, 2018-10-27:
if your refresh the web page, does any page show up? If it does not then that is also an indication your pi is not longer on your network or on the same IP address. Is you the IP address of your pi static or dynamic?

hedgemybets, 2018-10-27:
Thanks for the quick reply! Yes, I can ping the Pi with no packet loss and
my ssh connections are very responsive to monitor CPU using "top", even
while logged into the UI. Tested up to 5K byte pings, no loss. IP is DHCP,
but ifconfig shows the IP address I'm using, no TCP TX or RX loss or errors
in the stats.  I don't think there is any issue with the Pi's Internet
connectivity.

I inspected the web traffic with the Chrome developer tools and the calls
look like they are timing out at 4 seconds, see screenshot below. Later in
the list,I switched to the Maintenance view, and then some calls then
showed quickly as failed.
[image: Capture.PNG]
I switched the website back to non-SSL port 8000, just to see if the SSL
was having an effect, same issue. Any idea why genmon is consuming so much
CPU after I log in?

I haven't managed to get email settings to work, I need SSL, when I turn
that on, I get this:
2018-10-27 15:12:21,923 : Error SMTP Init : SSL:<True>: [SSL:
WRONG_VERSION_NUMBER] wrong version number (_ssl.c:661) : mymail.py:304

Thanks again for your help.

hedgemybets, 2018-10-27:
More info:
I looked at the performance of the underlying web service calls and before bringing up the main web page, I get immediate responses to everything but maint_json, which takes about 12-15 seconds to load and CPU goes to 95% when I make that call. The response appears valid, see below (I removed the serial number):

{"Maintenance": {"Model": "G0070421", "Generator Serial Number": "XXXXXXXXXX", "Controller": "Evolution, Air Cooled", "Nominal RPM": "3600", "Rated kW": "20", "Nominal Frequency": "60", "Fuel Type": "Propane", "Engine Displacement": "999 cc", "kW Hours in last 30 days": "63.48", "Fuel Consumption in last 30 days": "16.64 gal", "Estimated Fuel In Tank": "308.36 gal", "Controller Settings": {"Calibrate Current 1": "1513", "Calibrate Current 2": "1537", "Calibrate Volts": "1017"}, "Exercise": {"Exercise Time": "Weekly Saturday 08:00 Quiet Mode Off"}, "Service": {"Service A Due": "190 hrs or 04/19/2020 ", "Service B Due": "390 hrs or 04/19/2022 ", "Total Run Hours": "9", "Hardware Version": "V1.06", "Firmware Version": "V1.20"}}}

All others load in a second or two, with no CPU spikes. Could it be that that call is having an issue with my controller's data format?

Thanks...

jgyates, 2018-10-27:
Can you post your /var/log/genmon.log file?

It may be later tonight before I can look at it.

jgyates, 2018-10-28:
Now that I think about it and easier thing to do to test this theory is to enable the option "Disable Power / Current Display" on the setting page and save the setting and then see if the problem still exist. if it does not show the problem, then this is likley due to the processing required for the fuel calculations for 6 hours of kw readings. The log stores entries every 10 seconds and the maintenance page parses this three times (fuel in 30 days, kw hours in 30 days and fuel in tank). Needless to say this is a bit much for the file size and the power of a pi zero. Once you validate this I can work on a fix.

jgyates, 2018-10-28:
If you can validate, would it be possible to save your kwlog.txt file, zip it and attach it to this thread so I can test with the file?

jgyates, 2018-10-28:
I tested with your kwlog.txt file on my system (pi 3 with class 10 card). There is a small delay when pulling up the maintenance page, but the system does work as expected. I can optimize the way the file is read and how the maintenance page accesses it, but the speed of your SD card also plays a large factor.

jgyates, 2018-10-29:
I checked in an update that should be kinder to the SD card when calculating the power and fuel estimates. Let me know if you have any issues with the update. I will close this thread but feel free to comment here if you have any other questions / comments on this topic.

closed 2018-10-29 by jgyates